MichaIng/DietPi #3350: DietPi-Software | PiVPN: Make install unattended

MichaIng, 2020-01-20:
The PiVPN installer allows unattended install via given inputs. We could either use this always or depending on interactive/non-interactive call (automated firstrun installs). Needs some testing to find working+reasonable defaults.

bastianpaetzold, 2022-06-04:
Would it be enough to provide the location of an option file via dietpi.txt for handling the first boot case? Or would you prefer to have all options directly in dietpi.txt?

MichaIng, 2022-06-04:
Ah, it supports an options file? Then I'd say that would be sufficient, even better, so we don't need to maintain/update supported options, or break when those change. Is there an up-to-date sample options file URL, which we can add to our documentation and as a comment to `dietpi.txt`?

bastianpaetzold, 2022-06-04:
There are 2 files here: https://github.com/pivpn/pivpn/tree/master/examples
They seem pretty up-to-date

MichaIng, 2022-06-04:
That is great, serves for documentation pretty well. So we check whether `/boot/unattended_pivpn.conf` exists, set `aSOFTWARE_INTERACTIVE[$software_id]=1` only if it does not, else call the installer with `--unattended /boot/unattended_pivpn.conf`.

Not sure whether we shall always call the installer unattended when the config file exists, even when `dietpi-software` was called interactively? I'd say so. But shall we keep the file where it is, or (re)move it so that a reinstall will be always interactive?

bastianpaetzold, 2022-06-04:
I personally would expect to be able to set all options manually when I call `dietpi-software` interactively. But if the file is removed after an unattended install, it would be fine too, because you have to actively put a new file there and therefore there is no risk of confision why the script doesn't ask for input.
Instead of deleting it, it might actually better to rename or move it so you can quickly put it back in case something goes wrong during the install or you mad a mistake in the option file.

closed 2022-06-25 by MichaIng