MichaIng/DietPi #3351: Kodi crashes on boot (solution found)

sbrl, 2020-01-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster` or `10.2`
- Kernel version | `Linux MACHINE_NAME_HERE 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux`
- SBC device | `RPi 4 Model B (armv7l)` or (EG: RPi3)
- Power supply used | 2.4A brick with type-a to type-c 0.9m converter cable
- SDcard used | SanDisk 16GiB

#### Additional Information (if applicable)
- Software title | Kodi)
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yep, since is basically a fresh DietPi install

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. Install kodi
2. Set kodi to autostart
2. Reboot
3. Note that a kodi crashlog is present in your home directory

#### Expected behaviour
Kodi should start without crashing

#### Actual behaviour
Kodi crashes

#### Extra details
This is happening because dietpi does not check to make sure that the target user that kodi autostarts under is a member of the `video` group. This is vital to kodi starting correctly. Without this you'll get a kodi crash - but the crash doesn't actually give you an error message :-(

By guesswork I figured out the above. The following command fixed it:

```bash
sudo usermod -a -G video dietpi
```

Just wanted to leave a note for future DietPi users to leat them know how to fix the issue when they run into it :smiley_cat:

MichaIng, 2020-01-20:
@sbrl 
Many thanks for your report.

Due to a bug the dietpi user is not added to the bunch of system groups, including video. Usually not an issue since all services which require special access run as their own user which is added to the required groups on install. Also I plan to remove the `dietpi` user and keep the `dietpi` group for shared software permissions (downloader backend => downloader frontend => media players <= network shares...) only, on new images after DietPi v6.29 release. Instead, on first login it will be offered to create a custom user account.

To choose a user for autologin, other than root, is a relatively new feature and not yet 100% nicely implemented. Adding the user automatically to groups, required to run the autostart program, is one of them. Thanks for sorting out the Kodi case already. Will add this for next release.

Besides the video group, I could have imagined that the `render` group is required as well, which allows access to DRM nodes since Debian Buster. And Kodi 18 uses DRM if I am not mistaken 🤔. But you do not get any error messages or something when starting Kodi as dietpi user, respectively see error messages on console, after exiting Kodi again, right?

sbrl, 2020-01-23:
Ah, I see. Thanks for the info!

That would be a welcome feature for user accounts.

Oops, I should probably add the `render` group to my setup then! I'm currently sourcing a keyboard to setup the feature where you use your phone as a remote.

The physical console of my Pi isn't easy to read, unfortunately - as I'm using an adapter to plug it into an ancient standard definition CRT television (no, I can't afford to replace it). All I had to work from are the rather useless crash logs that appear in my home folder.

It would be nice if the kodi installer created a `kodi` user with it's own home folder, actually.

MichaIng, 2020-01-23:
@sbrl 
I was thinking about the same, having a service that could run much earlier than login prompt. We could do it similar like for Amiberry, a fast boot and a standard boot option.

Actually what I liked best is three options for simply all autostart programs:
1. Manual (console) login: On desktops this will be LightDM instead, actually not benefit but users might be confused by "ugly" console prompt prior to "beauty" graphics desktop 😄. So program will start automatically AFTER manual login.
2. Autologin: Program will start automatically after automated login, hence from within a user login session. Has not really an effect on programs started as systemd units, but assures they are started after all boot steps and background services and after exiting you're in your login shell session already.
3. Nologin: Program will start outside of any user session, like Amiberry fast boot currently. This can then happen based on required `After=` declarations much earlier and with least overhead. A systemd unit is required in this case with proper program run user etc. Most elegant definitely. Requires `systemd-logind` active as well, so as fast as the program is exited, a getty (login prompt) will appear.

sbrl, 2020-01-24:
That sounds like a great plan! I guess users would be able to choose between them?

MichaIng, 2020-01-24:
Yes that's the aim. Would make all of this much more flexible.

MichaIng, 2020-03-17:
Light solution for now: #3424
Make it more flexible out of scope for DietPi v6.29 at least. I'll keep it in mind.

closed 2020-05-03 by MichaIng