MichaIng/DietPi #3352: dietpi-banner - custom banner entry - wireguard status

bbsixzz, 2020-01-21:
Adding `wg` as a custom command in `dietpi-banner` doesn't look that great.

Maybe add its own entry like NordVPN status?

[image]

[image]

MichaIng, 2020-01-21:
@bbsixzz 
Jep, that output should be parsed and printed more compact for banner. If you have some suggestion I can write some code to do that and implement into DietPi.

I like the idea of having service status like these available in the banner. To not have the banner code itself growing, I think we need to implement some infrastructure where custom entry can be stored as files (e.g. `/DietPi/dietpi/dietpi-banner.d/`) where users and software installs can place custom files in a certain format, containing title, description and command to be run and printed in Banner.

bbsixzz, 2020-01-21:
The WAN IP option in `dietpi-banner` is currently a good indicator of VPN status.

Later with https://github.com/MichaIng/DietPi/issues/3285 it would be nice to show in banner which services are using WireGuard.

Also transfer received/sent and latest handshake would be helpful without cluttering the banner.

The infrastructure idea is sound as the complexity grows.