MichaIng/DietPi #3354: Debmatic / Pivccu / Raspberry Pi 3

Ewies, 2020-01-23:
I try to get the Debmatic or Pivccu to work but somehow it doesn't start under Dietpi.

Would support be possible?

I did the manual installation but I don't get any errors but somehow it doesn't start ...

`root@RetzlaffHomeServer:~# sudo debmatic-info
debmatic version: 3.49.17-40
Kernel modules: Available
Raw UART dev:   Not available
Rasp.Pi3 UART:  Assigned to GPIO pins
HMRF Hardware:  unknown
HMIP Hardware:  unknown
Board serial:   unknown
Radio MAC:      unknown
SGTIN:          unknown`

`root@RetzlaffHomeServer:~# sudo systemctl status debmatic
● debmatic.service - debmatic
   Loaded: loaded (/lib/systemd/system/debmatic.service; enabled; vendor preset: enabled)
   Active: active (exited) since Thu 2020-01-23 13:46:58 CET; 13min ago
  Process: 442 ExecStart=/usr/share/debmatic/bin/initsystem.sh (code=exited, status=0/SUCCESS)
 Main PID: 442 (code=exited, status=0/SUCCESS)

Jan 23 13:46:39 RetzlaffHomeServer systemd[1]: Starting debmatic...
Jan 23 13:46:58 RetzlaffHomeServer systemd[1]: Started debmatic.`

Steffen

MichaIng, 2020-01-23:
@Ewies 
You need to assure that the UART is enabled and not blocked by e.g. serial login console. E.g.:
```
dietpi-config
> Advanced Options
> Serial/UART
> ttyS0 (mini UART) device: [On]
> ttyS0 console: [Off]
reboot
```

Ewies, 2020-01-23:
Thank you for your quick reply ....

I had already tried that in a few settings, but unfortunately also no success ...

`ttyAMA0 console         : [Off]
 ttyS0 console               : [Off]
●─ Toggle device ●
ttyS0 (mini UART) device: [On]`

`root@DietPi:~# debmatic-info
debmatic version: 3.49.17-40
Kernel modules: Available
Raw UART dev:   Not available
Rasp.Pi3 UART:  Not assigned to GPIO pins
HMRF Hardware:  unknown
HMIP Hardware:  unknown
Board serial:   unknown
Radio MAC:      unknown
SGTIN:          unknown`

the installation always runs without error messages, so the error must be somewhere in the setting right?

MichaIng, 2020-01-24:
@Ewies 
Yes must be a configuration issue. Which UART device does it actually want to use? It seems it doesn't see the one it looks at attached to the right GPIO pins. On RPi there are two UARTs, ttyS0 (mini UART) and ttyAMA0 (full physical UART). On RPi1/2 ttyAMA0 is the primary UART ready for serial console, on RPi3/4 its ttyS0, since ttyAMA0 is by default ready for Bluetooth. But this can be switched with an dtoverlay.

DietPi is shipped with defaults like Raspbian (Lite), hence whichever complete guide works for Raspbian on your RPi3 model, should work for DietPi as well.
In any case, if its about GPIO/UART and such, I am no expert. Read about about the configs, e.g.:
- https://www.raspberrypi.org/documentation/configuration/uart.md
- On the device: `more /boot/overlays/README`

Ewies, 2020-01-24:
I don't know much about UART either, but with a standard Raspberry Noobs-light installation and the normal installation path for Debmatic it works without any problems.

Dietpi has to bend something about it and I had hoped that the makers of Dietpi could tell me what is different in the Dietpi version compared to the standard version of Raspberry?!?

with the standard version of Raspberry-Light...
`pi@raspberrypi:~ $ sudo debmatic-info
debmatic version: 3.49.17-40
Kernel modules: Available
Raw UART dev:   Available
Rasp.Pi3 UART:  Assigned to GPIO pins
HMRF Hardware:  RPI-RF-MOD
HMIP Hardware:  RPI-RF-MOD
Board serial:   58A9A70C9A
Radio MAC:      unknown
SGTIN:          3014F711A0001F58A9A70C9A`

MichaIng, 2020-01-24:
@Ewies 
No difference between Raspbian and DietPi on that end, as long as you have the serial console disabled (default on Raspbian on RPi3/4). If it works OOTB on Noobs => Raspbian, then the only thing I can think of is a missing APT package that debmatic expects or some raspi-config binary/script it calls or such. I'll have a look into this.

Ewies, 2020-01-24:
I found the config that runs with the Dietpi and Debmatic.

For all who also have the problem here the Config in /DietPi/config.txt

`
dtoverlay=pivccu-raspberrypi
dtoverlay=pi3-miniuart-bt
enable_uart=1
force_turbo=1
core_freq=250
`
I like DietPi very much and now I can use it to its full extent.

Thank you

MichaIng, 2020-01-27:
@Ewies 
It seems that even that Bluetooth is disabled, when some service needs/wants to use the ttyAMA0 as serial console or ttyS0 for other purpose than serial console, `dtoverlay=pi3-miniuart-bt` is required to have those connected to/listening on the correct GPIO pins.

Could you paste:
```
ps t S1,AMA0
```
This shows which process is attached to which UART/serial device. The question is if it wants to use ttyAMA0 as serial console or ttyS0 for other purpose than serial console. In both cases your settings could be simplified.

#### Background (sorry much text, ignore if not interested :wink:):
- If the RPi model is WiFi/BT-capable, by default the full hardware UART (ttyAMA0) is used by Bluetooth. Hence mini software UART (ttyS0) is available for serial console and attached to the related GPIO pins.
- Furthermore on those devices ttyS0 is disabled by default, since it requires a fixed GPU frequency, setting `core_freq=250` automatically, which has an performance impact in return. `core_freq=250` is hence implied when enabling ttyS0 via `enable_uart=1` and does not need to be set explicitly in this combination. `force_turbo=1` allows to raise the GPU frequency by fixing it at the upper range, hence this can be used to enable ttyS0 without performance impact. Then `core_freq=250` counteracts this, hence doesn't make any sense in this combination. But forcing turbe mode affects SoC power consumption and hence temperature, hence is a dangerous option generally, when you don't have some cooling capability or a careful temperature monitoring.
- `dtoverlay=pi3-miniuart-bt` switches ttyS0 and ttyAMA0, hence ttyAMA0 is now the primary UART, available for serial console while ttyS0 is available for other purpose on other GPIO pins. Since ttyAMA0 does not rely on fixed GPU/core frequency, it is enabled by default then, hence `enable_uart=1` is then not required, as well as `force_turbo=1` and `core_freq=250`. Only thing I am not sure about, is how other serial devices, attached to ttyS0, react to non-fixed GPU/core frequencies, hence if then `force_turbo=1` and `core_freq=250` is still required in some cases, but not for the serial console on ttyAMA0, but for the other serial device on ttyS0.

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng