MichaIng/DietPi #3355: Error when updating, "Writing more data than expected"

patho5, 2020-01-23:
#### Details:
- Date           | Thu 23 Jan 09:54:01 CST 2020
- Bug report     | N/A
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. run command 'dietpi-update'

#### Expected behaviour:

- Normal update, I expect

#### Actual behaviour:

- updater encounters an issue with the Packages.gz file that is downloaded from packages.sury.org

#### Extra details:

#### Additional logs:

Ign:1 http://ftp.nl.debian.org/debian stretch InRelease
Hit:2 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian stretch InRelease
Hit:3 http://ftp.nl.debian.org/debian stretch Release
Get:4 https://packages.sury.org/php stretch InRelease [6,760 B]
Get:5 https://download.docker.com/linux/raspbian stretch InRelease [31.1 kB]
Hit:6 https://archive.raspberrypi.org/debian stretch InRelease
Get:8 https://packages.sury.org/php stretch/main armhf Packages [132 kB]
Err:8 https://packages.sury.org/php stretch/main armhf Packages
  Writing more data than expected (131791 > 131756)
  Hashes of expected file:
   - Filesize:131756 [weak]
   - SHA256:2ba1f75e44d6d7e189e54e247772be73e6b1f1580b3d331fda18569f9c23040f
   - SHA1:2cefcd807a9ca56bd4ace953cdfe29e45c4e18b4 [weak]
   - MD5Sum:3c91df0b59a417f03bd317c7bc624511 [weak]
  Release file created at: Wed, 22 Jan 2020 14:34:57 +0000
Fetched 37.8 kB in 2s (15.5 kB/s)
Reading package lists...
E: Failed to fetch https://packages.sury.org/php/dists/stretch/main/binary-armhf/Packages.gz  Writing more data than expected (131791 > 131756)
   Hashes of expected file:
    - Filesize:131756 [weak]
    - SHA256:2ba1f75e44d6d7e189e54e247772be73e6b1f1580b3d331fda18569f9c23040f
    - SHA1:2cefcd807a9ca56bd4ace953cdfe29e45c4e18b4 [weak]
    - MD5Sum:3c91df0b59a417f03bd317c7bc624511 [weak]
   Release file created at: Wed, 22 Jan 2020 14:34:57 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

MichaIng, 2020-01-23:
@patho5 
Many thanks for your report.

This is most likely a temporary issue with the sury.org PHP repository. Please try:
```
/DietPi/dietpi/func/dietpi-set_software apt-cache clean
apt update
```
See if the same error message appears:

> E: Some index files failed to download. They have been ignored, or old ones used instead.

patho5, 2020-01-23:
MichaIng, this corrected the issue and I was able to update to the latest version with no issues.
Thank you for your quick response!

MichaIng, 2020-01-23:
Great, I mark this issue as closed then. The solution is something that should be available from this error prompt directly: https://github.com/MichaIng/DietPi/projects/4

closed 2020-01-23 by MichaIng