MichaIng/DietPi #3357: Amiberry stops on black screen after config and launch of emulator

digitalMedic, 2020-01-25:
### Creating a bug report/issue

#### Required Information
- DietPi version:  6.28
- Distro version:  10.2
- Kernel version:  Linux DietPi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
- SBC device:  RaspberryPi 4B
- Power supply used:  Official RasberryPi 4B power supply 5V 
- SDcard used: Kingston 64GB

#### Additional Information (if applicable)
- Software title:  Amiberry
- Was the software title installed freshly or updated/migrated?  Freshly
- Can this issue be replicated on a fresh installation of DietPi?  Yes

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. Install DietPi using BalenaEtcher and latest DietPi image
2. Install Amiberry from DietPi software install screen
3. Launch Amiberry, config emulator and hit Start

#### Expected behaviour

- Amiberry should start the emulator on pressing Start

#### Actual behaviour

- After pressing Start, emulator hangs on black screen requiring reboot of RPi

#### Extra details

- Had to manually install the following dependencies to get Amiberry to work.
"sudo apt-get install libsdl2 libsdl2-ttf libsdl2-image libxml2 libflac libmpg123 libpng libmpeg2-4"

MichaIng, 2020-01-26:
@digitalMedic 
Many thanks for your report.

Actually Amiberry is installed with all its dependencies via dietpi-software. The SDL2 libs are placed in `/mnt/dietpi_userdata/amiberry/lib` and loaded by the systemd service. Hence you must start it via `systemctl start amiberry` or autostart option.

There is another bug that can make it hang on RPi with onboard sound enabled: https://github.com/MichaIng/DietPi/issues/3356
I already patched our images with the fix and added it as live patch to MOTD script.

An assumption why or works in your case when installing the SDL2 libs from APT repo is that it might access audio in a different way which works around the audio issue, not sure.

Could you test it with the linked fix and removing the SDL2 libs via apt purge?

digitalMedic, 2020-01-26:
Hi.  I did a fresh install and updated through the launcher.  I then entered the sed command from the other issue.  Amiberry starts the emulator now, but the mouse is way off.  I have to use the keyboard to select options in the config.

MichaIng, 2020-01-26:
@digitalMedic 
Many thanks for testing, great it works with the fix. The current images btw have them included already.

I'll have a look into the mouse feature. When I tested the images initially, the mouse worked. But at least the Amiberry binary has been updated one time since then.

digitalMedic, 2020-01-26:
No problem.  The issue with the mouse now is that it is off vertically (the mouse click is registered higher than the pointer) by a number of inches.  Thanks for fixing the other error though.  :)

MichaIng, 2020-01-26:
@digitalMedic 
Ah okay, I'll test it by times.

actraiser30, 2020-02-03:
Hello,i installed the latest dietpi amiberry image (6.28) in an Raspberry pi4 (1GB) after reboot it runs amiberry fine, if i press quit amiberry the screen goes black like its lost the signal i have to press a few buttons in my keyboard for the screen to get back on after half a minute or so to get me to the DietPi login screen,this is also happening with older dietpi/amiberry releases on the Rpi4 only
any suggested fix?

MichaIng, 2020-02-04:
@actraiser30 
Can you verify that systemd-logind is active?
```
systemctl status systemd-logind
```
This service should spawn a getty (login prompt) automatically, as fast as Amiberry exits. And it should be enabled automatically when Amiberry is installed.

actraiser30, 2020-02-27:
hello any change that we can get the latest version v3.1.2 upon installing amiberry ?

MichaIng, 2020-02-27:
@actraiser30 
I'll recompile our binaries for DietPi v6.29.

midwan, 2020-03-03:
@MichaIng 
you might want to update the Amiberry version while you're at it, I've fixed a few bugs since then and we're currently on v3.1.2 (until a new version is released of course). Perhaps offering a method of downloading the sources and compiling would be an alternative also (if someone wants to get the latest version from `master`)?

MichaIng, 2020-03-03:
@midwan 
Some binaries are on v3.1.1 already, however yes I'll run a rebuild session the next days and upload the build scripts I use, so users can re-compile themselves the same way we do. Although of course following your official instructions works just the same 😉.

actraiser30, 2020-03-14:
hello in the https://github.com/midwan/amiberry/releases has two version for each raspberry eg

amiberry-rpi3-sdl2-v3.1.2

amiberry-rpi3-v3.1.2

which one is installed with amiberry in dietpi ??

midwan, 2020-03-14:
I would assume the one that gets installed is the Dispmanx version (e.g. the one without `sdl2` in the filename), which provides the best performance for the RPI.

actraiser30, 2020-03-14:
but stil amiberry uses sdl2 since its inside amiberry folder

MichaIng, 2020-03-14:
@actraiser30 
The DispmanX version uses SDL2 as well. And yes that's what the binaries DietPi offers are compiled against, although we do not use the pre-compiled binaries that @midwan offers.

actraiser30, 2020-03-14:
strange i used the pre-compiled binaries with amiberry installed from dietpi with the DispmanX it did not boot with the sdl2 binary it boot

MichaIng, 2020-03-14:
@actraiser30 
Did you try the raw DietPi-Software install without any manipulation? E.g. we compiled the Buster RPi binaries after we found that the ones from midwan do not work with our SDL2 version. Do you should not try to mix those.
Also note that the systemd service we ship to start Amiberry contains `Environment=LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib` to use those SDL2 libs. You must either use this service or do `export LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib` before executing the binary manually.

actraiser30, 2020-03-14:
I did a fresh install of amiberry in the latest buster image of dietpi-Amiberry then i tried both the binaries from midwan by copying them with mc to the appropirate folder overwriting the included binary, in the DispmanX version it did not boot in the sdl2 version it boot

i just wanted to use the latest 3.1.2 version

MichaIng, 2020-03-14:
@actraiser30 
If you use the midwan binaries, try to remove `rm /mnt/dietpi_userdata/amiberry/lib/*` and instead install the SDL2 libraries from Raspbian: `apt install libsdl2 libsdl2-ttf libsdl2-image`
Otherwise I'll recompile our binaries the next days.

midwan, 2020-03-14:
The binaries I provide are all compiled under the Foundation's latest distro (currently Raspbian Buster), so they may not work on others. :)

actraiser30, 2020-03-14:
Sostos o mitsos

MichaIng, 2020-03-22:
@midwan 
How did you manage to reduce amiberry binary size to ~2 MiB? Mine are still ~5.4 MiB (like yours pre-v3.1.0) and `strip` has no significant effect.

midwan, 2020-03-22:
@MichaIng I used an executable compressor, `upx` - it decompresses the executable during runtime. ;)

MichaIng, 2020-03-22:
@midwan 
Ah okay, as this does not increase performance or reduce resource usage, I will skip it. But good to know 😉.

midwan, 2020-03-22:
You can skip it, it's just for making the archive smaller but it doesn't change the performance.

MichaIng, 2020-03-28:
I recompiled all our binaries by the way. Those are pulled already on fresh installs + reinstalls now, however I added a reinstall to v6.29 update to apply it to all systems: https://github.com/MichaIng/DietPi/commit/96ae941c9647b3b87cac91a44b8a4f6b3b98d50e

MichaIng, 2020-05-18:
Marking as closed. Feel free to reopen if required.

closed 2020-05-18 by MichaIng

sromeroi, 2021-04-02:
> Marking as closed. Feel free to reopen if required.

Sorry for replying to this closed issue... 

Midwan just released Amiberry 4.0 final, is it possible to update to 4.0? Or, are you planning to update the binaries? Thanks!

midwan, 2021-04-02:
As of now, v4.1 actually - found a bug in v4.0 so I deployed a quick update.

sromeroi, 2021-04-02:
> As of now, v4.1 actually - found a bug in v4.0 so I deployed a quick update.

It's even better then :) (thanks BTW for your work on Amiberry!).

MichaIng, 2021-04-02:
Awesome, I'll update our binaries ASAP. @midwan any breaking changes that would require changed config files, dir structure or such? I'll also have a look into the repo.

midwan, 2021-04-02:
There have been _many_ changes actually (488 commits in v4.0 since v3.3 and a few more for v4.1).
There are 2 new directories (included in the ZIP archives) that are optional: `abr` and `controllers`.

Config files might need to be recreated, mostly due to the now invalid old Height values (they used to be doubled, now we use the proper full size). E.g. if you had 256 lines for height, you'll now get half the expected height in your output window.

MichaIng, 2021-04-02:
Probably the height values could be removed from existing config files instead of re-creating them completely?

midwan, 2021-04-03:
One more thing I remembered: The 32-bit RTG pixel format has changed, to match that of WinUAE (easier to merge stuff in the future this way). This is configurable from the new RTG Panel in the GUI, and the default values are marked with an Asterisk next to them (16-bit modes are not changed). If you change them here (and you should), then your configured Screenmode in Workbench will also be invalid, if it was a 32-bit one. This means that you'll have to open Screenmode and re-select your desired mode, then save it again.

sromeroi, 2021-04-08:
> Awesome, I'll update our binaries ASAP. @midwan any breaking changes that would require changed config files, dir structure or such? I'll also have a look into the repo.

Just to warn you in case you're recompiling amiberry... it looks like 4.1.2 is out fixing some issues in 4.0.

midwan, 2021-04-08:
I'm getting some bug reports, so there will be a few more updates it seems. :)