MichaIng/DietPi #3358: While the update process, an error ocured and said I should post a ticket here.

LeatherMolly, 2020-01-25:
New Issue - 25.01.2020 - While the update process, an error ocured and said I should post a ticket here.
The blue lines to post:
#### Details:
- Date           | Sat 25 Jan 15:22:07 CET 2020
- Bug report     | 9e7b7f07-aa5c-4ea8-8c32-0be03c112593
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPizza 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
- Exit code      | 4
- Software title | DietPi-Set_software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-01-25 15:21:27--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 192.168.178.31
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|192.168.178.31|:443... failed: Connection refused.

-------------------------------
my output out of commandline:
-------------------------------
~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
curl: (7) Failed to connect to raw.githubusercontent.com port 443: Connection refused
[ INFO ] DietPi-Update | No response from: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6 ()
[ INFO ] DietPi-Update | Checking mirror: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
[  OK  ] DietPi-Update | Using update server: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.26.3
[ INFO ] DietPi-Update | Latest version  : v6.28.0
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=12211 MiB | required=100 MiB
[  OK  ] DietPi-Update | Setting in /DietPi/dietpi.txt adjusted: CONFIG_AUTO_DIETPI_UPDATES=1
[  OK  ] DietPi-Update | Setting in /DietPi/dietpi.txt adjusted: CONFIG_AUTO_DIETPI_UPDATES=0
[ INFO ] DietPi-Backup | Checking for required APT packages: rsync
[ INFO ] DietPi-Backup | Flagged for install: rsync
[ INFO ] DietPi-Backup | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [25.3 kB]
Get:3 https://archive.raspberrypi.org/debian buster/main armhf Packages [263 kB]
Update rpimonitor Packages Status
Fetched 303 kB in 1s (257 kB/s)
Reading package lists...
[  OK  ] DietPi-Backup | G_AGUP
[... 521 lines cut ...]
Removing 'diversion of /boot/fixup4x.dat to /usr/share/rpikernelhack/fixup4x.dat by rpikernelhack'
Removing 'diversion of /boot/LICENCE.broadcom to /usr/share/rpikernelhack/LICENCE.broadcom by rpikernelhack'
Setting up libraspberrypi0 (1.20200114-1) ...
Setting up libtiff5:armhf (4.1.0+git191117-2~deb10u1) ...
Setting up libraspberrypi-bin (1.20200114-1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
[  OK  ] DietPi-Update | G_AGUG

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Installing new DietPi code

######################################################################################################################################################################################################### 100.0%
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts to RAMdisk
[  OK  ] DietPi-Update | Installing new DietPi system files to disk
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt ()
[FAILED] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[ INFO ] DietPi-Set_software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[FAILED] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[ INFO ] DietPi-Set_software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Bugreport | Bug report sent, reference code: 9e7b7f07-aa5c-4ea8-8c32-0be03c112593

Joulinar, 2020-01-25:
Hi,
these 2 lines looks quite strange to me
```
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 192.168.178.31
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|192.168.178.31|:443... failed: Connection refused.
```

Looks like `raw.githubusercontent.com` is resoled as `192.168.178.31`. But 192.168.178.x is a privat network. I guess you home network. Usually used by Internet Router like FritzBox.

LeatherMolly, 2020-01-25:
Hi Joulinar,
I tried to update again and I saw this 2 Lines too. 

I am running DietPi with PiHole and did not really change much of the original settings, as I am more or less a beginner in this sphere.

Does someone has a hint how and where I could look or check to change this behaviour?

Help is apreciated. Thanx

Joulinar, 2020-01-25:
Ok yep that could happen, thx for update. I will close the issue.

closed 2020-01-25 by Joulinar