MichaIng/DietPi #3360: Camera support (imx214) support on RockPro64

philfleck, 2020-01-27:
### Creating a feature request
Camera modul support in dietpi (Pine64 provided camera modul). The say, it is working with the android version, but we could not get it to run with dietpi. However, if i am not wrong, there is already an implementation driver in the Kernel device tree.

#### Is your feature request related to a problem? Please describe:
The shipped Kernel in stable does not support the camera. We experience performance issues when using the Beta Version.

#### Describe the solution you'd like:
Merge the existing imx214 driver into the used Kernel version of Stable?

#### Describe alternatives you've considered:
- Manual / Driver modul install (somewhat convinient)?

(Github Imx214)[https://github.com/torvalds/linux/blob/master/drivers/media/i2c/imx214.c]
(Camera Module)[https://store.pine64.org/?product=rockpro64-13-megapixel-camera-module-m12-mount-lens-model]

MichaIng, 2020-01-27:
@philfleck 
Many thanks for your request.

To clarify:
> The shipped Kernel in stable does not support the camera. We experience performance issues when using the Beta Version.

With "stable" you mean the old Stretch-based image and with "beta" the new Buster-based image, provided [here](https://github.com/MichaIng/DietPi/issues/2979)?

The old image is based on legacy but official Pine64-provided Linux kernel 4.4 while the new is based on mainline Linux kernel 5.4, which hence includes additional/updated drivers on the one hand, but might be slightly less optimised for this particular board. But actually since RK3399 is very widely used, there should be no performance downsides of mainline kernel to legacy one...

Since we usually do not do kernel development of compiling, we'll not provide an updated camera driver for the legacy kernel version, especially since all future images will be based on new mainline kernel. So I suggest two options:
- Describe the performance issues you face with the new Linux 5.4 image. Probably it is something we can fix or enhance or report back to Armbian devs, who do the great effort to develop those mainline kernels for a large number of SoCs/SBCs.
- Upgrade your old stable (Stretch) image to new mainline kernel (after doing a `dietpi-backup`!):
```
apt install linux-image-current-rockchip64 linux-dtb-current-rockchip64 linux-u-boot-rockpro64-current linux-stretch-root-current-rockpro64
apt purge linux-image-rockchip64 linux-dtb-rockchip64 linux-u-boot-rockpro64-default linux-stretch-root-rockpro64
```
- The above names are derived from the old image +  Armbian APT list.
- To recheck which kernel packages are installed on your system: `dpkg -l | grep '^linux-'`
- This also allows to check if the performance issues are really due to the new kernel and not related to other things.

philfleck, 2020-01-27:
> @philfleck

> With "stable" you mean the old Stretch-based image and with "beta" the new Buster-based image, provided [here](https://github.com/MichaIng/DietPi/issues/2979)?
exactly.

Thank you for making it clear. We are going to perform further test and let you know.

philfleck, 2020-03-05:
Hi we could not make it to work. We switched to Manjaro Arm and die some kernel patching we got from the imx214 devs. We will try future versions of dietpi and let you know. 
- best, phil

MichaIng, 2020-03-05:
@philfleck 
I see the kernel driver with last commit is part of Linux since 5.5, while Armbian currently is at 5.4, however 5.5 is already in testing. I think hence this will be solved soon.

But btw did you try a source build of the module to replace or add the driver that is shipped with the Armbian kernel?

belveder79, 2022-02-04:
Co-working with @philfleck on this, the current status is the following (sorry for misusing this to document what we've tried and what is the current status with latest Dietpi on Bullseye):

- it seems that the problem is not with the camera drivers per se, but with the way the MIPI ports are accessed...
- the rk3399 chipset supposedly uses the rkisp1 module and then v4l2 and media-ctl to set up a topology as described [here](https://linuxtv.org/downloads/v4l-dvb-apis/admin-guide/rkisp1.html) 
- the module rkisp1 itself was actually built in staging drivers in 5.10.63-rockchip64, but is not built any more in the edge image (5.15.18-rockchip64); anyway, the rkisp1 driver went out of staging and is now considered stable with 5.11.x onwards , so it should be there one or the other way
- I rebuilt the 5.15.18-rockchip64 version with the module and it loads, but modprobing it does not do anything (no kernel message, no additional device nodes), so I don't even get to the point where I can try accessing the camera...
- I made sure that the DTBs include the information, but for some reason nothing happens...

I will follow up on this here if I have something new, because it might be of interest to others... if ANYONE has made the camera work on an RK3399 chipset with kernel 5, I would be very happy to talk about it...

MichaIng, 2022-02-04:
This is awesome. When you found possibly required changes to the kernel or device tree, this should be suggested to Armbian:
- https://forum.armbian.com/forum/4-development/
- https://github.com/armbian/build

Here is the kernel configuration, a quick search though the list did not result in a line about a rkisp1 module, so probably it needs to be added indeed. As module (not builtin) it should be pretty reasonable for ROCKPro64, given that it's about the camera module explicitly built for this SBC.

The development forum is also a great place to get further input.

I'm happy to implement a toggle to `dietpi-config` to have required kernel modules, in case device tree overlays loaded and userland utilities (`v4l-utils` package).

marv, 2022-03-18:
@belveder79 Hey! I have the imx214 working with my RockPro64 board. The imx214 kernel driver needs more work to properly support exposure times, gain, frame rates etc., but I can capture images using libcamera just fine. Feel free to contact me (maybe on IRC? I'm marvs@Libera or marv@pine64) and let me know how to help :-)

belveder79, 2022-03-18:
@marv going to get in contact with you over IRC... I have a working DTS and it would definitely be good to have some conversation to get the camera support improved...

kbingham, 2023-02-20:
If you have patches to improve the IMX214 please post them to linux-media mailinglist.

 http://vger.kernel.org/vger-lists.html#linux-media

That's the path to getting the fixes directly into the linux-kernel.

MichaIng, 2023-02-20:
Makes much sense to send any patches upstream directly.

belveder79, 2023-02-20:
will do so... seems that I repair broke my version :grinning: - need to get back to a working state first...