MichaIng/DietPi #3361: Error

MiRo-Law, 2020-01-28:
#### Details:
- Date           | Tue 28 Jan 10:24:32 GMT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.93-v7+ #1290 SMP Fri Jan 10 16:39:50 GMT 20                                                                                        20 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | Connection test: ssh.dietpi.com
- Exit code      | 4
- Software title | DietPi-Bugreport

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2020-01-28 10:24:29--  https://ssh.dietpi.com/
Resolving ssh.dietpi.com (ssh.dietpi.com)... failed: Temporary failure in name r                                                                                        esolution.
wget: unable to resolve host address ‘ssh.dietpi.com’
```

MichaIng, 2020-01-28:
@MiRo-Law 
Many thanks for your report.

Could you simply retry it once? If it fails the same way, please run and paste the output of:
```
ping -c 1 quad9.net
```

MichaIng, 2020-02-20:
I mark this as closed due to missing reply and as it most likely was a temporary issue. Feel free to reopen if required.

closed 2020-02-20 by MichaIng