MichaIng/DietPi #3362: Error during installation of Home Assistant

Enmaboya, 2020-01-30:
#### Details:
- Date           | Thu 30 Jan 10:48:58 GMT 2020
- Bug report     | 5aa2abde-51f7-4ea0-a2c2-5ffff5a2c047
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGI gcc libc6-dev make zlib1g-dev libbz2-dev libreadline-dev libssl-dev libsqlite3-dev libffi-dev  - Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: Unable to locate package gcc
E: Unable to locate package libc6-dev
E: Unable to locate package make
E: Unable to locate package zlib1g-dev
E: Unable to locate package libbz2-dev
E: Unable to locate package libreadline-dev
E: Unable to locate package libssl-dev
E: Unable to locate package libsqlite3-dev
E: Unable to locate package libffi-dev

MichaIng, 2020-01-31:
@Enmaboya 
Many thanks for your report.

Could you try it manually:
```
apt update
apt install gcc libc6-dev make zlib1g-dev libbz2-dev libreadline-dev libssl-dev libsqlite3-dev libffi-dev
```
Then retry the Home Assistant install.
For a strange reason the APT lists seems to have been lost during dietpi-software run.

MichaIng, 2020-02-20:
I mark this as closed due to missing reply and as a clean apt update most likely solves it. Feel free to reopen if required.

closed 2020-02-20 by MichaIng