MichaIng/DietPi #3363: mariadb error at start

vimalaguti, 2020-02-01:
#### Details:                                                                                                                                                      
- Date           | Sat  1 Feb 09:04:42 CET 2020                                                                                                                    
- Bug report     | 00b6b2bc-1a1d-4670-a733-fe21e52d4a03                                                                                                            
- DietPi version | v6.28.0 (MichaIng/master)                                                                                                                       
- Image creator  | DietPi Core Team                                                                                                                                
- Pre-image      | Raspbian Lite                                                                                                                                   
- SBC device     | RPi 3 Model B+ (armv7l) (ID=3)                                                                                                                  
- Kernel version | Linux raspi 4.19.93-v7+ #1290 SMP Fri Jan 10 16:39:50 GMT 2020 armv7l GNU/Linux                                                                 
- Distro         | buster (ID=5)                                                                                                                                   
- Command        | systemctl restart mariadb                                                                                                                       
- Exit code      | 1                                                                                                                                               
- Software title | DietPi-Software                                                                                                                                 
                                                                                                                                                                   
#### Steps to reproduce:                                                                                                                                           
                                                                                                                        
1. there is a dedicated partition mounted at /mnt/random-numbers/dietpi_userdata/                                                                                                                                                             
2. install nextcloud
                                                                                                                                                                   
#### Expected behaviour:                                                                                                                                           
                                                                                                                                       
- start mariadb service
                                                                                                                                                                   
#### Actual behaviour:                                                                                                                                                                                                                                                                                    
                                                                                                                                        
- system service returns error
                                                                                                                                                                   
#### Extra details:                                                                                                                                                
                                                                                             
- when dietpi_userdata folder is /mnt/dietpi_userdata/ the installation
and run works fine.
                                                                                                                                                                   
#### Additional logs:                                                                                                                                              
```                                                                                                                                                                
Job for mariadb.service failed because the control process exited with error code.                                                                                 
See "systemctl status mariadb.service" and "journalctl -xe" for details.                                                                                           
```                      

```
mysqld[7303]: 2020-02-01  9:20:55 0 [Note] /usr/sbin/mysqld (mysqld 10.3.17-MariaDB-0+deb10u1) starting as process 7303 ...
mysqld[7303]: 2020-02-01  9:20:55 0 [Warning] Can't create test file /var/lib/mysql/raspi.lower-test
mysqld[7303]: [90B blob data]
mysqld[7303]: 2020-02-01  9:20:55 0 [ERROR] Aborting
systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
```
                               
```
ls -la /var/lib/
total 112
drwxr-xr-x 28 root   root    4096 Feb  1 01:24 .
drwxr-xr-x 12 root   root    4096 Jan 31 01:00 ..
drwxr-xr-x  2 root   root    4096 Jan 31 01:20 alsa
drwxr-xr-x  3 root   root    4096 Feb  1 01:23 apt
drwxr-xr-x  2 root   root    4096 Jan 31 23:54 aptitude
drwx------  3 root   root    4096 Jan 31 00:41 bluetooth
drwxr-xr-x  2 root   root    4096 Jan 31 00:41 dbus
drwxr-xr-x  2 root   root    4096 Sep 26 02:24 dhcp
drw-rw----  7 dietpi dietpi  4096 Jan 31 00:28 dietpi
drwxr-xr-x  7 root   root    4096 Feb  1 01:24 dpkg
drwxr-xr-x  2 root   root    4096 Jan 31 01:14 fail2ban
drwxr-xr-x  2 root   root    4096 Jan 31 23:24 man-db
drwxr-xr-x  2 root   root    4096 Aug 27  2017 minidlna
drwxr-xr-x  2 root   root    4096 Sep  9 15:42 misc
lrwxrwxrwx  1 root   root      26 Feb  1 01:12 mysql -> /mnt/dietpi_userdata/mysql
drwxr-xr-x  5 statd  nogroup 4096 Feb  1 01:11 nfs
drwxr-xr-x  2 root   root    4096 Jan  7 23:32 pam
drwxr-xr-x  4 root   root    4096 Feb  1 01:17 php
drwx------  3 root   root    4096 Sep 26 02:06 polkit-1
drwx------  2 root   root    4096 Jan  7 23:18 private
drwxr-xr-x  2 root   root    4096 Jan 31 23:00 python
drwxr-x---  2 redis  redis   4096 Feb  1 01:24 redis
drwxr-xr-x  2 root   root    4096 Jan  7 23:35 resolvconf
drwxrwxr-x  3 dietpi dietpi  4096 Feb  1 00:59 rpimonitor
drwxr-xr-x  3 root   root    4096 Sep 26 02:11 sudo
drwxr-xr-x  8 root   root    4096 Feb  1 01:01 systemd
drwxr-xr-x  4 root   root    4096 Jan 31 01:05 transmission-daemon
drwxr-xr-x  3 root   root    4096 Feb  1 01:20 ucf
drwxr-xr-x  2 root   root    4096 Sep 26 02:11 usbutils
```

                                                                                                           
The problem is similar to #3336

MichaIng, 2020-02-01:
@vimalaguti 
Many thanks for your report.

Can you please paste:
```
ls -Al /mnt $(readlink -f /mnt/dietpi_userdata/mysql/)
```

vimalaguti, 2020-02-01:
thank you for the support!

```
ls -Al /mnt $(readlink -f /mnt/dietpi_userdata/mysql/)
/mnt:
total 16
drwx------ 3 dietpi dietpi 4096 Feb  1 18:00 97a92e81-3095-47d8-ac48-bbb84029d585
lrwxrwxrwx 1 root   root     57 Feb  1 18:01 dietpi_userdata -> /mnt/97a92e81-3095-47d8-ac48-bbb84029d585/dietpi_userdata
drwxr-xr-x 2 root   root   4096 Jan  7 23:35 ftp_client
drwxr-xr-x 2 root   root   4096 Jan  7 23:35 nfs_client
drwxr-xr-x 2 root   root   4096 Jan  7 23:35 samba

/mnt/97a92e81-3095-47d8-ac48-bbb84029d585/dietpi_userdata/mysql:
total 122920
-rw-rw---- 1 mysql  mysql     8192 Feb  1 17:57 aria_log.00000001
-rw-rw---- 1 mysql  mysql       52 Feb  1 17:57 aria_log_control
-rw-r--r-- 1 dietpi root         0 Feb  1 01:14 debian-10.3.flag
-rw-rw---- 1 dietpi mysql 12582912 Feb  1 01:14 ibdata1
-rw-rw---- 1 dietpi mysql 50331648 Feb  1 17:57 ib_logfile0
-rw-rw---- 1 dietpi mysql 50331648 Feb  1 01:14 ib_logfile1
-rw-rw---- 1 mysql  mysql 12582912 Feb  1 17:57 ibtmp1
drwx------ 2 dietpi mysql     4096 Feb  1 01:14 mysql
-rw-rw---- 1 mysql  mysql    24576 Feb  1 17:57 tc.log
```

vimalaguti, 2020-02-01:
```
$ sudo cat /var/log/mysql/error.log

2020-02-01 17:57:26 0 [Note] InnoDB: Using Linux native AIO
2020-02-01 17:57:26 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2020-02-01 17:57:26 0 [Note] InnoDB: Uses event mutexes
2020-02-01 17:57:26 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-02-01 17:57:26 0 [Note] InnoDB: Number of pools: 1
2020-02-01 17:57:26 0 [Note] InnoDB: Using generic crc32 instructions
2020-02-01 17:57:26 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2020-02-01 17:57:26 0 [Note] InnoDB: Completed initialization of buffer pool
2020-02-01 17:57:26 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-02-01 17:57:26 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-02-01 17:57:26 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2020-02-01 17:57:26 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-02-01 17:57:26 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2020-02-01 17:57:26 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-02-01 17:57:26 0 [Note] InnoDB: Waiting for purge to start
2020-02-01 17:57:26 0 [Note] InnoDB: 10.3.17 started; log sequence number 1603959; transaction id 9
2020-02-01 17:57:26 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
2020-02-01 17:57:26 0 [Note] InnoDB: Cannot open '/mnt/dietpi_userdata/mysql/ib_buffer_pool' for reading: No such file or directory
2020-02-01 17:57:26 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-02-01 17:57:26 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2020-02-01 17:57:26 0 [Note] Recovering after a crash using tc.log
2020-02-01 17:57:26 0 [Note] Starting crash recovery...
2020-02-01 17:57:26 0 [Note] Crash recovery finished.
2020-02-01 17:57:26 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2020-02-01 17:57:26 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-02-01 17:57:26 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist

```

MichaIng, 2020-02-01:
Please try the following:
```
chown -R mysql:mysql /mnt/97a92e81-3095-47d8-ac48-bbb84029d585/dietpi_userdata/mysql
systemctl restart mariadb
```

vimalaguti, 2020-02-01:
Nope, didn't fix. 
Anyway I erased the flashcard and reinstalled dietpi. You might want to add a few lines about what user (regular linux user) isn't required to do because it's already handled. Eg. I created a different home partition, expanded the root partition, explain data are going to be saved in /mnt, and so on. Probably I messed up with the init script, anyway now it works fine.

closed 2020-02-01 by vimalaguti