MichaIng/DietPi #3364: Could not install Chromium from dietpi-software

kmplngj, 2020-02-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

buster

- Kernel version | `uname -a`

Linux DietPi 4.19.93+ #1290 Fri Jan 10 16:34:37 GMT 2020 armv6l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)

RPi Zero W (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)

2.1A

- SDcard used | (EG: SanDisk ultra)

SanDisk 32GB Ultra

#### Additional Information (if applicable)
- Software title

Chromium

- Can this issue be replicated on a fresh installation of DietPi?

Is fresh installed

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

1c7a9f07-5ae2-4a7e-97b9-f1bbdea6ef07

#### Steps to reproduce
Installed DietPi, Setup Wifi in dietpi-wifi.txt, Set Autostart to Chromium

Could not install Chromium in dietpi-software

#### Expected behaviour
Chromium should be installable-

#### Actual behaviour

No, Chromium to install.

THANKS!

Joulinar, 2020-02-02:
Hi,

many thanks for your report. Did you already finished the initial setup process? If not, pls do so first.

Next to that I would recommend to set Autostart after you successfully installed Chromium and not before.

Do you have any further error message if you try to install Chromium using `dietpi-software`

MichaIng, 2020-02-02:
@kmplngj 
Many thanks for your report.

This is intended, Chromium is currently not supported on ARMv6 RPi models by us. But I am not sure about the reason and if there are still related problems currently. You can manually test it, if it works fine, we can enable it for next release officially:
Edit `/DietPi/dietpi/dietpi-software`
Search for `aSOFTWARE_NAME[$software_id]='Chromium'`
Remove the line `aSOFTWARE_AVAIL_G_HW_ARCH[$software_id,1]=0` in this box, safe and use `dietpi-software` to install Chromium, which is now available.
Here the code block, with the last line to be removed: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L379-L389

kmplngj, 2020-02-02:
Works for me and my Smart Mirror. 

Thanks

closed 2020-02-02 by kmplngj

MichaIng, 2020-02-02:
@kmplngj 
Great. How is the performance on RPi Zero?

kmplngj, 2020-02-03:
@MichaIng Works for me. But I only use it in Kiosk Mode with my Smart Mirror.

MichaIng, 2020-02-04:
@kmplngj 
Okay navigation performance does not play a role then. This use case should be enough to enable Chromium for ARMv6 as well. I found other GPU/graphical software titles as well disabled for ARMv6. Not sure if there were issues with the GPU (or i.e. OpenGL) drivers on those models in the past, but generally they work now.