MichaIng/DietPi #3365: ROCK Pi S image

lwqcz, 2020-02-04:
### Creating an image request

#### Formal device information
- Device name | ROCK Pi S
- Official product URL | https://wiki.radxa.com/RockpiS
- GitHub resource URL (if available) | https://github.com/radxa
- Image download URL | https://dl.radxa.com/rockpis/images/debian/rockpis_debian_buster_minimal_arm64_191219_0406-gpt.img.gz

#### Is the SBC officially supported by the Debian installer?
- As I noticed on https://wiki.debian.org/InstallingDebianOn/ there is no direct Debian support.

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Yes, it seems to be that SBC vendor Radxa is providing official [Debian Buster image](https://wiki.radxa.com/RockpiS/downloads).

#### If not, are there install instructions for Debian available?

- N/A

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/

I've got nothing ... https://feathub.com/MichaIng/DietPi/+95

#### My DietPi image & notes
- I've used procedure described at https://github.com/MichaIng/DietPi/issues/1285 and later today I can post a link to compressed *.img file of Radxa official Buster modified by DietPi's procedure :-) Would you be interested in?
- I can also test this image as I have ROCK Pi S on my desk.
- Is there an option to have first-run unattended? ROCK Pi S has no video output AFAIK and therefore I would need to use serial console ...

MichaIng, 2020-02-04:
@lwqcz 
Many thanks for your request.

Which kernel version does the Radxa image ship? Linux 4.4 like the Armbian ones? https://dl.armbian.com/rockpi-s/

Fully unattended firstrun setup is currently not possible. The update and installs on firstrun can be automated via [dietpi.txt](https://github.com/MichaIng/DietPi/blob/master/dietpi.txt#L91), but you need to do one manual login. However this can be done via SSH (enabled by default), as long as you configured `dietpi.txt`/`dietpi-wifi.txt` with correct network info/credentials. Serial console is an alternative as well of course, and DietPi-PREP should enable it as well for first boot automatically, if the serial/UART device is available/enabled on the board at time of running the script + the serial device needs to be ttyS0/ttyAMA0/ttySAC0 one of this naming schemes to be recognised.

lwqcz, 2020-02-07:
I've tried to use `dietpi.txt` file to configure first boot, but the process stuck on some kernel/HW related error.
Radxa is using kernel 4.14-rockpi based on https://github.com/radxa/rockchip-bsp AFAIK.

I need to finish a review regarding this HW so I'm going to skip the DietPi test on this HW for now.

Any timeframe, when ROCKPis would be supported? Can I help somehow - maybe to try some alpha builds?

BTW: Fully unattended boot including first-boot is going to be supported anytime soon?

MichaIng, 2020-02-07:
@lwqcz 
Okay I downloaded the Radxa image and had a look inside. The kernel is 4.4 like the one from Armbian, no reason to use the Armbian image for this device. Also Armbian does not (yet?) officially support it.

I'll create an image from this and upload for testing.

lwqcz, 2020-02-07:
@MichaIng 
yeah, you are correct - it's `4.4.143-34-rockchip-g3c9d2019dba7`, sorry for wrong info.

Wifi and Bluetooth is working fine so far. I'm going to test 1wire and I2C.

Thank you for new image anyway. No need for my prepared image? I can upload it...

MichaIng, 2020-02-08:
Thanks for the offer. I'll this build it myself since I did some closer review in case and update DietPi-PREP of required. Also I'll e.g the new hardware definition directly, so banner shows device name, our scripts know correct CPU etc.

MichaIng, 2020-02-24:
@lwqcz 
Image ready for testing: https://dietpi.com/downloads/testing/DietPi_ROCKPiS-ARMv8-Buster.7z
Based on official Radxa Debian Buster as suggested. Let's see if this works fine so far.

lwqcz, 2020-03-02:
@MichaIng I've didn't noticed your last comment here, sorry. I'll test it tonight and I'll let you know.

lwqcz, 2020-03-05:
@MichaIng 
I've tested that image for few days and played with `dietpi-config` + `dietpi-software` + WIFI and found no issues at all. But bluetooth seems not to be working. And I've not tried I2C and 1wire as there are no overlays ready, or am I wrong?

MichaIng, 2020-03-05:
Many thanks for testing and feedback.

Probably there is some Bluetooth package available in the Radxa repo. How did you test it, currently not sure if it's enabled for ROCK Pi S in dietpi-config, probably it requires to listen on a different tty.

Similar thing on I2C and 1wire. If there is no overlay, probably it can be installed or enabled via kernel module. Basically the Radxa forum/docs might also help on this.

MichaIng, 2020-03-05:
For Bluetooth according to docs:
```
apt install rtl8723ds-firmware bluez
systemctl start bluetooth
```

lwqcz, 2020-03-06:
Kernel tries to setup bluetooth during boot:
```
[    1.420929] Bluetooth: Core ver 2.21
[    1.420998] Bluetooth: HCI device and connection manager initialized
[    1.421022] Bluetooth: HCI socket layer initialized
[    1.421039] Bluetooth: L2CAP socket layer initialized
[    1.421081] Bluetooth: SCO socket layer initialized
[    2.886560] Bluetooth: Virtual HCI driver ver 1.5
[    2.992086] of_get_named_gpiod_flags: parsed 'gpios' property of node '/gpio-leds/blue-led[0]' - status (0)
[    3.065633] Bluetooth: RFCOMM TTY layer initialized
[    3.065663] Bluetooth: RFCOMM socket layer initialized
[    3.065716] Bluetooth: RFCOMM ver 1.11
[    3.065746] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    3.065764] Bluetooth: HIDP socket layer initialized
[    3.067187] of_get_named_gpiod_flags: parsed 'uart_rts_gpios' property of node '/wireless-bluetooth[0]' - status (0)
[    3.067206] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: uart_rts_gpios = 135.
[    3.067239] of_get_named_gpiod_flags: parsed 'BT,power_gpio' property of node '/wireless-bluetooth[0]' - status (0)
[    3.067249] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,power_gpio = 139.
[    3.067262] of_get_named_gpiod_flags: can't parse 'BT,reset_gpio' property of node '/wireless-bluetooth[0]'
[    3.067273] of_get_named_gpiod_flags: can't parse 'BT,wake_gpio' property of node '/wireless-bluetooth[0]'
[    3.067290] of_get_named_gpiod_flags: parsed 'BT,wake_host_irq' property of node '/wireless-bluetooth[0]' - status (0)
[    3.067300] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_host_irq = 140.
[    3.067315] [BT_RFKILL]: bluetooth_platdata_parse_dt: clk_get failed!!!.
```

MichaIng, 2020-03-06:
@lwqcz 
Perfect with Bluetooth.

Regarding 1wire:
- Note that device tree overlays are just one way to enable a certain board feature.
- Kernel modules should be another way, if those are present instead:
```
modprobe wire
modprobe w1-gpio
modprobe w1-therm
```
- In the link I posted the guys indeed created dtoverlays manually. However I couldn't find some official instructions about that, which identifiers/addresses are required an all that 🤔.

lwqcz, 2020-03-06:
@MichaIng 
Maybe I am wrong, but kernel module is not enough configuration - the HW needs to be setup somehow, for instance - GPIO PIN needs to be specified and so on:
[`Overlays are only available with Mainline ...`](https://forum.armbian.com/topic/8719-rock-pi-4/page/3/?tab=comments#comment-87226) ... and more about this topic [here](https://forum.radxa.com/t/temperature-from-ds18b20/1997/4).

MichaIng, 2020-03-07:
@lwqcz 
You can add module options via modprobe (`/etc/modprobe.d/*.conf`) as well. But it depends on kernel + modules code.

See some old PR here with the intention to add options to RPi 1-wire module: https://github.com/raspberrypi/linux/pull/457/files
This has not been added, instead the device tree overlay allows it now, which generally is the modern approach of course.

Probably we can open some request at Radxa to add an official ROCK Pi S dtoverlay for 1-wire with pin options, else I guess you'd need to bump the topic on Radxa forum... btw. isn't this one working already, as it looks quite complete: https://forum.radxa.com/t/temperature-from-ds18b20/1997/33

MichaIng, 2020-03-07:
@lwqcz 
I added ROCK Pi S Bluetooth support to DietPi-Config: https://github.com/MichaIng/DietPi/commit/7e1feb7b1bc6713bf707616cf2f50051543fc9db

lwqcz, 2020-03-08:
RockPi and RockPiS uses different SoCs - RK3399 vs 3308 - so I assume that the approach to overlays and 1w can be rather different or am I wrong?

MichaIng, 2020-03-08:
@lwqcz 
Yes that's true, so playing around with the .dtso is only something when you can quickly plug and edit the SDcard + ext4 file system somewhere else, if it doesn't boot 😄. However a request to Radxa for adding the overlay to their APT package would be the most sustainable thing.

Btw, I just checked the Armbian packages: https://apt.armbian.com/pool/main/l/linux-4.4.207-rockpis/
I cannot find any hint or overlay for 1wire there as well 🤔.

lwqcz, 2020-03-09:
@MichaIng 
:ok: I'm going to ask Radxa to provide some clue about how to solve this at least. These buses didn't work even in their own Linux image.

MichaIng, 2020-05-24:
New image uploaded based on current DietPi v6.30 with some other minor fixes: https://dietpi.com/downloads/images/testing/DietPi_ROCKPiS-ARMv8-Buster.7z

@lwqcz 
We btw have overseen the second source for device tree overlays, respectively the better maintained source. The kernel package itself comes with all that is needed and more:
```
root@DietPi:~# ls -Al /boot/dtbs/4.4.143-47-rockchip-g495ff574976c/rockchip/overlay/
total 56
-rwxr-xr-x 1 root root 4373 Apr 21 02:39 README.rockchip-overlays
-rwxr-xr-x 1 root root 1290 Apr 21 02:39 i2c1-hym8563.dtbo
-rwxr-xr-x 1 root root  546 Apr 21 02:39 rk3308-console-on-uart0.dtbo
-rwxr-xr-x 1 root root  788 Apr 21 02:39 rk3308-console-on-uart1.dtbo
-rwxr-xr-x 1 root root  669 Apr 21 02:39 rk3308-console-on-uart2.dtbo
-rwxr-xr-x 1 root root  505 Apr 21 02:39 rk3308-i2c0.dtbo
-rwxr-xr-x 1 root root  262 Apr 21 02:39 rk3308-i2c1.dtbo
-rwxr-xr-x 1 root root  383 Apr 21 02:39 rk3308-i2c2.dtbo
-rwxr-xr-x 1 root root  504 Apr 21 02:39 rk3308-i2c3.dtbo
-rwxr-xr-x 1 root root  262 Apr 21 02:39 rk3308-pwm1.dtbo
-rwxr-xr-x 1 root root  383 Apr 21 02:39 rk3308-pwm2.dtbo
-rwxr-xr-x 1 root root  383 Apr 21 02:39 rk3308-pwm3.dtbo
-rwxr-xr-x 1 root root  881 Apr 21 02:39 rk3308-spi-spidev.dtbo
-rwxr-xr-x 1 root root  382 Apr 21 02:39 rk3308-uart0.dtbo
-rwxr-xr-x 1 root root  624 Apr 21 02:39 rk3308-uart1.dtbo
-rwxr-xr-x 1 root root  384 Apr 21 02:39 rk3308-uart2.dtbo
-rwxr-xr-x 1 root root  491 Apr 21 02:39 rk3308-w1-gpio.dtbo
-rwxr-xr-x 1 root root 1814 Apr 21 02:39 rockchip-fixup.scr
-rwxr-xr-x 1 root root 2480 Apr 21 02:39 rockpis-v11-spi2-waveshare35b-v2.dtbo
-rwxr-xr-x 1 root root 2512 Apr 21 02:39 rockpis-v11-spi2-waveshare35c.dtbo
-rwxr-xr-x 1 root root 2480 Apr 21 02:39 rockpis-v12-spi2-waveshare35b-v2.dtbo
-rwxr-xr-x 1 root root 2512 Apr 21 02:39 rockpis-v12-spi2-waveshare35c.dtbo
```
And there is the 1wire overlay. To enable it (as well mentioned in the readme) add `rk3308-w1-gpio` to the overlays line in `/boot/uEnv.txt`.

The `rockpis-dtbo` package ships the overlays in `/boot/overlays` which basically match the ones above but are fewer and older. And those would need to be added/enabled via `/boot/hw_intfc.conf`. So we have two implementations that likely conflict with each other, hence I removed the `rockpis-dtbo` package from the new image, which is also true for the current official Radxa image.

closed 2020-08-31 by MichaIng

lwqcz, 2020-11-24:
Unfortunately, 1wire still does not work. On DietPi or even on armbian-focal. I want to use different that the default PIN (to be PIN compatible with RPi) and uEnv or armbianEnv w1 PIN configs have no effect. Every time PIN number 17 is seen open for w1 module... And sorry for the delay 🤦

MichaIng, 2020-11-24:
Is the device path available? `ls /sys/bus/w1/`
And are the kernel modules enabled: `lsmod | grep w1`
In case enable them manually: `modprobe w1-gpio; modprobe w1-therm`

Generally, since we use the official Radxa bootloader, kernel (+overlays) and repo, I think we'll have more success checking things there. Found a similar thread, although it seems the device tree overlay has not been enabled there: https://forum.radxa.com/t/enable-1wire-for-ds18b20/3521?u=michaing

lwqcz, 2020-11-29:
I have spent some time trying and testing. Everything is working using default GPIO PIN 17 called `GPIO0_C1`. I have to luck getting other PINs to work, if I configured it that way, I was getting only "noise" pretending that there is some sensor, but nothing really worked.

MichaIng, 2020-11-29:
Little hint about the recent fixes and repacked image: #3925
This includes a new Linux meta package as Radxa renamed it and the new one points to a little newer kernel. If you want to test:
```
dietpi-backup 1
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /boot/dietpi.txt
dietpi-update 1
```
Probably it contains a fix for the noisy GPIO pin as well.

lwqcz, 2020-12-15:
I've tested the stable version for more than a week and it seems to be working 👍 

Anyway - can you add `1wire` enable/disable the option to `dietpi-config` or it would be better when I'll make an MR?
It is rather simple:
1. add to `/boot/uEnv.txt` - `rk3308-w1-gpio` option:
```
overlays=rk3308-uart0 rk3308-w1-gpio
```
2. add to `/etc/modules` file `1wire` modules:
```
w1-gpio
w1-therm
```

MichaIng, 2020-12-15:
Many thanks for testing 👍. Jep, let's make a new issue for this. Along with 1wire, we can switches for other hardware features that are available via overlays.

Btw, adding the modules to the list is definitely required? As I know from other cases that those might be loaded automatically, when the overlay is enabled.

I'd add those via `/etc/modules-load.d/dietpi-1wire.conf` or similar, to make it more transparent and easier to revert.