MichaIng/DietPi #3369: why ping 1.1.1.1?

lhaclove, 2020-02-06:
#### Details:
- Date           | Thu  6 Feb 12:25:44 CST 2020
- Bug report     | d57020f8-e1a3-487c-987e-0a5faef3522a
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | N/A
- Pre-image      | N/A
- SBC device     | NanoPi M1 Plus (armv7l) (ID=66)
- Kernel version | Linux DietPi 4.14.0 #83 SMP Fri Dec 8 17:44:19 CST 2017 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | ping -c 1 -W 10 1.1.1.1
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2020-02-06:
@lhaclove 
many thank for your report.

Hi, if you are not able to `ping 1.1.1.1` you would have serous issues as it's part of Cloudflares DNS Server. It is the globally available IP address for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `google.com`

MichaIng, 2020-02-07:
We really need to change the default test IP+domain for new images 🤔. It is a pain that there seem to be so much networks/routers that still override 1.* IPs internally.

@lhaclove 
May I ask: Are you on a home network with single router to www? Which router model do you use?

MichaIng, 2020-02-07:
Okay I changed our defaults: https://github.com/MichaIng/DietPi/commit/ffd9431cba7a9f5fda48faec5edd7bc9a12e82ed
Note that this is no solution. As long as the local network/router overrides/redirects 1.* IPs internally, related public servers are not reachable from the system, hence this should be fixed anyway.

I planned to add some general system check script, based on regularly reported issues not related to DietPi. Added ping 1.1.1.1 to the list.

closed 2020-02-07 by MichaIng

lhaclove, 2020-02-09:
you konw that, I'm Chinese and living in china.

MichaIng, 2020-02-11:
@lhaclove 
Are there public `1.*` IPs generally blocked in China? From what I find this seems to be not the case. Or is your ISP probably blocking international-public DNS IPs/services in general, to force you using the ISPs one? But I can't find any indication for that. Actually reading through some posts it is really not a country-specific issue but an issue with local/company networks, AP/hotspot providers and such: https://news.ycombinator.com/item?id=14862748

lhaclove, 2020-03-20:
Yeah, that's because of my ISP. So I changed to an accessible IP and it works fine now.

MichaIng, 2020-03-20:
@lhaclove 
Good to know, so it's indeed a two-layer issue. However solved for all future images.