MichaIng/DietPi #3370: Request to update Shairport-sync to most recent version

Hammer8, 2020-02-06:
### Creating a software request

Hi, I’m running 6.28 Dietpi on a pi 4 and am having some issues with Shairport and a usb DAC. Are there plans to update the Dietpi install to the most recent version? I tried Dietpi-software reinstall 124, which worked for updating Signalyst NAA and other items, but it did not update Shairport-sync. Thank you!

#### Formal software information
- Software title | 
- Short description | 
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2020-02-06:
Hi,
As far a I can see, Shairport-sync download URL is hard coded into `dietpi-software` script.

https://github.com/MichaIng/DietPi/blob/6ff308bdf88b6e3411bd61b4e3d16edded82fe38/dietpi/dietpi-software#L4755

@MichaIng 
I guess new software package would need to be created? Latest version on Shairport-sync GitHub seems to be 3.3.5

MichaIng, 2020-02-07:
@Hammer8 
Many thanks for your request. Yes indeed we ship a custom compiled version currently since it is newer than what Debian Stretch repo ships. Buster and Stretch-backports now ship the same version: https://packages.debian.org/buster/shairport-sync
Bullseye and even Buster-backports the most current: https://packages.debian.org/buster-backports/shairport-sync

Are there any important changes with the current version? I am thinking if we simply revert to Debian APT install on Buster and Bullseye systems, on non-RPi systems to backports, and only ship out custom compiled package for RPi-Stretch systems, where no backports are available, hence otherwise much older 2.8.6 would be shipped: https://packages.debian.org/stretch/shairport-sync

But if there are real benefits of the new version, I can update our custom compiled binaries.

Hammer8, 2020-02-07:
Hi, yes, looking at the release notes, there are a lot of bug fixes and enhancements added since the Dietpi version in the most recent version; especially from v3.2xx to v3.3.  A few of which I am especially interested in given I am not able to get the Dietpi version working with my USB DAC.  Additionally, there appears to be more troubleshooting features in the new version when things don’t work as expected.

Thank you for your help.

MichaIng, 2020-02-07:
@Hammer8 
Okay, since I have a bid limited time currently and am busy with redoing all our images, it can take some days. If I'm in mood and things go easy, I'll compile one RPi Buster binary over the weekend to test.

Joulinar, 2020-02-07:
@MichaIng 
are the following steps needed only to compile the stuff? if yes I could do it on my RPi3B+

```
$ git clone https://github.com/mikebrady/shairport-sync.git
$ cd shairport-sync
$ autoreconf -fi
$ ./configure --sysconfdir=/etc --with-alsa --with-soxr --with-avahi --with-ssl=openssl --with-systemd
$ make
```

LeeMenHin, 2020-02-07:
Here is an New AirPlay Mirroring Server Found in github ， support airplay2;
https://www.raspberrypi.org/forums/viewtopic.php?t=239964
GITHUB:
https://github.com/FD-/RPiPlay

MichaIng, 2020-02-07:
For reference, the thread with USB DAC issue: https://dietpi.com/phpbb/viewtopic.php?f=9&t=7242
Not sure whether new Shairport-Sync can solve it, but at least worth to give it a try.

@Joulinar 
Looks good, but I didn't have a closer look into compile options yet. Of course the mentioned build tools and dev packages are required first: `build-essential git xmltoman autoconf automake libtool libpopt-dev libconfig-dev libasound2-dev avahi-daemon libavahi-client-dev libssl-dev libsoxr-dev`

Joulinar, 2020-02-07:
@MichaIng 
Of course they are needed for building the package. I just skipped it in my post :laughing:

Joulinar, 2020-02-07:
ok I tried to setup the package on my RPi3B+ but I discover some differences between the package you created and the one I did now. 

`/usr/local/etc` is empty. Files in my installation are located on `/etc`
file in `/usr/local/share/man/man7` is `shairport-sync.7` and not `shairport-sync.7.gz`

Anyway attached the file. Just remove *.txt

[shairport-sync_3.3.5_armv7l.7z.txt](https://github.com/MichaIng/DietPi/files/4172751/shairport-sync_3.3.5_armv7l.7z.txt)

MichaIng, 2020-04-06:
@Joulinar 
Totally forgot about this issue. I already compiled new packages based on v3.3.6: https://dietpi.com/downloads/binaries/buster/

Strange that your config file is located in /etc since by default the binary should look inside /usr/local/etc which as well matches all other files locations. However possibly the binary searches the other default config file locations as well so that it does not make a difference. However, ah I see that I created this default config manually, although only because I didn't want a sample config which contains non-supported settings (based compile-time options).

Here my script:
<details>

```
{
. /boot/dietpi/func/dietpi-globals

# Download
G_RUN_CMD cd /tmp
G_RUN_CMD curl -sSfL https://github.com/mikebrady/shairport-sync/archive/master.tar.gz -O
G_RUN_CMD rm -Rf shairport-sync-master
G_RUN_CMD tar xf master.tar.gz
G_RUN_CMD rm master.tar.gz

# Compile
G_RUN_CMD cd shairport-sync-master
G_AGUP
G_AGI automake pkg-config make g++ libpopt-dev libconfig-dev libssl-dev libsoxr-dev libavahi-client-dev libasound2-dev
autoreconf -i -f || exit 1
./configure --with-alsa --with-avahi --with-ssl=openssl --with-soxr --with-metadata --with-systemd || exit 1
make CFLAGS='-g0 -O3' CXXFLAGS='-g0 -O3' || exit 1
# - Stip down binary size
strip --remove-section=.comment --remove-section=.note shairport-sync || exit 1

# Package dir: In case of Raspbian, force ARMv6
G_RUN_CMD cd /tmp
grep -q 'raspbian' /etc/os-release && DIR='shairport-sync_armv6l' || DIR="shairport-sync_$(uname -m)"
G_RUN_CMD rm -Rf $DIR
# - Control files, systemd service, config file, executable, copyright
G_RUN_CMD mkdir -p $DIR/{DEBIAN,lib/systemd/system,usr/local/{bin,etc,share/doc/shairport-sync}}

# Binary
G_RUN_CMD cp shairport-sync-master/shairport-sync $DIR/usr/local/bin/

# Copyright
curl -sSfL https://raw.githubusercontent.com/mikebrady/shairport-sync/master/LICENSES > $DIR/usr/local/share/doc/shairport-sync/copyright || exit 1

# systemd service
G_RUN_CMD cp shairport-sync-master/scripts/shairport-sync.service $DIR/lib/systemd/system/

# Cleanup
G_RUN_CMD rm -Rf shairport-sync-master

# Config file
cat << _EOF_ > $DIR/usr/local/etc/shairport-sync.conf
// Sample Configuration File for Shairport Sync
// Commented out settings are generally the defaults, except where noted.
// Some sections are operative only if Shairport Sync has been built with the right configuration flags.
// See the individual sections for details.

// General Settings
general =
{
	name = "%H"; // This means "Hostname" -- see below. This is the name the service will advertise to iTunes.
//		The default is "Hostname" -- i.e. the machine's hostname with the first letter capitalised (ASCII only.)
//		You can use the following substitutions:
//				%h for the hostname,
//				%H for the Hostname (i.e. with first letter capitalised (ASCII only)),
//				%v for the version number, e.g. 3.0 and
//				%V for the full version string, e.g. 3.3-OpenSSL-Avahi-ALSA-soxr-metadata-sysconfdir:/etc
//		Overall length can not exceed 50 characters. Example: "Shairport Sync %v on %H".
//	password = "secret"; // leave this commented out if you don't want to require a password
	interpolation = "soxr"; // aka "stuffing". Default is "auto". Alternatives are "basic" or "soxr". Choose "soxr" only if you have a reasonably fast processor and Shairport Sync has been built with "soxr" support.
//	output_backend = "alsa"; // Run "shairport-sync -h" to get a list of all output_backends, e.g. "alsa", "pipe", "stdout". The default is the first one.
//	mdns_backend = "avahi"; // Run "shairport-sync -h" to get a list of all mdns_backends. The default is the first one.
[... 182 lines cut ...]
Homepage: https://github.com/mikebrady/shairport-sync
Vcs-Git: git://github.com/mikebrady/shairport-sync.git
Vcs-Browser: https://github.com/mikebrady/shairport-sync
Description: AirPlay audio player
 Plays audio streamed from iTunes, iOS devices and third-party AirPlay
 sources such as ForkedDaapd and others. Audio played by a Shairport
 Sync-powered device stays synchronised with the source and hence with
 similar devices playing the same source. In this way, synchronised
 multi-room audio is possible without difficulty.
 .
 Shairport Sync does not support AirPlay video or photo streaming.
_EOF_

# Build deb package
G_RUN_CMD rm -Rf $DIR.deb
dpkg-deb --build $DIR || exit 1
G_RUN_CMD rm -Rf $DIR

# Upload
curl -T $DIR.deb sftp://root:t1tp4cda@ssh.dietpi.com/var/www/downloads/binaries/$G_DISTRO_NAME/

exit
}
```
</details>

MichaIng, 2020-04-07:
@Joulinar 
Do you have devices to the our new build? Since we have a report that it does not work: #3461
Could be due to strict compile-time options (`-g0 -O3`) instead of (Debian) default (`-g -O2`).

Joulinar, 2020-04-07:
hmm I don't have much experience on shairport-sync. However I flashed a new image to my RPi3B+ and did the update to v6.29 beta. Installed  shairport-sync and looks like it's working fine

```
root@DietPi3:~# systemctl status shairport-sync
● shairport-sync.service - Shairport Sync - AirPlay Audio Receiver
   Loaded: loaded (/lib/systemd/system/shairport-sync.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-07 12:26:57 CEST; 2min 25s ago
 Main PID: 520 (shairport-sync)
    Tasks: 5 (limit: 2319)
   Memory: 1.7M
   CGroup: /system.slice/shairport-sync.service
           └─520 /usr/local/bin/shairport-sync

Apr 07 12:26:57 DietPi3 systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
root@DietPi3:~# journalctl -u shairport-sync
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Tue 2020-04-07 12:28:27 CEST. --
Apr 07 12:26:57 DietPi3 systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
root@DietPi3:~#
```

MichaIng, 2020-04-07:
@Joulinar 
Okay, same with me. The service start (running binary, listening to TCP port etc) is not the issue, but playing audio only does not work. However interestingly the ARMv6 binary does work (on RPI3+, armv7l) while ARMv7 does not and if I got it right Raspbian repo as well not. So probably Debian vs Raspbian library incompatibility (ARMv7 binaries are compiled on Debian, ARMv6 binaries on Raspbian) but I think we need to wait for some more feedback to be sure.

Problem from my side as well that I have no AirPlay-like devices to really test Shairport-Sync.

C0D3-M4513R, 2020-04-07:
On ubuntu you can stream to airplay commands, if you install 
`paprefs pulseaudio-module-raop`, then run `paprefs` and check `Make discoverable Apple AirTunes sound devices available locally`. See [https://superuser.com/a/345976](https://superuser.com/a/345976)

closed 2020-05-03 by MichaIng

mistakenideas, 2020-05-04:
Should shairport-sync be automatically upgraded as part of the move to DietPi 6.29? I’ve just upgraded a RaspberryPi 3 B+ from 6.28 and although this process looks to have completed fine, shairport-sync is still at version 3.2.2-OpenSSL-Avahi-ALSA-soxr-metadata-sysconfdir:/usr/local/etc

MichaIng, 2020-05-04:
@mistakenideas 
Yes indeed we do not force this update.

mistakenideas, 2020-05-04:
@MichaIng 
Great. Thanks for confirming. The new version is working fine so far too.