MichaIng/DietPi #3371: Dropbear version outdated

WolfganP, 2020-02-06:
I was checking latest versions of some packages installed in my dietpi and found out that dropbear is outdated (almost 2 years old):

```
$ dropbear -V
Dropbear v2018.76
```

Latest version is **v2019.78** (as per https://github.com/mkj/dropbear/blob/master/CHANGES)

#### Required Information

```
$ cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

$ cat /etc/debian_version
10.2

$ uname -a
Linux DietPi 4.19.93-v7+ #1290 SMP Fri Jan 10 16:39:50 GMT 2020 armv7l GNU/Linux
```

Joulinar, 2020-02-06:
Hi,

I was playing a bit with dropbear this evening and dropbear is manged as debian package by running `apt-get`. I removed dropbear and reinstalled it on a VM. There I could see the following package was installed `dropbear-run`. If you try to update this package, you will get the following.

```
root@DietPiVM1:~# apt-get upgrade dropbear-run
Reading package lists... Done
Building dependency tree
Reading state information... Done
dropbear-run is already the newest version (2018.76-5).
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPiVM1:~#
```
Means there is no newer version within the debian source than currently installed. 

However you could add some more sources to your system.

https://packages.debian.org/en/sid/all/dropbear-run/download

```
nano /etc/apt/sources.list
deb http://ftp.de.debian.org/debian sid main
```

than run
```
apt-get update
apt-get upgrade dropbear-run
```

but be careful, for me this was trying to update quite a huge number of packages.

@MichaIng 
I guess there is a reason for the `old` dropbear package 😃

MichaIng, 2020-02-07:
@WolfganP @Joulinar 
> However you could add some more sources to your system.

Please don't do that guys. It would work on the ARMv7 RPi models, but e.g. break ARMv6 models (not Debian-repo compatible). Also we did such to install WireGuard once (now switched to Bullseye with strong pin priorities), and even with strict /etc/apt/preferences.d/ pin priorities, somehow some users had a half-Sid upgraded system, even the base packages, of course causing havoc, incompatible library versions and all such, especially on RPi where Raspbian (ARMv6) and Debian (ARMv7) ship packages for different architectures. There is a good reason why one stays with the Debian repo, at best backports, but not mix multiple version if there is no strong reason to do that!

Is there any issue with this Dropbear version? If so, and a newer version ships a solution, I suggest to compile it manually instead of applying and installing from Debian Sid. You know who Sid is (Toy Story)? Image what this can do 💥.

Shipping own compiled updated binaries is out of scope. Effort is too high and I really cannot see any "real" benefit of doing that. Important bug/security fixes are backported into older package versions by Debian package maintainers, usually.

Joulinar, 2020-02-07:
yep that was the explanation I was looking for @MichaIng 😉

WolfganP, 2020-02-07:
Understood, thx @MichaIng. I was looking at dropbear version in light of the latest news of sudo security issues, so I thought worthy to open the issue just in case.
Thx again for the explanation!

closed 2020-02-07 by WolfganP

MichaIng, 2020-02-07:
@WolfganP 
The sudo security fix has been quickly backported to all Debian version, AFAIK. So should be done with any serious Dropbear security issue.

WolfganP, 2020-02-07:
@MichaIng thx. The sudo security issue was definitely fixed a couple of days ago on v1.8.31 after further code revision (https://www.sudo.ws/alerts/pwfeedback.html, updated on Feb 5th)

Anyways, my dietpi install doesn't have the offending `pwfeedback` entry enabled (you may check your system with `sudo -l`), so no security risk (for this incident) even in older version (v1.8.27 currently installed)