MichaIng/DietPi #3372: Issue installing Sonarr on a fresh install of Dietpi on RPi 4 B with 4 GB ram

coolmax12, 2020-02-06:
#### Details:
- Date           | Thu  6 Feb 19:47:01 GMT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Fresh install of DietPi
2. Tried to install bunch of software on first run along with sonarr

#### Expected behaviour:

- I expect a smooth install as it's a fresh dietpi install with nothhing done to it.

#### Actual behaviour:

- Keeps saying the key is not right (or somethhing.. not super techhnical)

#### Extra details:

- I tried to google and found a command on an older thread that tries to update the key.

apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493

But it didn't work.

I get

Executing: /tmp/apt-key-gpghome.NIvmRkCx9m/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: keyserver receive failed: Server indicated a failure

#### Additional logs:
```
Ign:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://downloads.plex.tv/repo/deb public Release
Hit:3 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Hit:4 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Get:7 https://apt.sonarr.tv develop InRelease [13.3 kB]
Err:7 https://apt.sonarr.tv develop InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Reading package lists...
W: GPG error: https://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.
```

MichaIng, 2020-02-07:
@coolmax12 
Many thanks for your report.

The the instructions are still valid: https://github.com/Sonarr/Sonarr/wiki/Installation#linux
Also it worked well on my testing system:
```
2020-02-07 12:45:04 root@VM-Buster:~$ apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
Executing: /tmp/apt-key-gpghome.ZLc579SWVz/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: key EBFF6B99D9B78493: public key "NzbDrone <contact@nzbdrone.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
```
Could you simply retry it now? Probably it was a temporary issue with the key server or cert timeout or such.

coolmax12, 2020-02-07:
Thanks Michal,

I tried it.. doesn't work. I get the same error:

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys
0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
Executing: /tmp/apt-key-gpghome.NXF0idTkjo/gpg.1.sh --keyserver
keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: keyserver receive failed: Server indicated a failure

This is again on a fresh install. I tried it twice and I get the same
error. I noticed that I had "https://apt.sonarr.tv develop main" as the
source so I even tried to change it to the source mentioned in the link you
shared about sonarr. Even that won't work.

I have an older Pi3 and everything runs fine on that.

MichaIng, 2020-02-07:
@coolmax12 
Very strange... Since you only changed the repo branch from develop to stable, the same key is required. I'll try it on my RPi as well later.

coolmax12, 2020-02-07:
I fixed it!!

It looks like it's a DNS / port / something issue.

I used the solution here (and you may want to update your string with the
same hkp and port 80 to ensure it works for others.

https://unix.stackexchange.com/questions/399027/gpg-keyserver-receive-failed-server-indicated-a-failure

"hkp://keyserver.ubuntu.com:80" instead of just "keyserver.ubuntu.com" and
it worked!

MichaIng, 2020-02-07:
@coolmax12 
Hmm, this is not a fix but a workaround. An since you're on Buster, this should be fixed with the dirmngr version already 🤔. Port 80 usually means unencrypted HTTP, although hkp:// protocol here.

I retried on RPi, again with success:
```
root@BuildingRPi:~# apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
Executing: /tmp/apt-key-gpghome.Y4AVxLALhb/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: key EBFF6B99D9B78493: public key "NzbDrone <contact@nzbdrone.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
root@BuildingRPi:~# apt-key list
/etc/apt/trusted.gpg
--------------------
...
pub   rsa2048 2013-08-27 [SC]
      A236 C58F 4090 91A1 8ACA  53CB EBFF 6B99 D9B7 8493
uid           [ unknown] NzbDrone <contact@nzbdrone.com>
sub   rsa2048 2013-08-27 [E]
```

Were there some possibly related packages (ssl/crypto/dirmngr or similar) upgraded during Sonarr install/DietPi-Software run?

coolmax12, 2020-02-08:
I don't recall seeing such package updates.. if you can tell me how to
generate a detailed log or where to find a detailed log, I will gladly send
you a log dump?

Thanks!

MichaIng, 2020-02-10:
@coolmax12 
Okay, as long as it works now, it doesn't matter how you got the key. The repo is accessed via HTTPS anyway. Still not sure why getting the key fails in your case, but DietPi-Software should now work fine.

MichaIng, 2020-02-24:
Marking as closed, obviously was temporary or edge case only.

closed 2020-02-24 by MichaIng

Rothbardian, 2020-11-05:
I've been hitting this keyserver issue installing Sonarr, again on a fresh installation of DietPi (RasPi 4, 4GB RAM), and tried several different potential solutions. I was finally able to download the key using @coolmax12's solution from Feb 7, 2020:
`apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493`

Thanks all!

MichaIng, 2020-11-05:
Strange, however, I'm going to replace app `apt-key` usage with manual `gpg` key conversion and storing to `/etc/apt/trusted.gpg.d/sonarr.gpg`, so that issue is avoided then as well.
Reason for the change is mainly that `apt-key` has been deprecated and will be removed from APT in Q2/2022, so it makes sense to prepare and migrate.