MichaIng/DietPi #3373: - DietPi has encounted an error, and, is unable to continue

cirleno, 2020-02-08:
#### Details:
- Date           | Sat  8 Feb 00:27:40 -02 2020
- Bug report     | N/A
- DietPi version | v6.13
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1134 SMP Tue Aug 14 17:10:10 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Get:61 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf libgd3 armhf 2.2.4-2+deb9u5 [114 kB]
Get:62 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf libicu57 armhf 57.1-6+deb9u3 [7,425 kB]
Err:1 https://archive.raspberrypi.org/debian stretch/main armhf firmware-atheros all 1:20161130-3+rpt4
  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
Get:71 https://archive.raspberrypi.org/debian stretch/main armhf firmware-brcm80211 all 1:20161130-3+rpt4 [3,672 kB]
Get:63 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf libxslt1.1 armhf 1.1.29-2.1+deb9u1 [215 kB]
Get:64 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf mariadb-common all 10.1.38-0+deb9u1 [28.4 kB]
Get:65 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf mariadb-client-core-10.1 armhf 10.1.38-0+deb9u1 [4,359 kB]
Get:66 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u1 [713 kB]
Get:67 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-mysql armhf 7.0.33-0+deb9u6 [103 kB]
Get:68 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-intl armhf 7.0.33-0+deb9u6 [102 kB]
Get:69 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-xml armhf 7.0.33-0+deb9u6 [86.9 kB]
Get:70 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-json armhf 7.0.33-0+deb9u6 [15.2 kB]
Get:72 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-zip armhf 7.0.33-0+deb9u6 [17.3 kB]
Get:73 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-opcache armhf 7.0.33-0+deb9u6 [64.8 kB]
Get:74 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-fpm armhf 7.0.33-0+deb9u6 [1,064 kB]
Get:75 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-curl armhf 7.0.33-0+deb9u6 [24.3 kB]
Get:76 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-mcrypt armhf 7.0.33-0+deb9u6 [12.7 kB]
Get:77 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-gd armhf 7.0.33-0+deb9u6 [21.9 kB]
Get:78 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-mbstring armhf 7.0.33-0+deb9u6 [413 kB]
Get:79 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-readline armhf 7.0.33-0+deb9u6 [10.8 kB]
Get:80 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-cli armhf 7.0.33-0+deb9u6 [1,052 kB]
Get:81 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf php7.0-common armhf 7.0.33-0+deb9u6 [487 kB]
Get:82 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf proftpd-basic armhf 1.3.5b-4+deb9u2 [2,376 kB]
Get:83 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf python-cryptography armhf 1.7.1-3+deb9u1 [173 kB]
Get:84 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf python-parsedatetime all 2.1-3+deb9u1 [38.4 kB]
Get:85 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf redis-server armhf 3:3.2.6-3+deb9u3 [341 kB]
Get:86 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf redis-tools armhf 3:3.2.6-3+deb9u3 [381 kB]
Get:87 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf rsync armhf 3.1.2-1+deb9u2 [356 kB]
Get:88 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf sudo armhf 1.8.19p1-2.1+deb9u2 [1,031 kB]
Get:89 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf unzip armhf 6.0-21+deb9u2 [151 kB]
Get:90 http://raspbian.c3sl.ufpr.br/raspbian stretch/main armhf usbutils armhf 1:007-4+deb9u1 [226 kB]
Get:91 https://archive.raspberrypi.org/debian stretch/main armhf firmware-misc-nonfree all 1:20161130-3+rpt4 [2,321 kB]
Err:91 https://archive.raspberrypi.org/debian stretch/main armhf firmware-misc-nonfree all 1:20161130-3+rpt4
  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
Get:92 https://archive.raspberrypi.org/debian stretch/main armhf firmware-realtek all 1:20161130-3+rpt4 [342 kB]
Get:93 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel armhf 1.20190819~stretch-1 [35.4 MB]
Get:94 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi-bin armhf 1.20190819~stretch-1 [338 kB]
Get:95 https://archive.raspberrypi.org/debian stretch/main armhf libraspberrypi0 armhf 1.20190819~stretch-1 [843 kB]
Get:96 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-bootloader armhf 1.20190819~stretch-1 [3,581 kB]
Get:97 https://archive.raspberrypi.org/debian stretch/ui armhf python-six all 1.12.0-0+rpt1 [13.4 kB]
Get:98 https://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-sys-mods armhf 20181127 [10.2 kB]
Get:99 https://archive.raspberrypi.org/debian stretch/main armhf raspi-copies-and-fills armhf 0.11 [7,274 B]
Get:100 https://archive.raspberrypi.org/debian stretch/main armhf wireless-regdb all 2018.05.09-0~rpt1 [11.8 kB]
Get:101 https://archive.raspberrypi.org/debian stretch/main armhf wpasupplicant armhf 2:2.6-21~bpo9~rpt1 [837 kB]
Get:102 https://archive.raspberrypi.org/debian stretch/main armhf fake-hwclock all 0.11+rpt1 [7,126 B]
Fetched 103 MB in 2min 18s (741 kB/s)
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-atheros_20161130-3+rpt4_all.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20161130-3+rpt4_all.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

MichaIng, 2020-02-10:
Many thanks for your report. I remember this error from older stretch images but not sure about the solution. Please try to upgrade all packages from Raspbian only first:
```
mv /etc/apt/sources.list.d/raspi.list /etc/apt/sources.list.d/raspi.list.bak
apt update
apt upgrade
apt full-upgrade
mv /etc/apt/sources.list.d/raspi.list.bak /etc/apt/sources.list.d/raspi.list
apt update
apt upgrade
apt full-upgrade
```

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng