MichaIng/DietPi #3376: [Odroid XU4] [Bug] [Beta] Looping error on first boot

Technetium1, 2020-02-09:
#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`

`#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

`buster`

- Kernel version | `uname -a`

`Linux DietPi 4.14.165+ #1 SMP PREEMPT Mon Jan 20 10:57:56 UTC 2020 armv7l GNU/Linux`

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)

`Odroid XU3/XU4/HC1/HC2 (armv7l)`

- Power supply used | (EG: 5V 1A RAVpower)

`Odroid Provided`

- SDcard used | (EG: SanDisk ultra)

`Sandisk Ultra 32GB`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?

`Fresh install`

- Can this issue be replicated on a fresh installation of DietPi?

`Yes`

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

`Bug report sent, reference code: 1777adf1-046e-4224-8f91-1a1635c69039`

#### Steps to reproduce

`Flash image, login via SSH, get error loop that only ends when cancelling.`

#### Expected behaviour

`Shouldn't loop to an error infinitely.`

#### Actual behaviour

`Error looping until cancelled.`

#### Extra details
`I have no connectivity issues to any services.`

```bash

root@DietPi:/var/tmp/dietpi/logs# cat dietpi-firstboot.log
[ SUB1 ] DietPi-Set_swapfile > Apply
[  OK  ] DietPi-Set_swapfile | Disable swapfile
[  OK  ] DietPi-Set_swapfile | swapoff -a
[  OK  ] DietPi-Set_swapfile | Free space check: path=/var/ | available=28850 MiB | required=54 MiB
[  OK  ] DietPi-Set_swapfile | Generating new swapfile
[ INFO ] DietPi-Set_swapfile | Size     = 54 MB
[ INFO ] DietPi-Set_swapfile | Location = /var/swap
[  OK  ] DietPi-Set_swapfile | fallocate -l 54M /var/swap
[  OK  ] DietPi-Set_swapfile | mkswap /var/swap
[  OK  ] DietPi-Set_swapfile | swapon /var/swap
[  OK  ] DietPi-Set_swapfile | Setting in /DietPi/dietpi.txt adjusted: AUTO_SETUP_SWAPFILE_SIZE=54
[  OK  ] DietPi-Set_swapfile | Desired setting in /DietPi/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
[ INFO ] DietPi-Set_swapfile | Setting /tmp tmpfs size: 1024M
[  OK  ] DietPi-Set_swapfile | mount -o remount /tmp
[ INFO ] DietPi-FirstBoot | Setting Timezone Europe/London. Please wait...

Current default time zone: 'Europe/London'
Local time is now:      Mon Feb  3 23:39:40 GMT 2020.
Universal Time is now:  Mon Feb  3 23:39:40 UTC 2020.

[ SUB1 ] DietPi-Set_hardware > eth-forcespeed (disable)
[  OK  ] eth-forcespeed disable | Completed
[  OK  ] DietPi-Change_hostname | Desired setting in /etc/hosts was already set: 127.0.1.1 DietPi
[ SUB1 ] DietPi-Set_software > apt-mirror (https://deb.debian.org/debian/)
[  OK  ] DietPi-Set_software | Desired setting in /DietPi/dietpi.txt was already set: CONFIG_APT_DEBIAN_MIRROR=https://deb.debian.org/debian/
[  OK  ] apt-mirror https://deb.debian.org/debian/ | Completed
Generating Dropbear DSS host key.  Please wait.
Generating 1024 bit dss key, this may take a while...
Public key portion is:
ssh-dss AAAAB3NzaC1kc3MAAACBAL4sQ3mmAE6dRhJmC8RZLRjMckM8a49nZrrmHIN1pCurNYr7Rm3Gce4dlRiRIMQ2ZbjCU3nMqJM4nuiV53LJAEcVs9ejklIPv+nPhhveNU0GhwdtpRrdiDR5kU+ZaK4ETPpccO0IYfok7i4Kk+b6mVSoeJdpT8DS7PwRctMnGSbBAAAAFQDemmgVBEwBhVLRRdgjb4rc4Qu4AQAAAIAjYhRogwszRjD0hAGiCJfN3XtWrqAyqcyeQHotbay5WIsU7DeAF441J4MXIbFi2pUzigXk0bsw6rZxiXVi0Gnb4dNjzZmohVgz8lmaoSCbEgGfphON1G6GjpLeLUX9IuzRJ+EdK/w1MRK9Nayby24ttlam+vnuQZmkvfl9eYLPOAAAAIEAor9jYtFN+JuBLZwoDQur72tZdOT5+o+zxtSFsH3noLKweaEo1CgbVP3HjOt+rl/uo2Y+/KsDzqxJPB8K22XrK8rpGqhOYbCgkO2ghb38YBSUXYGdUhBEJPhII8rtLJk9S/OZ/f74tCVIkx84dyUUL2TgI8Q7FYwvkYUYqZmmxhw= root@DietPi
Fingerprint: sha1!! 64:ca:4a:4d:75:20:75:c6:b7:25:95:98:74:49:12:3f:9f:56:70:e7
Generating Dropbear RSA host key.  Please wait.
Generating 2048 bit rsa key, this may take a while...
Public key portion is:
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQC7f1VPBZnFtJ/8QDWxiPRCv/zaxWza4rKTqGQ/SO4rCuXRNOd+b48SsZxr4Z3jhB0zRv0KeblaG6faA7GoEyAOd+g4JFxiNPmRg/sMGKC8BKKfyzMzErskZRvXfJ7324IoWhn14xYWnOQ3kDg40zvYgDoMtuqWY/GstozeZ9oi+IxNw9joAm9sC3l95rRJUlCw+KmMXrs4dKvt+51XGWdP9rDszXMyDL4YdJ14yOBCTJ+dPnw+54RHCROvtu+hr3uLdS4+wr4c9CUX+kvF6+EfYHI6z5UHC1sTS5qd/0kphjKH1gjEWlQVhX+4PKyZvkXid/qS/RQMIDO7kUV/Jzt5 root@DietPi
Fingerprint: sha1!! 78:97:ef:f6:63:32:a8:30:00:e9:6a:e1:f1:ad:58:68:4b:62:61:30
Generating Dropbear ECDSA host key.  Please wait.
Generating 256 bit ecdsa key, this may take a while...
Public key portion is:
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBNUcQSIbCvaekaXowPZMKDR3vx9OP1Quexun19ZePQk6QN/QEjfIl0LMxBAYR4SdjHeTtzTaxJm4ADf9uedMREk= root@DietPi
Fingerprint: sha1!! b9:a4:5a:66:9b:39:83:8a:a1:d6:4a:22:f4:8c:d1:23:4e:a4:35:c0
Initializing machine ID from random generator.
[ SUB1 ] DietPi-Set_hardware > wifimodules (disable)
[  OK  ] wifimodules disable | Completed
[ SUB1 ] DietPi-Set_hardware > enableipv6 (enable)
[  OK  ] DietPi-Set_hardware | Setting in /etc/hosts adjusted: ::1 localhost ip6-localhost ip6-loopback
[  OK  ] DietPi-Set_hardware | Setting in /etc/hosts adjusted: ff02::1 ip6-allnodes
[  OK  ] DietPi-Set_hardware | Setting in /etc/hosts adjusted: ff02::2 ip6-allrouters
[  OK  ] DietPi-Set_hardware | Desired setting in /DietPi/dietpi.txt was already set: CONFIG_ENABLE_IPV6=1
[  OK  ] enableipv6 enable | Completed
[ SUB1 ] DietPi-Set_hardware > preferipv4 (enable)
[  OK  ] DietPi-Set_hardware | Desired setting in /etc/wgetrc was already set: prefer-family = IPv4
[  OK  ] DietPi-Set_hardware | Desired setting in /DietPi/dietpi.txt was already set: CONFIG_PREFER_IPV4=1
[  OK  ] preferipv4 enable | Completed
Removed /etc/systemd/system/multi-user.target.wants/dietpi-firstboot.service.
```

MichaIng, 2020-02-09:
@Technetium1 
Many thanks for testing and reporting. It is this issue, right? https://github.com/MichaIng/DietPi/issues/2979#issuecomment-583046383

I am currently redoing the image and apply the fix. Please test this:
```
update-ca-certificates --fresh
/DietPi/dietpi/dietpi-login
```
- This should then succeed with the curl command and finish firstrun setup.

If this works, no need to reflash btw, since I do nothing else with the beta image as well 😉.

MichaIng, 2020-02-09:
🈴 Hmm above still did no work, exact solution from the other thread:
```
update-ca-certificates
update-ca-certificates --fresh
/DietPi/dietpi/dietpi-login
```
However this should not make any difference 🤔...

MichaIng, 2020-02-09:
@Sudrien 
I tried to fix it on the image for a while now but cannot get it working. I saw your find about the `*.0` symlinks and indeed I cannot get them to be created. Tried to reinstall all related packages, configuring and `update-ca-certificates --fresh` several times, re-checked config files are default etc.
This last command was all that solved it for you, did I get this right?

MichaIng, 2020-02-09:
Okay I think I found it. Those symlinks are hashes and those are used/required by some programs, i.e. curl. Strange is that `openssl rehash` does not (in my case) do what it should, which is the reason reinstalling/re-configuring all these packages does not resolve it. Instead what works for me is `c_rehash`, which recreates the hash symlinks and fixes curl...

Sudrien, 2020-02-09:
@MichaIng Yes, `update-ca-certificates --fresh` was the only thing I needed to do from command line

MichaIng, 2020-02-09:
@Sudrien
Probably it has something to do with the systemd-nspawn container, not sure. However I just uploaded the fixed image. No need to anyone to reflash, when having it mixed manually though.

Technetium1, 2020-02-10:
@MichaIng Thank you for fixing this! I can confirm that it worked!

closed 2020-02-10 by Technetium1