MichaIng/DietPi #3378: Software installer: Error while Nextcloud installation

Hafii, 2020-02-12:
───────┤ DietPi-Software | Error handler├───────

 - Command: ncc maintenance:install
 - Exit code: 1
 - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:5
 - Image creator: DietPi Core Team
 - Pre-image: Raspbian Lite

Log file content:
Error while trying to create admin user: Failed to connect to the database: An
exception occurred in driver: SQLSTATE[HY000] [1698] Access denied for user
'tmp_root'@'localhost'
 ->

MichaIng, 2020-02-12:
@Hafii 
Many thanks for your report. I'll check and provide some manual steps later.

MichaIng, 2020-02-12:
Okay try to go though these steps manually:
```
systemctl start mariadb
systemctl start redis-server
ncc(){ sudo -u www-data php /var/www/nextcloud/occ "$@"; }
mysql -e "grant all privileges on *.* to 'tmp_root'@'localhost' identified by 'dietpi' with grant option"
ncc maintenance:install --no-interaction --database 'mysql' --database-name 'nextcloud' --database-user 'tmp_root' --database-pass 'dietpi' --admin-user 'admin' --admin-pass 'dietpi' --data-dir '/mnt/dietpi_userdata/nextcloud_data'
mysql -e 'drop user tmp_root@localhost'
```
- You can adjust `--admin-user` and `--admin-pass` values to your needs of course, which is the initial admin account to log into Nextcloud web UI.

If the above steps report success, **please do not log into Nextcloud yet**, but redo/finish the install first:
```
dietpi-software install 114
```

Hafii, 2020-02-13:
Thank you for the quick response.
Still the same Error(with changed Admin Name) when executing: nn maintainance:install .....
Here the Error message:
Error while trying to create admin user: Failed to connect to the database: An exception occurred in driver: SQLSTATE[HY000] [1698] Access denied for user 'oc_admin'@'localhost'

MichaIng, 2020-02-14:
@Hafii 
Ah, it might be this: https://github.com/nextcloud/server/issues/18513
Strange only that it should have been fixed. We worked around the issue by pulling an older Nextcloud version, but all has been backported now. I'll try to replicate.

MichaIng, 2020-02-14:
Ah dammit, the MySQL 8 fix, that broke MariaDB with Nextcloud 18+, has been backported to Nextcloud 17 meanwhile: https://github.com/nextcloud/server/commit/f3ad49b866ae6e616559cb2d49c3632078735a5b
Since we pull NC17 as workaround, this does not work anymore, of course.

Meanwhile as above on NC18 the complete fix has been merged, but not yet into current NC17...

Fix at least is simply to pull most current Nextcloud again... Hang on...

Hafii, 2020-02-15:
I am using stock dietpi, first and only thing i want to do is install nextcloud with the software installer. First time i got this issue was Christmas holidays(don't know which dietpi version is was, maybe v6.26...) now i updates dietpi and still get this error.

MichaIng, 2020-02-15:
@Hafii 
This error happened once indeed because of the mentioned regression with NC17.0.2.

Did you flash the new DietPi image freshly or just retried installing Nextcloud on the same image? Probably there are issues because of the partial install done until failure.

Please try the following to purge all related installed software and start from scratch:
```
apt purge mysql-server lighttpd redis-server
apt autoremove --purge
rm -Rf /etc/{mysql,apache2} /mnt/dietpi_userdata/{mysql,nextcloud_data}
```
Then freshly install Nextcloud:
```
dietpi-software install 114
```

CLUB777ONE, 2020-02-17:
dietpi-software install 114 - It does not work because the file "nextcloud-17.0.1.tar.bz2" is not available on the Nextcloud server.

Current DietPi uses this link to download nextcloud: https://download.nextcloud.com/server/releases/nextcloud-17.0.1.tar.bz2

Meanwhile, the Nextcloud v. 18.0.1 has been released. Is there a way to update the link in DietPi myself? If so, in which DietPi-File could I replace the old link with the new one?

https://download.nextcloud.com/server/releases/nextcloud-18.0.1.zip

MichaIng, 2020-03-17:
Okay, basically the workaround has been removed and newest Nextcloud 18 has the issue solved, hence there should be no issue anymore with DietPi v6.29.

On our current testing images, I applied the fix as well and it is hacked into running systems via MOTD (a practice which I aim to replace with a transparent live-patches system).

closed 2020-05-03 by MichaIng