MichaIng/DietPi #3380: Checking network connectivity fails (via ssh)

lokmanhakim, 2020-02-13:
#### Details:
- Date           | Thu 13 Feb 16:40:13 +08 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B (armv6l) (ID=0)
- Kernel version | Linux DietPi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 1.1.1.1
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:
1. ...whenever try to install dietpi software, dietpi will run network check but it always fails.
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- connected via SSH
- when ping normally, it goes through

#### Additional logs:
```
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2020-02-13:
@lokmanhakim 
many thanks for your report.

if you are not able to `ping 1.1.1.1` you would have local issues as it's part of Cloudflares DNS Server. It is the globally available IP address for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `google.com`

It's a similar issue like this #3337

MichaIng, 2020-02-13:
Aside of the workaround provided by Joulinar, if there is any chance, check your router/local network settings, if it somehow uses 1.1.1.1 (respectively the whole 1/8 IP range) internally. This is the case for e.g. older Cisco routers. If possible, then disable this internal usage to allow connection to these IPs on global level.

Background about that topic: https://news.ycombinator.com/item?id=14862748

teamallnighter, 2020-03-25:
│ Checking network connectivity                                                            
  │                                                                                          
  │  - Command: ping -c 1 -W 5 8.8.8.8                                                       
  │  - Exit code: 1                                                                          
  │  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5         
  │  - Image creator: DietPi Core Team                                                       
  │  - Pre-image: Raspbian Lite                                                              
  │                                                                                          
  │ Log file content:                                                                        
  │ PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.                                             
  │                                                                                          
  │ --- 8.8.8.8 ping statistics ---                                                          
  │ 1 packets transmitted, 0 received, 100% packet loss, time 0ms                            
  │

Joulinar, 2020-03-25:
So you are even not able to ping Google DNS server? What about Quad9 DNS server 9.9.9.9 ?

teamallnighter, 2020-03-25:
No. but when I go to into config and do a network test it goes to google

MichaIng, 2020-03-25:
@teamallnighter 
Hmm, when you run the following from console:
```
ping 8.8.8.8
```
Are all packages lost or are probably some answers received? Probably the connection is simply a bid unstable.

Joulinar, 2020-03-25:
Can you do a `ping google.com` from SSH command line

Superbatros, 2020-04-22:
Hello !

I have the problem teamallnighter encounter. I made the change in diepi.txt file but it has no effects (I'm connected throug ssh).
I tried ping google.com but all the packet sent are lost even if I see I'm connected with diepi-config network menu.
Could you please help me?

Joulinar, 2020-04-22:
ok it seems DNS resolution is not working correctly. Do you use an own DNS server or do you use a public one? How does DNS configuration on your system looks like?

Superbatros, 2020-04-22:
> ok it seems DNS resolution is not working correctly. Do you use an own DNS server or do you use a public one? How does DNS configuration on your system looks like?

I use a public DNS (I guess : I use the default configuration given by my internet provider). 

I just manage to run dietpi-software : I change one.one.one.one by 1.1.1.1 in dietpi.txt file. Now it works but I don’t understand why... thanks for your help !

Joulinar, 2020-04-22:
what did you changed exactly? `CONFIG_CHECK_DNS_DOMAIN=one.one.one.one` into `CONFIG_CHECK_DNS_DOMAIN=1.1.1.1`?? If yes, you are now going to bypass DNS check. As far as understood you are not able to ping FQDN from os level. So that's something to find out.

pls can you do following and post the output.
```
ping -c 1 -W 5 one.one.one.one
ping -c 1 -W 5 dns.google
ping -c 1 -W 5 dns.quad9.net
```

Superbatros, 2020-04-22:
> 
> 
> what did you changed exactly? `CONFIG_CHECK_DNS_DOMAIN=one.one.one.one` into `CONFIG_CHECK_DNS_DOMAIN=1.1.1.1`??

Yes I dit that and it works : the DNS resolve test works (I don't know if it is bypassed or not) and I manage to run the installation of diet pi, the update and upgrade after first launch.

Joulinar, 2020-04-22:
yes if you changed `CONFIG_CHECK_DNS_DOMAIN` from a `FQDN` to an `IP address`, you are going to bypass the DNS check completely.

Superbatros, 2020-04-22:
> 
> 
> yes if you changed `CONFIG_CHECK_DNS_DOMAIN` from a `FQDN` to an `IP address`, you are going to bypass the DNS check completely.

Ok! Understood. As it was enough to go through the installation of diet pi, I think I won't change it.

MichaIng, 2020-04-22:
@Superbatros 
Hmm strange that obviously DNS resolution in other cases works, only for ping not. Could you run the following commands and paste them here:
```
ping -c 1 -W 5 google.com
ping -c 1 -W 5 github.com
wget --spider google.com
wget --spider github.com
```

closed 2020-07-09 by MichaIng