MichaIng/DietPi #3381: [FR] Home Assistant + Addons in DietPi-Software

Hukuma1, 2020-02-14:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Right now the installation of Home Assistant on Diet-Pi via DietPi-Software is a no-thrills (and unsupported as I recall) version.

#### Describe the solution you'd like:

- A modified install of the current DietPi-Software Home Assistant script could get us entirely there for full support from HA team. They have a script here: https://github.com/home-assistant/hassio-installer

#### Describe alternatives you've considered:

- Right now it's simple, but messy. Most of the dependencies for HA are already installed by default from DietPi. The only missing ones were `docker, jq, and avahi-daemon`. So you can install DietPi, add those deps and then run the HA script to install the most versatile/supported version of HA: 

`curl -sL https://raw.githubusercontent.com/home-assistant/hassio-installer/master/hassio_install.sh | bash -s`

It works this way, but a bit more messy and would love to integrate HA more into DietPi like the other software is. I'm not a coder, but would love to help to contribute by testing if that helps speed the integration along for everyone to enjoy it fully. The version with addons is essentials I think. Adding any addons post-install is a huge nuisance.

#### Additional context

- Found a 6 month old request for the same exact thing on feathub below. Hopefully this invigorates some new interest! :)

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/+105

MichaIng, 2020-02-14:
@Hukuma1 
Note that "Home Assistant" is not "Hass.io". The first is the actual software while the second is a whole OS/environment around it with various deep/intrusive system changes, config possibilities etc. If you really want all of this, then I suggest you flash the Hass.io images, since it will break many DietPi scripts and its other install options. We want and need clean/modular non-intrusive software installs, which do what they are intended to and not turn the whole system into a single-purpose machine.

The above is the same reason why we needed to drop support for OpenMediaVault, RetroPie, had an own Pi-hole installer script for a while and cannot support NextCloudPi. Those do or did not only install the core software, but adjusted the whole system around it. This is great if you only want to use this single software, but it is bad if you want to run other software beside it, where you need a compatible generic OS. Especially difficult are tools which configure system settings in a specific format/style which breaks other (i.e. ours) config scripts and expectations. There have been much done our side and many plans are there to make DietPi most robust and compatible with all kind of 3rd-party configs/setup, but it is a long way, taking into account the very strict/intrusive way DietPi started as (great for simple scripting, much automation etc, bad for compatibility with custom/3rd-party modifications).

Not sure what you mean by "unsupported". HA does not offer a payed/guaranteed support plan, hence you can report to their forums or GitHub repo and you'll get an answer earlier or later. I know that their non-Hass.io install docs/recommendations do not use `pyenv` but `python-venv`, which means that you have a different Python version based on the underlying Debian version, Raspbian vs Debian, backports vs non-backports etc, which does not make it more stable or replicable but more difficult, compared to `pyenv`, where the whole Python environment is 100% the same, where ever you install it. If you have a support request/issue open on HA, report back, then I can give further details about our install method and why I believe it is even more "stable" then venv.

Of course any issues with the install process itself should be reported here first, since this might have nothing to do with HA.

Hukuma1, 2020-02-14:
Ah, I wasn't aware it was such a technical hurdle. That's a shame. Thanks for the explanation at least.

I know HA just launched a re-branding of sorts (https://www.home-assistant.io/blog/2020/01/29/changing-the-home-assistant-brand/) so as I understand it Hass.io install will become "Home Assistant" and the stripped down Home Assistant will be "Home Assistant Core" I believe.

I guess the good news is that installing it how I've been doing it on top of DietPi has so far been working as expected.

closed 2020-02-14 by Hukuma1

MichaIng, 2020-02-16:
@Hukuma1 
Good to know about the re-branding. This step is reasonable to reduce support effort on custom systems. Since Hass.io, either as OS image or via its install scripts, creates a certain system setup around Home Assistant, it is then easier to support. However makes it more difficult for any distribution maintainer to implement it. We'll stay with HA Core then for the mentioned reasons.

However all this does not mean that you cannot simply use Hass.io installer to install it on top of DietPi, which is not much more then a modified Debian with a software install and config UI script set. Simply take in mind that some system setup/config scripts from DietPi and Hass.io (they have drive management and network setup tools, AFAIK?) will conflict each other, hence use only one tool set and not both in parallel.