MichaIng/DietPi #3384: Cannot remove ALSA Xserver XFCE NoMachine WiFi Hotspot

Crasher124, 2020-02-15:
#### Details:
- Date           | Sat 15 Feb 11:57:47 CET 2020
- Bug report     | 30e3b511-95ac-4b3d-8465-6b9df23603d3
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGP xorg* xserver-xorg* x11-* xinit xcompmgr xterm xfonts-base
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. Uninstall ALSA Xserver XFCE NoMachine WiFi Hotspot

#### Expected behaviour:
- It shoud remove the programs

#### Actual behaviour:
- it dosent remove the programs

#### Extra details:
- i did the removing via ssh

#### Additional logs:
```
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

MichaIng, 2020-02-15:
@Crasher124 
Many thanks for your report. Yeah I recognised this when creating images where desktop is preinstalled. Can you please run the following and paste output:
```
apt purge $(dpkg --get-selections | mawk '{print $1}' | grep '^(xorg|xserver-xorg|x11-|xinit$|xcompmgr$|xterm$|xfonts-base$')
```

Crasher124, 2020-02-15:
Here is the output
`root@DietPi:~# apt purge $(dpkg --get-selections | mawk '{print $1}' | grep '^(xorg|xserver-xorg|x11-|xinit$|xcompmgr$|xterm$|xfonts-base$')
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@DietPi:~#
`

MichaIng, 2020-02-15:
@Crasher124 
Whoopsie, forgot the extended regex option for grep:
```
apt purge $(dpkg --get-selections | mawk '{print $1}' | grep -E '^(xorg|xserver-xorg|x11-|xinit$|xcompmgr$|xterm$|xfonts-base$')
```

Crasher124, 2020-02-15:
[image]

Is this better?

MichaIng, 2020-02-15:
@Crasher124 
Still a syntax error, damn thing that APT itself does not silently ignore non-installed packages 😄:
```
apt purge $(dpkg --get-selections | mawk '{print $1}' | grep -E '^(xorg|xserver-xorg|xinit$|xcompmgr$|xterm$|xfonts-base$|x11-xserver-utils$|x11-utils$|x11-xkb-utils$)')
```
I just recognised that x11-common is required for the Java package (if installed), hence this must not be purged. I adjusted the pattern above.

Crasher124, 2020-02-16:
It worked thanks 😃

MichaIng, 2020-02-16:
@Crasher124 
Finally, great. Running this does not mark Xserver as uninstalled within `dietpi-software`, however it is practically uninstalled 😉. You could mark it manually:
```
sed -i '/aSOFTWARE_INSTALL_STATE\[6\]=2/aSOFTWARE_INSTALL_STATE\[6\]=0/' /DietPi/dietpi/.installed
```

Crasher124, 2020-02-17:
I did run the command but it is still in dietpi software. It's not important to have it not anymore listed,
But thanks for the help. 😃

MichaIng, 2020-02-17:
@Crasher124 
I need to better review my commands :sweat_smile: it's (missed `s` above to do the actual replacement):
```
sed -i 's/aSOFTWARE_INSTALL_STATE\[6\]=2/aSOFTWARE_INSTALL_STATE\[6\]=0/' /DietPi/dietpi/.installed
```

closed 2020-02-18 by MichaIng