MichaIng/DietPi #3385: Nextcloud file is not available on the Nexcloud server. How replace the old link with the new one?

CLUB777ONE, 2020-02-17:
dietpi-software install 114 - It does not work because the file "nextcloud-17.0.1.tar.bz2" is not available on the Nextcloud server.

Current DietPi uses this link to download nextcloud: https://download.nextcloud.com/server/releases/nextcloud-17.0.1.tar.bz2

Meanwhile, the Nextcloud v. 18.0.1 has been released. Is there a way to update the link in DietPi myself? If so, in which DietPi-File could I replace the old link with the new one?

https://download.nextcloud.com/server/releases/nextcloud-18.0.1.zip

Joulinar, 2020-02-17:
Hi,

many thanks for your report. Nextcloud download link will be updated on v6.29. However you can change it already within `/DietPi/dietpi/dietpi-software` file. Below the code section from current dev branch

https://github.com/MichaIng/DietPi/blob/d8b4979c77a571359d0295594866bbbd198850e2/dietpi/dietpi-software#L3448

CLUB777ONE, 2020-02-17:
It works great, thanks

> link

closed 2020-02-17 by CLUB777ONE

MichaIng, 2020-02-17:
@CLUB777ONE 
Many thanks for your report. Huh since when the remove old tarballs... Actually they did not: https://download.nextcloud.com/server/releases/
The file shows up there, but when I try to download it, the browser indeed loads forever...

I'll hack some `sed` into MOTD to fix it for systems quickly.

CLUB777ONE, 2020-02-18:
don't mention it. Thank you for the wonderful DietPi!

MichaIng, 2020-02-18:
The 17.0.1 tarball btw downloads fine again, hence this was a temporary server issue only.