MichaIng/DietPi #3386: cannot install ampache

drieksje, 2020-02-18:
#### Details:
- Date           | Tue 18 Feb 20:29:02 CET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET 2020 aarch64 GNU/Linux
- Distro         | stretch (ID=4)
- Command        | G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql php7.3-redis
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: Package 'php7.3-fpm' has no installation candidate
E: Package 'php7.3-curl' has no installation candidate
E: Package 'php7.3-gd' has no installation candidate
E: Package 'php7.3-mbstring' has no installation candidate
E: Package 'php7.3-xml' has no installation candidate
E: Package 'php7.3-zip' has no installation candidate
E: Package 'php7.3-mysql' has no installation candidate
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ...
2. ...

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2020-02-18:
Hi,

many thanks for your report. can you try the following to see if this is working

```
apt-get update
apt-get install php7.3-fpm php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-mysql 
```

drieksje, 2020-02-19:
Hi,
When i'm running the commands, it say' s packets arent available, or
missing, or to old version. Running systemctl, give, apache2,redisserver
and proftp arrent running,  can i remove them by command?  And then
re-install?
Do i need apache2 when i'm running tomcat8?
Manny thanks for helping me!!!
Dirk; noob in debian, linux

MichaIng, 2020-02-19:
@drieksje 
Ampache requires the webserver + PHP to provide the web interface. If you have Apache installed already, that is fine.

There should be Ondrejs PHP repo added to your system, which provides the php7.3 packages. Can you run and paste the output:
```
cat /etc/apt/sources.list.d/dietpi-php.list
```

drieksje, 2020-02-20:
Hi, 
back again, after installing Ondrejs repo, and following the 2days Geek installation guide, i could install PHP 7.3 package. So everything is running **exept** Apache2, so i run the command you gave but still can 't install ampache and get errorcode.......
#### Details:
- Date           | Thu 20 Feb 10:34:42 CET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET 2020 aarch64 GNU/Linux
- Distro         | bullseye (ID=6)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Ign:1 http://ppa.launchpad.net/ondrej/php/ubuntu focal InRelease
Hit:2 http://fuzon.co.uk/meveric all InRelease
Ign:3 http://ppa.launchpad.net/ondrej/php/ubuntu focal Release
Hit:4 http://fuzon.co.uk/meveric stretch InRelease
Get:5 http://deb.debian.org/debian unstable InRelease [142 kB]
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
Get:8 http://packages.dotdeb.org jessie InRelease [9,839 B]
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Get:9 https://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Hit:10 https://packages.sury.org/php stretch InRelease
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
Hit:11 https://deb.debian.org/debian-security stretch/updates InRelease
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Get:12 https://deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Get:13 https://deb.debian.org/debian bullseye InRelease [113 kB]
Ign:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Err:6 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main arm64 Packages
  404  Not Found
Ign:14 https://download.webmin.com/download/repository sarge InRelease
Ign:7 http://ppa.launchpad.net/ondrej/php/ubuntu focal/main all Packages
Get:15 http://deb.debian.org/debian unstable/main arm64 Packages.diff/Index [27.9 kB]
Get:16 http://deb.debian.org/debian unstable/main arm64 Packages 2020-02-20-0815.41.pdiff [21.8 kB]
Get:16 http://deb.debian.org/debian unstable/main arm64 Packages 2020-02-20-0815.41.pdiff [21.8 kB]
Hit:17 https://download.webmin.com/download/repository sarge Release
Get:18 https://deb.debian.org/debian stretch-backports/main arm64 Packages.diff/Index [27.8 kB]
Get:19 https://deb.debian.org/debian stretch-backports/main arm64 Packages 2020-02-20-0815.41.pdiff [33 B]
Get:19 https://deb.debian.org/debian stretch-backports/main arm64 Packages 2020-02-20-0815.41.pdiff [33 B]
Get:20 https://deb.debian.org/debian bullseye/main arm64 Packages.diff/Index [27.9 kB]
Get:21 https://deb.debian.org/debian bullseye/main arm64 Packages 2020-02-20-0815.41.pdiff [8,683 B]
Get:21 https://deb.debian.org/debian bullseye/main arm64 Packages 2020-02-20-0815.41.pdiff [8,683 B]
Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1
Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1
Fetched 562 kB in 6s (92.1 kB/s)
Reading package lists...
W: The repository 'http://ppa.launchpad.net/ondrej/php/ubuntu focal Release' does not have a Release file.
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1
E: Failed to fetch http://packages.dotdeb.org/dists/jessie/InRelease  Unable to find expected entry 'main/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch http://ppa.launchpad.net/ondrej/php/ubuntu/dists/focal/main/binary-arm64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php7.3.list:1

drieksje, 2020-02-20:
so you see i have to tray 3 thimes to install php 7.3, can i remove files in that directory??????

MichaIng, 2020-02-20:
@drieksje 
Slowly:
- You added Ondrejs Ubuntu repo while DietPi is Debian
- You added dotdeb.org repo to an ARM system while this just ships x86_64 packages
- You added Debian unstable repo, which one should never do on a production system
- You as well have the Bullseye repo on a Stretch system, which could come from our WireGuard install

However we must clean that up, as you can see there are several errors and warnings about multiple entries etc. Please paste:
```
ls -Al /etc/apt/sources.list{,.d/*}
cat /etc/apt/sources.list{,.d/*}
```

drieksje, 2020-02-20:
Ok,  have paste the commands what to do now,  i have a wireguard tunnel
setup and running as server.  I get also the popup for keybordsettings
(whats the meaning physicle keybord-wired??? Or bluetooth????).  Thanks for
helping me!

Joulinar, 2020-02-20:
You pls copy the results from the 2 commands here into the GitHub post. The command did not fix anything. They will just list down the repros you have connected. 

```
ls -Al /etc/apt/sources.list{,.d/*}
cat /etc/apt/sources.list{,.d/*}
```

drieksje, 2020-02-20:
OK

e
root@DietPi:~# cat /etc/apt/sources.list{,.d/*}
deb http://packages.dotdeb.org jessie all
deb-src http://packages.dotdeb.org jessie alldeb https://deb.debian.org/debian/ stretch main contrib non-free
deb https://deb.debian.org/debian/ stretch-updates main contrib non-free
deb https://deb.debian.org/debian-security/ stretch/updates main contrib non-free
deb https://deb.debian.org/debian/ stretch-backports main contrib non-free
deb https://packages.sury.org/php/ stretch main
deb https://packages.sury.org/php/ stretch main
deb https://deb.debian.org/debian/ bullseye main
deb https://deb.debian.org/debian/ bullseye main
deb http://fuzon.co.uk/meveric all c2
deb http://fuzon.co.uk/meveric all c2
deb http://fuzon.co.uk/meveric all main
deb http://fuzon.co.uk/meveric all main
deb http://fuzon.co.uk/meveric all testing
deb http://fuzon.co.uk/meveric all testing
deb http://fuzon.co.uk/meveric stretch backports
deb http://fuzon.co.uk/meveric stretch backports
deb http://fuzon.co.uk/meveric stretch main
deb http://fuzon.co.uk/meveric stretch main
deb http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
deb http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
# deb-src http://ppa.launchpad.net/ondrej/php/ubuntu focal main
deb https://packages.sury.org/php/ stretch main
deb http://deb.debian.org/debian/ unstable main
deb http://deb.debian.org/debian/ unstable main
deb https://download.webmin.com/download/repository sarge contrib
deb https://download.webmin.com/download/repository sarge contrib

MichaIng, 2020-02-20:
@drieksje 
Please paste result of the first command as well, which list the file names, so we know which files to delete or alter.

drieksje, 2020-02-20:
root@DietPi:~# ls -Al /etc/apt/sources.list{,.d/*}
-rw-r--r-- 1 root root 382 feb 19 22:14 /etc/apt/sources.list
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/dietpi-php.list
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/dietpi-php.list.save
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/dietpi-wireguard.list
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/dietpi-wireguard.list.save
-rw-r--r-- 1 root root  38 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-C2.list
-rw-r--r-- 1 root root  38 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-C2.list.save
-rw-r--r-- 1 root root  40 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-main.list
-rw-r--r-- 1 root root  40 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-main.list.save
-rw-r--r-- 1 root root  43 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-testing.list
-rw-r--r-- 1 root root  43 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-testing.list.save
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-backports.list
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-backports.list.save
-rw-r--r-- 1 root root  44 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-main.list
-rw-r--r-- 1 root root  44 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-main.list.save
-rw-r--r-- 1 root root 378 feb 19 21:41 /etc/apt/sources.list.d/ondrej-ubuntu-php-focal.list
-rw-r--r-- 1 root root 314 feb 19 21:41 /etc/apt/sources.list.d/ondrej-ubuntu-php-focal.list.save
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/php7.3.list
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/unstable.list
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/unstable.list.save
-rw-r--r-- 1 root root  66 feb 19 21:41 /etc/apt/sources.list.d/webmin.list
-rw-r--r-- 1 root root  66 feb 19 21:41 /etc/apt/sources.list.d/webmin.list.save

MichaIng, 2020-02-20:
@drieksje 
Okay, lets clean this up:
```
# Remove wrong or dangerous repos
rm -vf /etc/apt/sources.list.d/{ondrej-ubuntu-php-focal,php7.3,unstable}.list*
# Regenerating correct default repo, since you /etc/apt/sources.list itself has the wrong dotdeb.org repo, even with wrong version jessie
/DietPi/dietpi/func/dietpi-set_software apt-mirror default
G_AGUP
apt show php7.3
```
The last command should now show a valid available package. You should then be able to rerun the install:
```
dietpi-software install 40
```

drieksje, 2020-02-20:
### Details:
- Date           | Thu 20 Feb 15:11:39 CET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET
- Distro         | bullseye (ID=6)
- Command        | G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbs
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in
W: Target Packages (main/binary-all/Packages) is configured multiple times in /e
E: Package 'php7.3-fpm' has no installation candidate
E: Package 'php7.3-curl' has no installation candidate
E: Package 'php7.3-gd' has no installation candidate
E: Package 'php7.3-mbstring' has no installation candidate
E: Package 'php7.3-xml' has no installation candidate
E: Package 'php7.3-zip' has no installation candidate
E: Package 'php7.3-mysql' has no installation candidate
```

MichaIng, 2020-02-20:
@drieksje 
Still something wrong, it seems, please paste again the output of these two commands:
```
ls -Al /etc/apt/sources.list{,.d/*.list}
cat /etc/apt/sources.list{,.d/*.list}
```

drieksje, 2020-02-20:
ls -Al /etc/apt/sources.list{,.d/*.list}
-rw-r--r-- 1 root root 300 feb 20 15:05 /etc/apt/sources.list
-rw-r--r-- 1 root root  48 feb 19 21:41 /etc/apt/sources.list.d/dietpi-php.list
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/dietpi-wireguard.list
-rw-r--r-- 1 root root  38 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-C2.list
-rw-r--r-- 1 root root  40 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-main.list
-rw-r--r-- 1 root root  43 feb 19 21:41 /etc/apt/sources.list.d/meveric-all-testing.list
-rw-r--r-- 1 root root  49 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-backports.list
-rw-r--r-- 1 root root  44 feb 19 21:41 /etc/apt/sources.list.d/meveric-stretch-main.list
-rw-r--r-- 1 root root  66 feb 19 21:41 /etc/apt/sources.list.d/webmin.list

drieksje, 2020-02-20:
cat /etc/apt/sources.list{,.d/*.list}
deb https://deb.debian.org/debian/ bullseye main contrib non-free
deb https://deb.debian.org/debian/ bullseye-updates main contrib non-free
deb https://deb.debian.org/debian-security/ bullseye-security main contrib non-free
deb https://deb.debian.org/debian/ bullseye-backports main contrib non-free
deb https://packages.sury.org/php/ stretch main
deb https://deb.debian.org/debian/ bullseye main
deb http://fuzon.co.uk/meveric all c2
deb http://fuzon.co.uk/meveric all main
deb http://fuzon.co.uk/meveric all testing
deb http://fuzon.co.uk/meveric stretch backports
deb http://fuzon.co.uk/meveric stretch main
deb https://download.webmin.com/download/repository sarge contrib

MichaIng, 2020-02-20:
@drieksje 
Wow, okay it seems the unstable repo without pin priorities was already in place too long, now I see the distro has changed: `Distro | bullseye (ID=6)`
So some main system packages have been upgraded to unstable/testing/Bullseye, that is bad...

Do you have a `dietpi-backup` in place from before you experimented with PHP7.3 install? I mean you could leave it now on Bullseye, but this is the testing repo and you'll get a lot of APT package upgrades every day + major software version changes, hence some manual tinkering is required by times.

As general recommendation:
- Do not add/change entries in sources.list as long as you do not know exactly what you are doing.
- 3rd party sources (for a single software title) are mostly fine, but when adding additional full Debian repo suites, the system can be easily by accident completely upgraded to a new Debian version. And an unstable Debian version is nothing you want, besides for testing/development.
- And in case of SBCs the bigger problem is that the kernel/bootloader/firmware repositories might not be compatible with the new Debian version/libraries anymore. I think in case of Meverics Odroid repo it is okay for kernel/bootloader/firmware, but e.g. his Kodi and Chromium build, providing GPU acceleration, will most likely not work on Debian Bullseye...

drieksje, 2020-02-20:
Ok,
Have also seen that i'm on buelsey distro, so i have no backup of the
dietpi debian9, i will tray kodi and google browser,  i will let you know
if they still working!
Kind regards: dirk..... Thumbs up😎

MichaIng, 2020-02-20:
@drieksje 
If you want to stay with this, you could do further cleanup:
- Both, Ondrejs PHP repo and the dedicated Bullseye source for WireGuard are not required anymore, since Bullseye repo serves all this already:
```
rm /etc/apt/sources.list.d/dietpi-php.list*
rm /etc/apt/preferences.d/dietpi-openssl
rm /etc/apt/sources.list.d/dietpi-wireguard.list*
rm /etc/apt/preferences.d/dietpi-wireguard
```
- Meveric does not serve his repo for Bullseye yet, but at least for Buster, hence you should alter the older Stretch entries:
```
mv /etc/apt/sources.list.d/meveric-{stretch,buster}-backports.list
mv /etc/apt/sources.list.d/meveric-{stretch,buster}-main.list
sed -i 's/stretch/buster/' /etc/apt/sources.list.d/meveric-buster-backports.list
sed -i 's/stretch/buster/' /etc/apt/sources.list.d/meveric-buster-main.list
```

MichaIng, 2020-03-11:
Okay I mark this issue as closed, feel free to reopen if required.

closed 2020-03-11 by MichaIng