MichaIng/DietPi #3388: Updating Radarr, Sonarr, SABnzbd

19eighties, 2020-02-23:
### Creating a software update request

Hi, I'm wondering if these software selections (Radarr, Sonarr, SABnzbd) could be updated to the latest versions in the next DietPi update since it has been some time since they were updated. Thanks.

MichaIng, 2020-02-24:
@19eighties 
Sonarr, Radarr and Lidarr are always installed with newest upstream version. Sonarr can be upgraded via APT: `apt update && apt update` It's the nzbdrone package.
Radarr and Lidarr can be upgraded via internal web UI updater. However, currently the service does not restart afterwards, hence you need to run `systemctl start radarr` once done. This is fixed for v6.29: #3349

SABnzbd has no internal updater? What works in every case is a reinstall: `dietpi-software reinstall 139`
NB: Reinstalls can be done in every case to repair or update software. Check the list and IDs of your installed software: `dietpi-software list | grep ' =2 '`

19eighties, 2020-03-01:
Thanks for the info, it will be a good reference for future updates.

closed 2020-03-01 by 19eighties