MichaIng/DietPi #3389: Chromium Autostart / Kiosk Mode

kmplngj, 2020-02-23:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

buster

Linux magicmirrordisplay 4.19.93+ #1290 Fri Jan 10 16:34:37 GMT 2020 armv6l GNU/Linux

RPi Zero W (armv6l)

#### Additional Information (if applicable)
- Chromium Autostart / Kiosk Mode

#### Steps to reproduce
- After correct shutdown on the next start Chromium asked to "restores pages"

#### Expected behaviour
- Never ask to restore pages. I could not interact with my smart mirror.

#### Actual behaviour
- Shows "restore pages" dialog

#### Extra details
There are some points that could be done when my search results are correct. But this did not worked for me.

- https://raspberrypi.stackexchange.com/questions/68734/how-do-i-disable-restore-pages-chromium-didnt-shut-down-correctly-prompt

I changed /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh 

`# - Add URL for first run:
URL=$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=' /DietPi/dietpi.txt | sed 's/^[^=]*=//$
CHROMIUM_OPTS+=" --homepage $URL"`

`# - Add URL for first run:
URL=$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=' /DietPi/dietpi.txt | sed 's/^[^=]*=//$
CHROMIUM_OPTS+=" --app $URL"`

MichaIng, 2020-02-23:
@kmplngj 
Many thanks for your report.

Jep that is a known limitation with the kiosk mode, which is not 100% non-interactive. The app mode is indeed the solution but one needs to take care to not mix up the xinit arguments with Chromium arguments and take care the syntax. Let me see where we discussed/tested this...

... https://github.com/MichaIng/DietPi/issues/2938#issuecomment-506393195

So it needs to be: `--app=$URL`

kmplngj, 2020-03-15:
Hi,

I'm not 100% sure if this really helped or the fact that I made the preferences readonly.

I have another thing. The Browser always displays the failed update messages. To disable this `--disable-component-update` should be added to the command line. Perhaps you want to fix this. 

Thanks
Jan

MichaIng, 2020-03-15:
@kmplngj 
Thanks for reporting. Using `--app=$URL` instead of `--kiosk` does not hide this?
However instead of hiding it, it would be better to check why the update fails. If you made parts of it readonly, it is of course expected and in case of a browser probably not the best idea do block updates. If Chromium is run as unprivileged user, then it hopefully does not even attempt to update itself.

kmplngj, 2020-03-21:
Using `--app=$URL` does not hide the update notice. It seems even `--disable-component-update` does not do it. 😞

MichaIng, 2020-03-21:
@kmplngj 
Okay that is bad, then we need to go through the available options again: https://peter.sh/experiments/chromium-command-line-switches/

docgalaxyblock, 2020-04-24:
@kmplngj I'm also building one ^^ (and also get this problem)
@MichaIng I found a solution at this guide http://werner.rothschopf.net/201501_raspberry_kioskmode.htm
He uses this tags:
chromium-browser --noerrdialogs --disable-translate --kiosk --incognito
and it works fine

I suggest to implement a list were you can check different tags and a "custom tags" selection (were you can put in the tags that yourself need from your last link) when at the dietpi-autostart chromium entry

Greetings from Germany ^^

docgalaxyblock, 2020-04-24:
So maybe a menu entry like this
--------------Chromium tags selection----------------------
Reference: https://peter.sh/experiments/chromium-command-line-switches/
Default: CHROMIUM_OPTS="--kiosk --noerrdialogs --test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"

     _____________________________________________________________________________

             Cancle                                                       Ok

MichaIng, 2020-04-25:
@docgalaxyblock 
Makes sense, there are quite many options available for different use cases. I'll see if I find some time for v6.30 to implement such, meanwhile anyone is welcome do work on this of course 😉.

kmplngj, 2020-05-04:
I'm still getting the update error dialog in Chromium with the parameters "chromium-browser --noerrdialogs --disable-translate --kiosk --incognito". :/

Joulinar, 2020-05-04:
did you tried using `--app` instead of `--kiosk?`?

docgalaxyblock, 2020-05-04:
I'm using this without any error dialogs:
CHROMIUM_OPTS="--kiosk --incognito --noerrdialogs --test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"

MichaIng, 2020-05-07:
@docgalaxyblock @kmplngj 
Many thanks for your info guys. As of my testing [here](https://github.com/MichaIng/DietPi/issues/2938#issuecomment-506393195) sadly the `--kiosk --noerrdialogs` does not make it 100% non-interactive, as there are other prompts possible that are no errors. I guess `--app --noerrdialogs` would then be the way to go it one has no mouse attached.

kmplngj, 2020-09-30:
Any news on this?

I tried a DietPi fresh install. Does not work. I tried changing to -app --noerrdialogs. Also no luck. Chrome is showing the failed update message.

MichaIng, 2020-09-30:
Okay so `--kiosk --noerrdialogs` as you tested before does not work, `--app --noerrdialogs` does not, I'm out of ideas how get it silenced. Also seems like each Chromium behaves a bid different...

I found `--disable-component-update` to probably disable the update attempt? It is not about Chromium itself I guess since those APT-packaged builds have internal updater disabled in favour of APT-based updates?

kmplngj, 2020-10-04:
I found that this is a bug in the Chromium version that is used by DietPi.  `--noerrdialogs` and -`-disable-component-update` are buggy.

I get this _fixed_ with `--check-for-update-interval=31536000`

Best regards
Jan

MichaIng, 2020-10-04:
> --check-for-update-interval=31536000

Great to have a solution for that. Probably `--check-for-update-interval=0` might work as well to disable update checks?

There is not really a Chromium version that DietPi uses, there is [chromium-browser](http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/) provided by the RPi repository that we pull on RPi and there is [chromium](https://packages.debian.org/chromium) from the Debian repository for all other cases and of course the version depends on the underlying Raspbian/Debian version as well.

The problem is, from what I found in general with Chromium, is that there is this huge number of command line options, that is nowhere really officially documented and changes from version to version and seems to be different between builds and platforms. So I would call that more an issue in the way how the Chromium project in general handles those command options 😄. Sometimes more is less in combination with a clear documentation that can be relied on and change logs which contain early information when an option has been deprecated, when it will be removed and about the replacement in case 😎.

Joulinar, 2020-10-05:
Hi,

Maybe you like to have to our forum https://dietpi.com/phpbb/viewtopic.php?p=27145#p27145

There I have spend some time to find a way 😉 

`sudo touch /etc/chromium-browser/customizations/01-disable-update-check;echo CHROMIUM_FLAGS=\"\$\{CHROMIUM_FLAGS\} --check-for-update-interval=31536000\" | sudo tee /etc/chromium-browser/customizations/01-disable-update-check`

ovaar, 2020-11-27:
I found that adding the following lines to `/var/lib/dietpi/dietpi-autostart/chromium-autostart.sh` helped. Taken from the dietpi forum: https://dietpi.com/phpbb/viewtopic.php?p=23969#p23969

```sh
# Reset chromium state before xinit start.
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' ~/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"Normal"/' ~/.config/chromium/Default/Preferences

xinit $FP_CHROMIUM $CHROMIUM_OPTS
```

MichaIng, 2020-11-27:
A nice workaround as well, many thanks for sharing.