MichaIng/DietPi #3390: pihole error

msongz, 2020-02-23:
```
[  OK  ] DietPi-Software | Comment in /etc/lighttpd/conf-available/99-dietpi-pihole.conf converted to setting: server.error-handler-404 = "/html/pihole/index.php"
/DietPi/dietpi/dietpi-software: line 8748: cd: /var/www/html/admin: No such file or directory
Failed to enable unit: Unit file pihole-FTL.service does not exist.
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable dhcpcd
Removed /etc/systemd/system/multi-user.target.wants/dhcpcd.service.
Removed /etc/systemd/system/dhcpcd5.service.
/DietPi/dietpi/dietpi-software: line 8760: pihole: command not found
/DietPi/dietpi/dietpi-software: line 8763: pihole: command not found
/DietPi/dietpi/dietpi-software: line 8766: pihole: command not found
[ SUB1 ] DietPi-Services > dietpi_controlled
```

 clean install pi hole on rpi 4

MichaIng, 2020-02-23:
@msongz 
Many thanks for your report.

Obviously something during Pi-hole installer script went wrong before already. Please try to run it manually first: `curl -sSL https://install.pi-hole.net | bash`

msongz, 2020-02-24:
@MichaIng thanks! 
after installing pihole by this method,
pihole blocks aria2 web page `http://pi4/aria2/docs`

Joulinar, 2020-02-24:
Hi,
are you sure PiHole is blocking `pi4` DNS query? Because `pi4` isn't a valid DNS name that can be resolved by PiHole and the upstream DNS provider. Therefore you would need to ensure that `pi4` is an available entry in `/etc/hosts`. Otherwise PiHole will send the request to an upstream DNS provider and for sure, they don't know what `pi4` is.

msongz, 2020-02-24:
@Joulinar sorry, not clear about it.
i preset my pi4's ip address to 'pi4'.

MichaIng, 2020-02-24:
Now that this worked, rerun the install with dietpi-software. The installer will rerun however now the things that it previously failed for exist, so it should succeed. Aria2 fails currently ecpectidly since Pi-hole installer overrides the Lighttpd config and switches the webroot from /var/www to /var/www/html.

Ah probably it is required to reinstall Lighttpd first, forgot about the --no-install-webserver option...

dietpi-software reinstall 83
dietpi-software install 93

MichaIng, 2020-03-11:
Btw I observed as well Pi-hole installer failing without error message during the internal APT installs in some cases. I am not sure about the reason, `debconf-apt-progress` is used which hides the install process behind whiptail menus, but whenever some interactive input is required it exits, hence the installer fails. But I as well ran into a failure during `resolvconf` install (pre-installed on DietPi by default, but not on my testing system), which was completely noninteractive. A bid strange. However Pi-hole v5 installer does not use `debconf-apt-progress` anymore which solves such kind of issues.

msongz, 2020-03-15:
root@pi4  ~  cat /etc/apt/apt.conf.d/*                                                                                                                                                                 ✔  ⚡  138  02:52:27
// DO NOT EDIT! File autogenerated by /etc/kernel/postinst.d/apt-auto-removal
APT::NeverAutoRemove
{
};
/* Debug information:
# dpkg list:
# list of installed kernel packages:

# list of different kernel versions:

# Installing kernel:  (4.19.97-v8+)
# Running kernel: ignored (4.19.75-v7l+)
# Last kernel:
# Previous kernel:
# Kernel versions list to keep:

# Kernel packages (version part) to protect:
4\.19\.75-v7l\+
4\.19\.97-v8\+
*/
// Pre-configure all packages with debconf before they are installed.
// If you don't like it, comment it out.
DPkg::Pre-Install-Pkgs {"/usr/sbin/dpkg-preconfigure --apt || true";};
Acquire::Languages "none";
# Please use "dietpi-config" > "Advanded Options" > "APT Cache" to edit these settings.
Dir::Cache "/tmp/apt";
Dir::Cache::archives "/var/cache/apt/archives";
Dir::Cache::pkgcache "";
Dir::Cache::srcpkgcache "";
Dpkg::options {
   "--force-confdef";
   "--force-confold";
}
Acquire::ForceIPv4 "true";
APT::Install-Recommends "false";
APT::Install-Suggests "false";
APT::AutoRemove::RecommendsImportant "false";
APT::AutoRemove::SuggestsImportant "false";

msongz, 2020-03-15:
```
Initialised database
 Reading database
 Checking network connectivity: ping -c 1 -W 10 9.9.9.9
 If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
 Checking network connectivity: ping -c 1 -W 10 9.9.9.9
 If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
 Checking network connectivity: ping -c 1 -W 10 9.9.9.9
 If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
 Checking network connectivity: ping -c 1 -W 10 9.9.9.9
 If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
```

Joulinar, 2020-03-15:
Can you tr to ping some other IP address?

MichaIng, 2020-05-18:
Marking as closed, the new Pi-hole v5 installer includes vast changes. Feel free to reopen if required.

closed 2020-05-18 by MichaIng