MichaIng/DietPi #3391: Pi-hole v5 needs 'www-data' user added to 'pihole' group

maartenlangeveld, 2020-02-24:
Just testing Pihole v5 (beta), in order to add domains (blacklist, whitelist) through pihole's web interface, user 'www-data' needs to be part of 'pihole' group.

MichaIng, 2020-02-24:
@maartenlangeveld 
Many thanks for your report. Yes indeed there is also some more testing and tweaking required/possible with v5. I'll do that the next days, the new `pihole` group is already good to know.

maartenlangeveld, 2020-02-24:
Hi Michael,

With pleasure.

Have also tested your new NanoPi NEO2 buster dietpi image but cannot get my NanoHatOLED working... Seems that I2C is not supported in the Buster build (works well with strech build). Is there a solution?

Many thanks & kind regards,

Maarten

On Mon, 24 Feb 2020 14:34:40 +0100, MichaIng <notifications@github.com> wrote:

> @maartenlangeveld
> Many thanks for your report. Yes indeed there is also some more testing and tweaking required/possible with v5. I'll do that the next days, the new `pihole` group is already good to know.

MichaIng, 2020-02-25:
@maartenlangeveld 
Many thanks for testing. Could you please try to edit `/boot/armbianEnv.txt` and add:
`i2c0` to the overlays line, so it should look like this:
```
overlays=usbhost1 usbhost2 i2c0
```
There is also an `i2c1` overlay available, not sure which one uses which pins.
And if you don't use Docker, add the following as new line as well, which should lead to minor performance enhancements:
```
docker_optimizations=off
```

maartenlangeveld, 2020-02-25:
@MichaIng 

That works! Thanks again for your truly great support !!

Joulinar, 2020-03-02:
just for reference regarding PiHole v5

https://discourse.pi-hole.net/t/error-message-attempt-to-write-a-readonly-database/27092/17

`usermod -a -G pihole www-data`

MichaIng, 2020-03-02:
@Joulinar 
Jep many thanks. Will run some further tests the next days, I think there are more things that should be adjusted for Pi-hole v5 compatibility.

MichaIng, 2020-03-07:
Done: https://github.com/MichaIng/DietPi/commit/b90b49fe701546da69f87b10de6f70e6009b5cb0
Changelog: https://github.com/MichaIng/DietPi/commit/d5a44d0151fa1ccec311be9a54418d0013d95d2b

Joulinar, 2020-03-08:
@MichaIng 
I guess it will impact new installation only? Right?
What able existing installation? Should we do anything on it as well?

MichaIng, 2020-03-08:
@Joulinar 
Good point. We can check on update if dietpi group exists and in case add www-data user to it.

MichaIng, 2020-03-11:
Pi-hole v5 as well depends on PHP intl module now: https://github.com/MichaIng/DietPi/commit/c18f6c66d9ab24f6259faaf04fb85b21c91b2c04

MichaIng, 2020-03-11:
v5 compatibility guaranteed with next DietPi-Update: https://github.com/MichaIng/DietPi/commit/9dc959687c4a9eaf1c46a7716809f3897d842f3a

Joulinar, 2020-03-13:
@MichaIng 
I tested 2 scenario on my Buster VM

1. upgrade from v6.28 to v6.29 with an already existing PiHole installation
2. install new PiHole on v6.29

Both are working fine and PiHole 5 was working ootb after upgrade.

MichaIng, 2020-03-13:
@Joulinar
That is great, many thanks for testing.

mistakenideas, 2020-05-03:
Tested upgrading from DietPi 6.28 to 6.29 on RaspberryPi B Rev 1. Only really running Pi-Hole 4 on this which I then [switched to the Pi-Hole 5 Beta branch](https://pi-hole.net/2020/01/19/announcing-a-beta-test-of-pi-hole-5-0/). All seems to be running okay so far.

MichaIng, 2020-05-03:
@mistakenideas 
Many thanks for testing ❤️.

closed 2020-05-03 by MichaIng