MichaIng/DietPi #3392: Plex Media Server issue

adamr20, 2020-02-24:
#### Details:
- Date           | Mon 24 Feb 22:25:25 GMT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... Attempting to update PMS via apt update upgrade

#### Actual behaviour:

- ... encountering below issue

#### Extra details:

- ...

#### Additional logs:
```
(Reading database ... 42706 files and directories currently installed.)
Preparing to unpack .../plexmediaserver_1.18.7.2457-77cb9455c_armhf.deb ...
PlexMediaServer install: Pre-installation Validation.  
PlexMediaServer install: ERROR:  Multiple configuration files found in \'/etc/systemd/system/plexmediaserver.service.d\'.  Please correct and install again.  
PlexMediaServer install:         Files found are: /etc/systemd/system/plexmediaserver.service.d/dietpi-group.conf
/etc/systemd/system/plexmediaserver.service.d/dietpi-process_tool.conf  
PlexMediaServer install:         Please correct this error and try again.  
dpkg: error processing archive /var/cache/apt/archives/plexmediaserver_1.18.7.2457-77cb9455c_armhf.deb (--unpack):
 new plexmediaserver package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/plexmediaserver_1.18.7.2457-77cb9455c_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2020-02-25:
Hi,

pls can you have a look to directory `/etc/systemd/system/plexmediaserver.service.d`. Looks like you have 2 config files but it should be just 1.

MichaIng, 2020-02-25:
Is there a general issue when having two drop-in configs? One is created during install to run Plex as dietpi user to grant media and download dir access + assure other media programs have access to Plex created files again. The other config is from `dietpi-services` when adjusting process priorities/governor policies/nice.

MichaIng, 2020-02-25:
... `/var/lib/dpkg/info/plexmediaserver.preinst`
```
    # if there is a lingering /etc/systemd/system/plexmediaserver.service, this is wrong.  Flag the PlexUser to correct
    if [ -e /etc/systemd/system/plexmediaserver.service ]; then

      Output user "SystemD configuration error detected:"
      Output user "File \"/etc/systemd/system/plexmediaserver.service\" is leftover from an earlier Debian/Ubuntu bug."
      Output user "Please transfer any customizations \(APP_DIR, User, Group, TMPDIR, and/or UMask\) to a proper \"$OverrideDir/override.conf\""
      Output user "and reinstall."
      Output both "Unit override '/etc/systemd/system/plexmediaserver.service' found.  Only '/etc/systemd/system/plexmediaserver.service.d/override.conf' is supported."
      Fail=1
      fi

      # Scan this sector
      OverrideDir="/etc/systemd/system/plexmediaserver.service.d"

      # Validate OverrideDir contents if it exists
      if [ -d $OverrideDir ]; then

      # Examine all possible (should only be one but be prudent)
      Count=$(find $OverrideDir -name \*.conf -print | wc -l)
      if [ $Count -eq 0 ]; then
        Output user "INFO: $OverrideDir exists but no override 'conf' files found. Ignored."
        OverrideDir=""
        HaveOverride=0

      elif [ $Count -gt 1 ]; then
        Output both "ERROR:  Multiple configuration files found in \'$OverrideDir\'.  Please correct and install again."
        Output user "        Files found are: $(ls -1 $OverrideDir/*.conf)"
        Output user "        Please correct this error and try again."
        Fail=1

      else
        # Lock phasers on target
        OverrideFile="$(find $OverrideDir -name \*.conf -print)"
        HaveOverride=1

      fi
    fi
```
What the hack, no existing service file allowed, only `override.conf` as drop-in allowed. All these checks do not make any sense. The existing "file" could be a mask, which might be wanted by admins which want to start it manually outside of systemd. `override.conf` or another name or multiple ones all have the same effect, being able to break Plex start (if wrong/invalid settings are contained) or working well with desired effect (in case of the two we add). Limiting things to this single fixed named drop-in config has hence no rational reason but only limits admins needs/flexibility. Bad that it is closed source, hence no quick PR to the source code possible.

MichaIng, 2020-03-09:
Issue will be fixed with PMS v1.19: https://forums.plex.tv/t/plex-media-server-linux-installation-packaging-update-issues/529917/187?u=michaing
It still handles those override configs, we have to see what it does to `dietpi` group membership.

MichaIng, 2020-03-23:
Quick link to track releases:
- Forum: https://forums.plex.tv/t/plex-media-server/30447/324
- Download: https://www.plex.tv/media-server-downloads/

MichaIng, 2020-04-22:
PMS v19 has been released, issue solved.

closed 2020-05-03 by MichaIng