MichaIng/DietPi #3393: HWClock support

Gokujo, 2020-02-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
On earlier versions there was an setting option for hardware clock. Now I am updating an RPi3B+ with hwclock and it is realy nesseccary for deploying. And it would be great if I could controll it with console (update the time on it, update time on device, ...)
`    sudo ntpdate-debian -u 8.8.8.8 >> /home/.../logs/ntpdate.log && sudo hwclock --set --date="$(date "+%m/%d/%y %H:%M:%S")"`
something like this

MichaIng, 2020-02-25:
@Gokujo 
Many thanks for your request.

Actually a real RTC is always supported, so is the `hwclock` always available.
- The `dietpi-config` option you mention only shows up if the `hwclock` succeeds, hence there is at least an RTC connector available that is recognised by the system.
- The option only allowed to purge/reinstall the `fake-hwclock` package, which sets system clock to what it stored on last shutdown (respectively periodically every hour).
- So as fast as you attached an RTC that is recognised by the system, `hwclock` must not print a failure, which leads to `dietpi-config` showing the option to purge `fake-hwclock`. Or you do it manually: `apt purge fake-hwclock`

And for network sync btw `ntpdate` is deprecated: https://packages.debian.org/buster/ntpdate
Do not use it anymore but use `/DietPi/dietpi/func/run_ntpd 1` instead if you want to do a manual re-sync. But this is done periodically according to your chosen dietpi-config/dietpi.txt setting.

MichaIng, 2020-05-18:
Marking as closed, I hope my answer explains everything enough. Feel free to reopen if required.

closed 2020-05-18 by MichaIng

ChristopheH22, 2021-03-14:
Hello,

I'm facing a similar issue. On dietpi 6.34 I had access to the menu to set a hardware clock. It is not the case anymore on DietPi 6.35/7.0.

My config : Raspberry Pi Zero W + DietPi 6.35 + DS3231

Based on your explanation MichaIng, I tried the command hwclock but it is not working (but was ok on 6.34) :
_root@DietPi:~# hwclock --verbose
hwclock from util-linux 2.33.1
System Time: 1615726546.837390
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method._

The device DS3231 is correctly detected on I2C bus:
_root@DietPi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --_

Do you know what could be the root cause of that ?

MichaIng, 2021-03-14:
Since the `util-linux` that ships `hwclock` didn't get an update recently, obviously the hardware clock interface has changed or got broken, most likely because of the kernel upgrade from 5.4 to 5.10.
Is the related kernel module loaded? `lsmod`

Since I have no experience with I2C-attached hardware clocks, please try to repeat the install steps for it.

You could try to downgrade the kernel back to v5.4:
```
apt install rpi-update
rpi-update b7c8ef64ea24435519f05c38a2238658908c038e
```

ChristopheH22, 2021-03-14:
Thank you MichalIng for your answer.
Here is the output of lsmod command:
[image]

I will try your recommendation to downgrade to 5.4.
Other question, is it possible to come back to 6.34 DietPi install ? I'm afraid not since if I use the 6.34 IMG file, it will proceed to an update automatically.

Joulinar, 2021-03-14:
There is no way to downgrade DietPi. As well, new installation will be updated to latest version.

MichaIng, 2021-03-14:
The DietPi update (i.e. DietPi scripts and services) is not the issue here, hence that won't help anyway. Better to find out which system change (kernel or other APT package) caused the issue.

AFAIK all RTCs come with a related kernel module, but I cannot see one loaded in your case. But you added the device tree overlay via `dtoverlay=i2c-rtc,ds3231` in `/boot/config.txt` right? That should be actually sufficient and invoke the module automatically.
<details>
<summary>`rtc-ds1307` should be it:</summary>

```
2021-03-14 20:40:59 root@micha:/mnt/sda# modinfo rtc-ds1307
filename:       /lib/modules/5.10.17-v7+/kernel/drivers/rtc/rtc-ds1307.ko
license:        GPL
description:    RTC driver for DS1307 and similar chips
srcversion:     9D261CFE626ECA3F8AC0FF6
alias:          i2c:rx8130
alias:          i2c:isl12057
alias:          i2c:rx8025
alias:          i2c:pt7c4338
alias:          i2c:mcp7941x
alias:          i2c:mcp7940x
alias:          i2c:m41t11
alias:          i2c:m41t00
alias:          i2c:m41t0
alias:          i2c:ds3231
alias:          i2c:ds1341
alias:          i2c:ds1340
alias:          i2c:ds1388
alias:          i2c:ds1339
alias:          i2c:ds1338
alias:          i2c:ds1337
alias:          i2c:ds1308
alias:          i2c:ds1307
alias:          of:N*T*Cepson,rx8130C*
alias:          of:N*T*Cepson,rx8130
alias:          of:N*T*Cisil,isl12057C*
alias:          of:N*T*Cisil,isl12057
alias:          of:N*T*Cepson,rx8025C*
alias:          of:N*T*Cepson,rx8025
alias:          of:N*T*Cpericom,pt7c4338C*
alias:          of:N*T*Cpericom,pt7c4338
alias:          of:N*T*Cmicrochip,mcp7941xC*
alias:          of:N*T*Cmicrochip,mcp7941x
alias:          of:N*T*Cmicrochip,mcp7940xC*
alias:          of:N*T*Cmicrochip,mcp7940x
alias:          of:N*T*Cst,m41t11C*
alias:          of:N*T*Cst,m41t11
alias:          of:N*T*Cst,m41t00C*
alias:          of:N*T*Cst,m41t00
alias:          of:N*T*Cst,m41t0C*
alias:          of:N*T*Cst,m41t0
alias:          of:N*T*Cmaxim,ds3231C*
alias:          of:N*T*Cmaxim,ds3231
alias:          of:N*T*Cdallas,ds1341C*
alias:          of:N*T*Cdallas,ds1341
alias:          of:N*T*Cdallas,ds1340C*
alias:          of:N*T*Cdallas,ds1340
alias:          of:N*T*Cdallas,ds1388C*
alias:          of:N*T*Cdallas,ds1388
alias:          of:N*T*Cdallas,ds1339C*
alias:          of:N*T*Cdallas,ds1339
alias:          of:N*T*Cdallas,ds1338C*
alias:          of:N*T*Cdallas,ds1338
alias:          of:N*T*Cdallas,ds1337C*
alias:          of:N*T*Cdallas,ds1337
alias:          of:N*T*Cdallas,ds1308C*
alias:          of:N*T*Cdallas,ds1308
alias:          of:N*T*Cdallas,ds1307C*
alias:          of:N*T*Cdallas,ds1307
depends:        regmap-i2c
intree:         Y
name:           rtc_ds1307
vermagic:       5.10.17-v7+ SMP mod_unload modversions ARMv7 p2v8
```

ChristopheH22, 2021-03-15:
Hi MichaIng,

You are right, the line `dtoverlay=i2c-rtc,ds3231` was missing. I'm not sure to understand what happened on my side. 
Maybe I didn't proceed to a reboot after changing the RTC option from "fake-hwclock" to "hwclock" and then, each time I sent the command "hwclock" the answer came from the fake hwclock and not the real one...
After upgrading to 7.0 (with reboot requested) then the "hwclock" failed because the chip was not configured in the file `/boot/config.tx`

Do you think it would be a good idea for future release to :
- Propose to reboot the system when options like this one are changed. 
- Include a menu to choose the RTC hardware added to the SBC in order to automatically configure the file `/boot/config.tx`

Thanks a lot for your help.

MichaIng, 2021-03-15:
> Do you think it would be a good idea for future release to

Only both together would make sense.

The hardware/emulated RTC option does nothing else than purging/reinstalling `fake-hwclock` and it is only shown when running `hwclock` "succeeds", i.e. does not return an error code. However, that `hwclock` succeeds does not mean that there is actually a functional RTC installed, it already does when only an RTC port is present, sadly. So we have no really a change to know whether there is an RTC, but only when it's definitely not there, degrading the menu option to "if you know that you have a functional RTC, select that fact here to have the obsolete fake-hwclock removed".

A reboot is only required when changing device tree overlays, so what we could do is add a menu that shows the known RTC models, supported by the `i2c-rtc` overlay, and if one is selected, `fake-hwclock` could be removed automatically and a reboot offered. Although, the RTC's only purpose is to the the initial system time at boot, and well, then you obviously have rebooted anyway, so the overlay becomes active as fast as it serves any purpose, making the offered reboot obsolete? 😃

ChristopheH22, 2021-03-15:
Thank you MichaIng for the information.