MichaIng/DietPi #3394: Autosetup failed

Gokujo, 2020-02-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.28
- Distro version | 10.3 (Buster)
- Kernel version | Linux ****** 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)

#### Additional Information (if applicable)
[config.txt](https://github.com/MichaIng/DietPi/files/4249351/config.txt)

#### Actual behaviour
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/4249353/dietpi-firstrun-setup.log)
[dietpi-boot.log](https://github.com/MichaIng/DietPi/files/4249360/dietpi-boot.log)

#### Extra details
Some issues that I noticed additional to this situation:

- It takes to long on boot with postboot options. Idk what should I do to disable this
[image]
It can take up to 5 minutes for postbooting - not a good option

- I tried to startup into desktop with dietpi user but it doesn't have rights to access /dev/tty1

- I set resolution 1280 x 720 in config.txt as well as in dietpi.txt on first run but i got a resolution 800x...

Joulinar, 2020-02-25:
Hi,

best to my knowledge, `autostart` is working with `root` user only atm. Pls can you check using `dietpi-config` to change/adjust resolution.

Gokujo, 2020-02-25:
> Hi,
> 
> best to my knowledge, `autostart` is working with `root` user only atm. Pls can you check using `dietpi-config` to change/adjust resolution.

[My older issue](https://github.com/MichaIng/DietPi/issues/3159#issuecomment-542267859)
@MichaIng said it should be set to an user. now I am using root-user

MichaIng, 2020-02-25:
Autostart works with every user, but as mentioned in the input screen, you need to give it sufficient permissions for now. We plan to automate this, so the selected user automatically gets required permissions to execute the chosen autostart program: #3351

In `config.txt`:
```
#framebuffer_width=1280
#framebuffer_height=720
```
@Gokujo is this your current content? In case you edited the numbers but did not uncomment/activate the settings 🙂:
```
framebuffer_width=1280
framebuffer_height=720
```
Yes please use `dietpi-config` for this, so it's assured to be set correctly in all required places.

In your `dietpi-boot.log` systemd-timesyncd times out. Please try to set this to your local router (dietpi-config > Network Options: Misc > NTP mirror > Gateway). This should be fastest as long as your router serves as NTP server for the local network and 99% do. Sync should not take more than 2 seconds.

If DietPi-Postboot again takes much longer, then most likely some service start hangs.

About the autostart user:
- Generally root works in every case of course
- A custom user requires manual permissions grant currently
- But I remember your issue with the MATE desktop and was able to replicate and workaround. By default the MATE desktop does not fully initiate (for security reasons) when starting as root. At least desktop icons are hidden. This can be worked around with `caja --force-desktop`, but since there seem to be good reasons to block it with root, it is better then to choose a non-root login user.

Gokujo, 2020-02-25:
@MichaIng my fault, after comparing with new config file I forgott to uncomment them. damn.

> dietpi-config > Network Options: Misc > NTP mirror 
I had to add a custom IP, but it needs 41 seconds to sync
[dietpi-boot.log](https://github.com/MichaIng/DietPi/files/4249719/dietpi-boot.log)
postboot took this time 200 seconds. I would know which proccess(es) is(are) hanging

Joulinar, 2020-02-25:
it's the NTP time sync taking 40 sec. We have seen this behaviour in the past as well but did not found the cause until now.

MichaIng, 2020-02-25:
@Gokujo 
If you run `/DietPi/dietpi/func/run_ntpd 1` manually from console, does it take that long as well?

Gokujo, 2020-02-25:
> @Gokujo
> If you run `/DietPi/dietpi/func/run_ntpd 1` manually from console, does it take that long as well?

no, only one second

Joulinar, 2020-02-25:
@MichaIng 
my observation until now, the ntp sync during boot is the only one taking that long. during daily operation it will run quite fast

MichaIng, 2020-02-25:
Okay @Joulinar indeed looks exactly like our test/report in forum. I was not able to replicate anywhere (or my RPi) so far 🤔.

It must have something to do with network init, but since `dietpi-boot` verifies the default route being up, I am not sure what is missing, taking THAT long.

If one of you wants to help debugging, you could edit `/DietPi/dietpi/boot` and replace `/DietPi/dietpi/func/run_ntpd 1` with:
```
until ping -c 1 0.debian.pool.ntp.org; do sleep 0.5; done
```
This pings the default Debian NTP pool but does no actual sync. The ping command as well outputs whether name resolution or final connection fails, hence we can somehow track the network init state by the outputs in `dietpi-boot.log`. If this succeeds immediately, then we know that it is not the network state but `systemd-timesyncd` itself.

Joulinar, 2020-02-25:
@MichaIng 
I set NTP server to my local `Gateway `and I have quite some differences between my RPi3B+ and my RPi4B. Both are behaving quite different. While my Pi4 is hanging, the Pi3 is running fine. However this picture is changing as well. Still strange topic.

MichaIng, 2020-02-25:
Hmm, first guess then would be that RPi4 Ethernet initiates different/slower then RPi3 one, but THAT long is unexpected of course. And Gokujo has the issue with RPi3+, so there must be a different reason. Since RPi images can be swapped you could try to boot RPi4 with the RPi3 SDcard vise versa to see if it indeed has something to do with any settings that we missed to check, or kernel/firmware version or such.

Joulinar, 2020-02-25:
I know it's a stupid question but where to find the fancy `dietpi-boot.log`

Joulinar, 2020-02-25:
My last boot was looking like this

```
root@DietPi4:~# systemd-analyze blame
         43.115s dietpi-boot.service
```

and now it's this
```
root@DietPi4:~# systemd-analyze blame
          5.372s dietpi-boot.service
```
 I don't get it at all 😕

MichaIng, 2020-02-25:
... edit of above:
Hmm, first guess then would be that RPi4 Ethernet initiates different/slower then RPi3 one, but THAT long is unexpected of course. And Gokujo has the issue with RPi3+, so there must be a different reason. Since RPi images can be swapped you could try to boot RPi4 with the RPi3 SDcard vise versa to see if it indeed has something to do with any settings that we missed to check, or kernel/firmware version or such, or how the router handles the devices.

E.g. when having DHCP enabled once, my fritzbox stores the hostname and the device in an internal list by its hostname.  If I set static IP right on first boot, the device is stored differently without its hostname. Probably some routers handle connected devices faster/different when they have retrieved their IP via DHCP initially and use this same IP afterwards as static IP?

____

The dietpi-boot.log is in /var/tmp/dietpi/logs/

Joulinar, 2020-02-25:
@MichaIng 
Do we like to stay in this topic for the NTP challenge? I mean we are hijacking the original topic 😄

MichaIng, 2020-02-25:
@Joulinar 
Okay, so NTP challenge is one thing, indeed makes sense to open a new issue for that to test/debug/discuss.

@Gokujo 
About the other hanging process:
- If it's related to boot but the service fails in every case, check the output of: `dietpi-services restart`
- Else, do you have a monitor to watch the boot output?
- Otherwise you should be able to derive it from `journalctl`. If a service times out, that usually produces a journal entry.
- Also `systemctl` should show the related service as failed.
- Which software do you have installed, actually? `dietpi-software list | grep ' =2 '`
- And are there any postboot.d scripts? `ls -Al /var/lib/dietpi/postboot.d/`

Gokujo, 2020-02-26:
due some experiments I noticed that on boot the wifi doesn't connect to internet but rasp says that connection was estabilished.

[Google Drive](https://drive.google.com/file/d/1CsKbvZO4u4oVHCnW_W7jQC-xvUHhcrwq/view?usp=sharing) Files are to big. But problem with with postboot remains

> If it's related to boot but the service fails in every case, check the output of: `dietpi-services restart`

``` bash
root@JAS-TT1:~# dietpi-services restart

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart

[  OK  ] DietPi-Services | restart : avahi-daemon
[  OK  ] DietPi-Services | restart : apache2
[  OK  ] DietPi-Services | restart : cron
root@JAS-TT1:~#
```

> Otherwise you should be able to derive it from journalctl. If a service times out, that usually produces a journal entry.

[journal.log](https://github.com/MichaIng/DietPi/files/4254059/journal.log)
It has issues with hardware clock

> Which software do you have installed, actually? `dietpi-software list | grep ' =2 '`

[software.log](https://github.com/MichaIng/DietPi/files/4254070/software.log)

> And are there any postboot.d scripts? `ls -Al /var/lib/dietpi/postboot.d/`

0

MichaIng, 2020-02-26:
@Gokujo 
```
Feb 26 08:20:43 JAS-TT1 sh[386]: wlan0=wlan0
Feb 26 08:20:45 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
Feb 26 08:20:45 JAS-TT1 sh[391]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
Feb 26 08:20:47 JAS-TT1 systemd-timesyncd[542]: Timed out waiting for reply from 192.168.0.10:123 (192.168.0.10).
Feb 26 08:20:52 JAS-TT1 dhclient[438]: No DHCPOFFERS received.
Feb 26 08:20:52 JAS-TT1 sh[391]: No DHCPOFFERS received.
Feb 26 08:20:52 JAS-TT1 sh[391]: No working leases in persistent database - sleeping.
Feb 26 08:20:52 JAS-TT1 dhclient[438]: No working leases in persistent database - sleeping.
Feb 26 08:20:52 JAS-TT1 sh[391]: eth0=eth0
Feb 26 08:20:56 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3
Feb 26 08:20:59 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
Feb 26 08:21:02 JAS-TT1 systemd[1]: systemd-fsckd.service: Succeeded.
Feb 26 08:21:06 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 1
Feb 26 08:21:07 JAS-TT1 dhclient[438]: No DHCPOFFERS received.
Feb 26 08:21:07 JAS-TT1 dhclient[438]: No working leases in persistent database - sleeping.
Feb 26 08:21:11 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3
Feb 26 08:21:14 JAS-TT1 dhclient[438]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
Feb 26 08:22:54 JAS-TT1 systemd-timesyncd[542]: Synchronized to time server for the first time 192.168.0.10:123 (192.168.0.10).
```
- As you can see, there is a timestamp shift once network time sync has completed. So `systemd-analyze` reports misleading results here. Best is to watch the boot process live on monitor so see if/where it really hangs.

However there are at best some conflicting network interfaces/services:
- WiFi and Ethernet are both enabled, WiFi with static IP and Ethernet with DHCP?
- Ethernet DHCP discovery fails and is re-attempted every few seconds. If there is no DHCP server on this network, assign a static IP to Ethernet.
- I could not reliably derive the actual device IP(s) from journal. However do you need both, Ethernet and WiFi enabled concurrently? Currently it seems Ethernet is not used, hence I would disable it. And not that you cannot have both connected to the same network without a custom routing table, else the system does not know where to send which request to.
- Also if I don't get it wrong, the static wpa_supplicant service is enabled, which is not required and conflicts with ifupdown. Please run: `systemctl disable wpa_supplicant`

Gokujo, 2020-02-26:
> WiFi and Ethernet are both enabled, WiFi with static IP and Ethernet with DHCP?

yes, only wifi is needed. I tried to disable and restart. time sync is still there but postboot disapeared

so, to main problem - autosetup. autoconfig worked well, but software installation couldn't be performed. I tried to full format the sd card and reinstall last image

MichaIng, 2020-02-26:
@Gokujo 
> but software installation couldn't be performed

What you mean by this? Initial `dietpi-software` run hang or failed somewhere? What as the error message?

Gokujo, 2020-02-26:
> What you mean by this?

try this config on first run (auto setup)
[dietpi.txt](https://github.com/MichaIng/DietPi/files/4255526/dietpi.txt)

i got a message that software couldn't be installed and I should reboot to retry
the [log file](https://github.com/MichaIng/DietPi/files/4249353/dietpi-firstrun-setup.log) I attached on topic start

MichaIng, 2020-02-26:
@Gokujo 
Which software couldn't be installed? I see everything got installed and configured correctly and the reboot is expected to have everything done. Any issues with service startup or autostart after reboot?

Gokujo, 2020-02-26:
> @Gokujo
> Which software couldn't be installed? I see everything got installed and configured correctly and the reboot is expected to have everything done. Any issues with service startup or autostart after reboot?

whole software on the list. I had to install it manually. Configuration has been applied without problems.

Joulinar, 2020-02-26:
Hi,

a couple of comments on the `dietpi.tx`t file

You are bypassing internet connection check by changing the IP to a local network device but you stay with the DNS check on the global Cloudflare DNS `one.one.one.one`. This doesn't make really sense as this is the same as pinging `1.1.1.1`

```
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=192.168.0.10
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```
I guess `11 `is an typo and incorrect value for `DietPi-Survey`
```
# DietPi-Survey: 1=opt in | 0=opt out | -1=ask on first call
SURVEY_OPTED_IN=11
```
It is recommended to also set `CONFIG_BOOT_WAIT_FOR_NETWORK=2` if you set `AUTO_SETUP_AUTOMATED=1`

MichaIng, 2020-05-18:
Marking as closed. Feel free to reopen if required.

closed 2020-05-18 by MichaIng