MichaIng/DietPi #3395: Checking network connectivity, inability to ping

seron, 2020-02-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.28.0 (MichaIng/master)
- Distro version | buster (ID=5)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | Laptop USB port
- SDcard used | Kingston CANVAS Select Plus 16 GB

#### Additional Information (if applicable)
- Software title | DietPi-Login
- Fresh installation of DietPi
- Automatic bugreport seems to fail too, 462d99eb-1378-4351-b4b0-87f9ee5f39e9

#### Steps to reproduce
logon to dietpi with default username and password

#### Expected behaviour
I'm not sure as this is the first time I try dietpi.

#### Actual behaviour
in the terminal:
[FAILED] DietPi-Login | Checking network connectivity: ping -c 1 -W 5 1.1.1.1

in the DietPi-Login | Error handler:
 Checking network connectivity

  - Command: ping -c 1 -W 5 1.1.1.1
  - Exit code: 1
  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:5
  - Image creator: DietPi Core Team
  - Pre-image: Raspbian Lite

 Log file content:
 PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

 --- 1.1.1.1 ping statistics ---
 1 packets transmitted, 0 received, 100% packet loss, time 0ms

#### Extra details
I do not expect a connectivity error as the Internet can be reached (files downloaded), but ping seems to not work from my location.

Joulinar, 2020-02-25:
@seron 

many thanks for your report.

Hi, if you are not able to `ping 1.1.1.1` you would have local issues as it's part of Cloudflares DNS Server. It is the globally available IP address for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change this, you can do it as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `8.8.8.8` and `google.com`

It's a similar issue like this #3337

MichaIng, 2020-02-25:
For reference, default has been changed for new images (not applied to all ones yet): https://github.com/MichaIng/DietPi/commit/ffd9431cba7a9f5fda48faec5edd7bc9a12e82ed

seron, 2020-02-26:
Thanks. It worked.

Will changes to `/DietPi/dietpi.txt` be saved permanently?

Joulinar, 2020-02-26:
Yes

closed 2020-02-26 by seron