MichaIng/DietPi #3396: Minio is out-of-date, ARM variants

GitHubGeek, 2020-02-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.28`
- Distro version | `buster`
- Kernel version | `Linux hoya 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B (armv7l)`
- Power supply used | Samsung Charger
- SDcard used | Samsung EVO+ 128GB

#### Additional Information (if applicable)
- Software title | Minio
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `N/A`

#### Steps to reproduce

1. Install Minio from dietpi-software > Software Optimised
2. Installation messages indicate minio binary downloaded from https://dl.min.io/server/minio/release/...
3. The issue is upstream have stopped providing fresh ARM builds since August 2018

#### Expected behaviour

- dietpi-software should download the latest source release from either `go get` or GitHub, and then compile minio from source.

MichaIng, 2020-02-26:
@GitHubGeek 
Many thanks for your report.

> The issue is upstream have stopped providing fresh ARM builds since August 2018

Shame on them... I am no fans of applying a source build within `dietpi-software`, which requires a lot of additional build dependencies, takes a lot of time (potentially) and may fail for OOM, any library incompatibilities and so on. Since it is a single binary file only, most likely I'll pre-compile them to be downloaded from our server instead.

GitHubGeek, 2020-02-27:
Your approach makes a lot of sense, RPI's are underpowered to compile such big program. Thanks again!

MichaIng, 2021-04-29:
ARM binaries are again up to date. Marking this as solved.

closed 2021-04-29 by MichaIng