MichaIng/DietPi #3398: dietpi-drive_manager - Check & Repair - "cat: /run/initramfs/fsck.log: No such file or directory"

maartenlangeveld, 2020-02-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.3
- Kernel version | `uname -a`
Linux DietPi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
1. Run dietpi-drive_manager
2. Select:  /                        : /dev/mmcblk0p2 | ext4 | Capacity: 15G | Used: 6.1G (45%)
3. Select: Check & Repair  :  Check and optionally repair filesystem
4. After system reboot : cat /run/initramfs/fsck.log

#### Expected behaviour
- cat /run/initramfs/fsck.log showing results of fsck

#### Actual behaviour
- cat /run/initramfs/fsck.log shows "cat: /run/initramfs/fsck.log: No such file or directory"

MichaIng, 2020-02-27:
@maartenlangeveld 
Many thanks for your report.

You mix up fsck done as part of initramfs (which is neither present on our RPi images, nor active on Raspbian) and fsck initiated by systemd (as in your case):
- If initramfs is present and its execution includes fsck (not always the case), `/run/initramfs/fsck.log` is created, as well as the flag file `/run/initramfs/fsck-root`.
- If that flag file is present later, systemd-fsck is skipped, else it does the check instead (as in your case). Log is done as usual to journald.
- Check `systemctl cat systemd-fsck-root` to verify this policy.

maartenlangeveld, 2020-02-27:
@MichaIng 

Thanks, I do not understand to be honest, my knowledge of Linux is very limited. I just followed the instructions given by dietpi-drive_manager. So it's not a fault but standard behaviour?

```
root@DietPi:~# systemctl cat systemd-fsck-root
# /lib/systemd/system/systemd-fsck-root.service
#  SPDX-License-Identifier: LGPL-2.1+
#
#  This file is part of systemd.
#
#  systemd is free software; you can redistribute it and/or modify it
#  under the terms of the GNU Lesser General Public License as published by
#  the Free Software Foundation; either version 2.1 of the License, or
#  (at your option) any later version.

[Unit]
Description=File System Check on Root Device
Documentation=man:systemd-fsck-root.service(8)
DefaultDependencies=no
Conflicts=shutdown.target
Before=local-fs.target shutdown.target
Wants=systemd-fsckd.socket
After=systemd-fsckd.socket
ConditionPathIsReadWrite=!/
ConditionPathExists=!/run/initramfs/fsck-root

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/lib/systemd/systemd-fsck
TimeoutSec=0
```

MichaIng, 2020-02-28:
@maartenlangeveld 
Jep, this is standard behaviour on RPi at least:
```
ConditionPathExists=!/run/initramfs/fsck-root
```
This means systemd-fsck is skipped of the flag file (hence log file as well) exists from fsck during initramfs.

MichaIng, 2020-03-09:
I mark this as closed, feel free to reopen if required.

closed 2020-03-09 by MichaIng