MichaIng/DietPi #3399: unable to update using  dietpi-update

sanfx, 2020-02-27:
#### Details:
- Date           | Thu 27 Feb 15:27:05 GMT 2020
- Bug report     | 2f306df7-4233-4c8e-aede-fbd118d43b07
- DietPi version | v6.25.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.93+ #1290 Fri Jan 10 16:34:37 GMT 2020 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:
trying to run dietpi-update

#### Actual behaviour:
update fails after the log below:
- ...

#### Extra details:
I can ping google.com 

#### Additional logs:
```
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:4 https://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:2 http://packages.ntop.org/apt/stretch_pi armhf/ InRelease [2,155 B]
Get:3 http://packages.ntop.org/apt/stretch_pi all/ InRelease [2,153 B]
Get:5 https://archive.raspberrypi.org/debian buster/main armhf Packages [278 kB]
Err:2 http://packages.ntop.org/apt/stretch_pi armhf/ InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 6F13614FD1EB60BE
Err:3 http://packages.ntop.org/apt/stretch_pi all/ InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 6F13614FD1EB60BE
Reading package lists...
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 1h 28min 49s). Updates for this repository will not be applied.
W: GPG error: http://packages.ntop.org/apt/stretch_pi armhf/ InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 6F13614FD1EB60BE
E: The repository 'http://apt.ntop.org/stretch_pi armhf/ InRelease' is not signed.
W: GPG error: http://packages.ntop.org/apt/stretch_pi all/ InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 6F13614FD1EB60BE
E: The repository 'http://apt.ntop.org/stretch_pi all/ InRelease' is not signed.
```

Joulinar, 2020-02-27:
hi,

seems you are using some Stretch repositories on a Buster system

```
http://raspbian.raspberrypi.org/raspbian buster
http://apt.ntop.org/stretch_pi
```

Do you use ntop Software?
Pls can you past the following

```
ls -Al /etc/apt/sources.list{,.d/*}
cat /etc/apt/sources.list{,.d/*}
```

MichaIng, 2020-02-28:
Exactly. Their instructions for RPi as well miss the key import, try:
```
curl -sSL http://packages.ntop.org/apt/ntop.key | apt-key add -
```
And replace in the related sources file:
`stretch_pi` with `buster_pi`.

Also for system time seems to be a bid out of sync, try:
```
/DietPi/dietpi/func/run_ntpd 1
```

sanfx, 2020-02-29:
I tried all your instructions and still managed to reach this point where I get the same error again:

```
 Command: Connection test:                                                             
       │ https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt                      
       │  - Exit code: 4                                                                          
       │  - DietPi version: v6.25.3 (MichaIng/master) | HW_MODEL:1 | HW_ARCH:1 | DISTRO:5         
       │  - Image creator: DietPi Core Team                                                       
       │  - Pre-image: Raspbian Lite                                                              
       │                                                                                          
       │ Log file content:                                                                        
       │ Spider mode enabled. Check if remote file exists.                                        
       │ --2020-02-29 10:45:55--                                                                  
       │ https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt                      
       │ Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Connection    
       │ timed out.                                                                               
       │ wget: unable to resolve host address ‘raw.githubusercontent.com’     
```

Joulinar, 2020-02-29:
still you are unable to resolve host address `raw.githubusercontent.com`

This could be an issue with your local DNS server. pls can you try to ping `raw.githubusercontent.com`

what DNS server you use? A local one or one of the public DNS Server?

MichaIng, 2020-02-29:
If ping succeeds, then test with wget manually:
```
wget --spider raw.githubusercontent.com
```

sanfx, 2020-02-29:
```
root@DietPi:~# ping raw.githubusercontent.com
PING github.map.fastly.net (151.101.0.133) 56(84) bytes of data.
64 bytes from 151.101.0.133 (151.101.0.133): icmp_seq=1 ttl=58 time=31.3 ms
64 bytes from 151.101.0.133 (151.101.0.133): icmp_seq=2 ttl=58 time=35.3 ms
64 bytes from 151.101.0.133 (151.101.0.133): icmp_seq=3 ttl=58 time=33.9 ms
^C
--- github.map.fastly.net ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 5ms
rtt min/avg/max/mdev = 31.335/33.515/35.343/1.661 ms
root@DietPi:~# wget --spider raw.githubusercontent.com
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2020-02-29 11:18:00--  https://raw.githubusercontent.com/
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.192.133, 151.101.128.133, 151.101.64.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.192.133|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://github.com/ [following]
Spider mode enabled. Check if remote file exists.
--2020-02-29 11:18:00--  https://github.com/
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.
```

Joulinar, 2020-02-29:
Ok all looks working fine. Can you try again

sanfx, 2020-02-29:
yes its working but I spotted something,

`Failed to restart pihole-FTL.service: Unit pihole-FTL.service not found.`

however I uninstalled pihole long back, but I dont think I ever installed on this raspberry pi

Joulinar, 2020-02-29:
but update was working now?

MichaIng, 2020-03-01:
@sanfx 
Does the error appear right after DietPi update, i.e. reproducable via `dietpi-services start`? Strange since it loops through only found/exiating services to start :thinking:.

sanfx, 2020-03-01:
yea strange I tried `dietpi-services start` and it doesn't come in list.

closed 2020-03-01 by sanfx