MichaIng/DietPi #3400: Can't connect with wifi

Mad1s, 2020-02-27:
#### Details:
- Date           | Thu 27 Feb 13:00:11 EST 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGI iw rfkill
- Exit code      | 100
- Software title | DietPi-Set_hardware

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/i/iw/iw_5.0.1-1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/u/util-linux/rfkill_2.33.1-0.1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2020-02-27:
Hi,

It seems you have local issues to resolve host names correctly to fetch the archives.

Mad1s, 2020-02-27:
Hi

I'm using DietPi. What can I do to resolve, this problem? 
Can this be easily fixable?

Joulinar, 2020-02-27:
ok do you have a valid internet connection?

Mad1s, 2020-02-27:
Yes. I have valid internet connection. When I'm trying to connect wifi, then bumps up this kind of error.
 - Command: G_AGI iw rfkill                                                                                           │  - Exit code: 100                                                                                                     │  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5                                      │  - Image creator: DietPi Core Team                                                                                    │  - Pre-image: Raspbian Lite
               │
               │ Log file content:
               │ E: Failed to fetch
               │ http://raspbian.raspberrypi.org/raspbian/pool/main/i/iw/iw_5.0.1-1_armhf.deb
               │ Temporary failure resolving 'raspbian.raspberrypi.org'
               │ E: Failed to fetch
               │ http://raspbian.raspberrypi.org/raspbian/pool/main/u/util-linux/rfkill_2.33.1-0.1_armh
               │ f.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
               │ E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
               │                                                                                                                                                                                                              │ If problems persist, please report at https://github.com/MichaIng/DietPi/issues for                                   │ investigation.                                                                                                        │                                                                                                                       │ Unable to continue, DietPi-Set_hardware will now terminate.

Joulinar, 2020-02-27:
yes because you are not able to resolve the host name `raspbian.raspberrypi.org`. It might be clocked by your DNS server.

MichaIng, 2020-02-28:
Hmm I see at least one issue with our script as well:
If there is currently no internet access, it still tries to pull the WiFi APT packages, which then of course fails. Depending on which packages are missing, this makes sense since WiFi cannot be enabled without, but in your case it might still be possible to connect and install the packages afterwards to allow complete WiFi configuration.

Do you have an Ethernet cable to plug first to install those packages? Else I'll give you further instructions about how to skip the install at first.

MichaIng, 2020-07-05:
I mark this as closed. Temporary issue most likely. Since those two packages are currently required to assure WiFi can be used, it generally makes sense that the command fails.

closed 2020-07-05 by MichaIng