MichaIng/DietPi #3404: Dietpi: problema with DNS???

carminerodio86, 2020-02-29:
#### Details:
- Date           | Sat 29 Feb 12:51:12 GMT 2020
- Bug report     | 75cb8204-fbcc-42d8-9602-a1fa0af7447e
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 one.one.one.one
- Exit code      | 2
- Software title | DietPi-Login

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
ping: one.one.one.one: Name or service not known
```

carminerodio86, 2020-02-29:
These are my Network configurations. Internet test ran with success.
 Ethernet     : Available | [On] | Connected                   │
│                WiFi         : Available | [On] | Disconnected                │
│                Onboard WiFi : [On]                                           │
│                             ●─ Additional Options ──────────●                │
│                IPv6         : [On]                                           │
│                Prefer IPv4  : [On]                                           │
│                Proxy        : [Off]                                          │
│                Test         : [Success] | Online

Joulinar, 2020-02-29:
Hi,

many thanks for your report. pls can you try to do the following

`ping -4 -c 1 one.one.one.one`

carminerodio86, 2020-02-29:
root@DietPi:~# ping -4 -c 1 one.one.one.one
ping: one.one.one.one: Name or service not known
root@DietPi:~#

Joulinar, 2020-02-29:
what about `ping 1.1.1.1` or p`ing 8.8.8.8`

carminerodio86, 2020-02-29:
64 bytes from 1.1.1.1: icmp_seq=5 ttl=59 time=23.9 ms

Joulinar, 2020-02-29:
and `ping google.com`

carminerodio86, 2020-02-29:
64 bytes from lhr35s11-in-f14.1e100.net (216.58.206.78): icmp_seq=3 ttl=46 time=77.9 ms

Joulinar, 2020-02-29:
so you are unable to resolve host name of `one.one.one.one` which is quite strange as this is part of Cloudflares DNS Server. It is the globally available DNS Name for Cloudflares 1.1.1.1 DNS resolver:

https://www.dnswatch.info/dns/dnslookup?la=en&host=1.1.1.1&submit=Resolve

Anyway if you like to change the connection check setting, you can do as follow. 

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```
you can change it to `8.8.8.8` and `google.com`

carminerodio86, 2020-02-29:
Ok. Now it seems working fine. First I was unable to start the software installation, but now the problem seems overcome. You have been very good and helpful. Thank you very much

Joulinar, 2020-02-29:
Did you changed something on config files or was it just working fine now without any change done?

carminerodio86, 2020-02-29:
I resolved applying your two indications (changing 1.1.1.1 and one.one.one.one in 8.8.8.8 and google.com rispectively)

Joulinar, 2020-02-29:
ok good. At least ist fix now.

But would be good to find out why you are not able to resolve `one.one.one.one`. Maybe it's blocked by your DNS Server/Provider.

MichaIng, 2020-03-01:
Strange indeed that 1.1.1.1 is reachable but the related domain name not. The "one" top level domain is relatively new (2015) which might be a reason for some outdated DNS caches: https://en.m.wikipedia.org/wiki/.one_(domain)

Another reason to not use Cloudflare as default for connection and DNS tests at all.

Joulinar, 2020-03-01:
@MichaIng 
Which is a pity as Cloudflare is one if the best and fastest DNS providers atm

MichaIng, 2020-03-01:
Jep world-wise average the fastest :cry:.

closed 2021-02-14 by MichaIng