MichaIng/DietPi #3405: [RockPro64 => Buster] /etc/resolve.conf has always 1.0.0.1 as first nameserver

St3S0f, 2020-02-29:
### Creating a bug report/issue
There's an entry in /etc/resolvconf/resolv.conf.d/head that fixes the first nameserver in /etc/resolve.conf to 1.0.0.1, disregarding the DHCP provided entries.
Works as expected on the stable images.

### Workaround
comment the line, then issue 'resolvconf -u'

#### Required Information
- DietPi version | v6.28
- Distro version | 10.3
- Kernel version | Linux zippy 5.4.20-rockchip64 #20.02.1 SMP PREEMPT Mon Feb 17 04:45:00 CET 2020 aarch64 GNU/Linux
- SBC device | RockPro64
- Power supply used | official
- SDcard used | SanDisk ultra

MichaIng, 2020-03-01:
@St3S0f 
Many thanks for your report. I'll check and in case repack with correct default file.

I aim to add a forces replacement of old configs with new maintainer shipped ones for new images, which should resolve such issues regardless of the used pre-images. Moreover for this particular case I as well aim to remove resolvconf package by default. It is not required for dynamic DNS server entries. DHCP clients / ifupdown etc alter resolv.conf directly, in case, instead of invoking the resolvconf command, which works just the same. Only benefit of resolvconf is that it does not require disk I/O since the symlink target is in /run tmpfs.

MichaIng, 2020-03-02:
@St3S0f 
I repacked the package with default resolv.conf.d content and added some further fine tuning.

You can do the following:
- `rm /etc/network/if-up.d/dietpi-disable_offload`
  - Not required anymore with newest Armbian kernel, having offload enabled might increase Ethernet performance/reduce CPU load slightly.
- `G_CONFIG_INJECT 'docker_optimizations=' 'docker_optimizations=off' /boot/armbianEnv.txt`
  - If you don't use docker, should should enhance performance again slightly. If you use docker, this is useful/required for some features though.

closed 2020-03-13 by MichaIng