MichaIng/DietPi #3406: HDMI Output doesn't rotate

michidk, 2020-03-02:
### Creating a bug report/issue

#### Required Information
- DietPi version: 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version: buster
- Kernel version: Linux infoscreen 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- SBC device: RPi 4 Model B (armv7l)
- Power supply used: 5V 1A
- SDcard used: SanDisk ultra

#### Additional Information (if applicable)
- Software title: chromium

#### Steps to reproduce
1. Install DietPi with Chromium selected
2. set `display_hdmi_rotate=1` in the config and place it in `/boot/config.txt` and `/DietPi/config.txt`

#### Expected behavior
- Screen (hdmi) should be rotated by 90 degrees

#### Actual behaviour
- Screen is not rotated

michidk, 2020-03-02:
Seems like this is an issue with RP4.
The documentation says

> If using the VC4 FKMS V3D driver (this is the default on the Raspberry Pi 4), then 90 and 270 degree rotations are not supported. The Screen Configuration utility provides display rotations for this driver.
(https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)

Joulinar, 2020-03-02:
Hi,

many thanks for your report. Did you tried using `dietpi-config` to rotate the screen?

DietPi for RPi is based an Raspbian Lite, so it's using standard packages.

michidk, 2020-03-02:
Hi @Joulinar,

yes I also tried it with the GUI (`dietpi-config`)

michidk, 2020-03-02:
While researching about this issue, most people seem to have fixed it by uncommenting `dtoverlay=vc4-kms-v3d`. But I have the same issue when just having `display_hdmi_rotate=1` and nothing else in the config file. Is DietPi falling back to a default config of some sorts, if a value is not set?

michidk, 2020-03-02:
Okay I got it to work. It just works differently for RP4.
First you have to set `dtoverlay=vc4-kms-v3d` in the config.
Then you have to rotate using xrandr: `DISPLAY=:0 xrandr --output HDMI-1 --rotate left`

I think this might be worth a comment in the example configuration and GUI. Maybe one could even add support for that, so that `rotate_display` and when changing the rotation in the GUI, just uses xrandr to rotate the display.

MichaIng, 2020-03-02:
Good to know, jep this should be added to the UI info then and e.g. rotation should be reset and blocked then until we find a matching replacement via xrandr or similar.

So you mean using display_hdmi_rotate without the fkms diver, output is rotated but GUI applications are shown un-correctly rotated (portrait format screen) and when using xrandr it's like a landscape monitor manually rotated, so that GUI applications are wrong rotated?

michidk, 2020-03-02:
@MichaIng `display_hdmi_rotate` didnt have any effect for me. 
When using xrandr --rotate the problem is that half the screen is still black and the actual screen is cut of.

[image]

MichaIng, 2020-03-02:
@michidk 
> display_hdmi_rotate didnt have any effect for me.

Jep makes sense then with fkms driver enabled.

But now I am confused since you use a landscape monitor, why do you want to rotate it? In case of portrait monitor (or probably you can rotate it mechanically?) did you try to adjust the framebuffer entries in `config.txt` accordingly?

Joulinar, 2020-03-02:
is that Chromium?

I guess you need to swap rotation as well

```
SOFTWARE_CHROMIUM_RES_X=1024
SOFTWARE_CHROMIUM_RES_Y=768
```

MichaIng, 2020-03-02:
Good point, at least for kiosk mode.

michidk, 2020-03-02:
So we basically use it as a display to advertise future events. The monitor is mounted in portrait mode. 

I also tried ```
framebuffer_width=1080
framebuffer_height=1920
SOFTWARE_CHROMIUM_RES_X=1080
SOFTWARE_CHROMIUM_RES_Y=1920
```

The framebuffer just affected the console display and not the output of the xserver.

Joulinar, 2020-03-02:
question: during reboot the full screen is used? Just Chromium is not displayed correctly?

michidk, 2020-03-02:
@Joulinar During startup the full screen is used and not rotated. But this is because I use xrandr after initializing the xserver. 
I never got the console output to be rotated at all, but this is not important to me anyways.

If I set the framebuffer height and width, it will affect the console output which is displayed during startup.

Joulinar, 2020-03-02:
If I have time tomorrow, I will play around with my Pi4.

MichaIng, 2020-03-02:
> If I set the framebuffer height and width, it will affect the console output which is displayed during startup.

Makes sense, since this is indeed the "console" framebuffer. Hmm this might even rotate with `display_hdmi_rotate` then, since KMS driver should not yet be in use there (?).

michidk, 2020-03-02:
`display_hdmi_rotate ` didnt rotate the console nor the xserver output, even without using `dtoverlay=vc4-fkms-v3d`

MichaIng, 2020-03-02:
@michidk 
Very strange, if I remember right I tested this a while ago, although not on RPi4, where probably things are different then 🤔.

Little not about /DietPi/config.txt, since you used echo to add the entries: Assure that those are only present once, as they are present in our default config already: https://github.com/MichaIng/DietPi/blob/master/config.txt#L51

michidk, 2020-03-02:
I have my own config.txt file, which is based on your default one. When not testing, I copy it on the sdcard in the root folder.
For testing I always clear the whole file and just add the entries I'm intersted in. So this shouldn't have infered with it.

Joulinar, 2020-03-03:
quite frustrating the screen rotation topic

https://www.raspberrypi.org/forums/viewtopic.php?t=247619

MichaIng, 2020-03-03:
@Joulinar 
At least the thread just verifies that config.txt entries do not work but xrandr instead. For desktops this is fine, we could install arandr (GUI frontend for xrandr) by default as well, or suggest it when dietpi-config rotation option is called. But this does not work for rotating the console. Hopefully the lcd_rotate entry does work at least, since this is a more frequent use case.

michidk, 2020-03-03:
Xrandr only works with the `dtoverlay=vc4-kms-v3d` setting.

I confirmed that xrandr is rotating the screen correctly, but I have issues getting chrome to render on the full screen.

michidk, 2020-03-03:
Finally got it to work. Apparently, you have to use "--app-window-size" instead of "--window-size" for chromium to use the full display.

Joulinar, 2020-03-04:
@michidk 
are you able to rotate chromium already if it was set to autostart? Or do you need to flip the screen manually once system is started using `DISPLAY=:0 xrandr --output HDMI-1 --rotate left`???

Because I found another way to rotate chromium without using `xrandr` by hacking `/etc/X11/xorg.conf.d/99-dietpi-dpms_off.conf`

michidk, 2020-03-04:
@Joulinar I need to flip the screen manually. Once the xserver is running, I execute the rotate command.

Could you send me your `99-dietpi-dpms_off.conf`? Because I tried it that way before but didnt get it to work.

Joulinar, 2020-03-04:
@michidk 
I did the following 

1. hide error messages in Chromium on autostart
`/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`
```
CHROMIUM_OPTS="--test-type --window-size=$RES_X,$RES_Y --start-fullscreen --start-maximized --window-position=0,0"
CHROMIUM_OPTS+=" --app=$URL"
```

2. hack the xserver
`/etc/X11/xorg.conf.d/99-dietpi-dpms_off.conf`

```
Section "Monitor"
        Identifier   "Monitor0"
        VendorName   "Monitor Vendor"
        ModelName    "Monitor Model"
        Option  "Rotate" "left" # You SEE: Rotate LEFT
EndSection

Section "Screen"
        Identifier "Screen0"
        Device     "Card0"
        Monitor    "Monitor0"
        SubSection "Display"
                Viewport   0 0
                Depth    24
                Modes "1920x1080" # RESOLUTION
        EndSubSection
EndSection

Section "ServerLayout"

    Identifier "ServerLayout0"
    Option "StandbyTime" "0"
    Option "SuspendTime" "0"
    Option "OffTime"     "0"
    Option "BlankTime"   "0"

EndSection
```
3. resolution for Chromium
`/DietPi/dietpi.txt`
```
SOFTWARE_CHROMIUM_RES_Y=1920
SOFTWARE_CHROMIUM_RES_X=1080
```

4. Resolution for the screen and the RPi4 driver to rotate
`/DietPi/config.txt `
```
framebuffer_width=1920
framebuffer_height=1080
dtoverlay=vc4-fkms-v3d
```

5. hide the mouse after Chromium has started
`G_AGI unclutter && echo '/usr/bin/unclutter -idle 0.1 &' > /etc/chromium-browser/customizations/99-dietpi-unclutter`

Now Chromium is able to rotate on autostart as well.

michidk, 2020-03-04:
@Joulinar interesting. Thanks for your help!
I created my own startup script:
```
# turn off screensaver
xset -dpms
xset s off
xset s noblank

# Resolution to use for kiosk mode, should ideally match current system resolution
RES_X=$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_RES_X=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')
RES_Y=$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_RES_Y=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')

# URL
URL=$(grep -m1 '^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=' /DietPi/dietpi.txt | sed 's/^[^=]*=//')

# Chromiom args
CHROMIUM_OPTS=\" --no-sandbox --homepage $URL --kiosk --window-size=$RES_X,$RES_Y --app-window-size=$RES_X,$RES_Y --window-position=0,0 --incognito --noerrdialogs --disable-infobars \"

# Chromium binary
FP_CHROMIUM=\"$(command -v chromium-browser)\"

xinit $FP_CHROMIUM $CHROMIUM_OPTS -- -nocursor
```
 
As you can see, I hide the cursor via the `-nocursor` flag.

But where exactly do you rotate your screen? You only change the resolution in your `/etc/X11/xorg.conf.d/99-dietpi-dpms_off.conf`. I tried it with `Option "Rotate" "left"` without success.
 
Also does your xserver run as root? My chrome only starts up with the `--no-sandbox` flag, but then also shows a warning that I shouldn't use that flag. I can't figure out how to remove that warning. Also tried running the xserver under the dietpi user without success.

Joulinar, 2020-03-04:
this is what my screen let rotate. Basically I added the entire Section "Screen"

```
Section "Monitor"
        Identifier   "Monitor0"
        VendorName   "Monitor Vendor"
        ModelName    "Monitor Model"
        Option  "Rotate" "left" # You SEE: Rotate LEFT
EndSection

Section "Screen"
        Identifier "Screen0"
        Device     "Card0"
        Monitor    "Monitor0"
        SubSection "Display"
                Viewport   0 0
                Depth    24
                Modes "1920x1080" # RESOLUTION
        EndSubSection
EndSection
```

For the rest I'm using DietPi Standards

Joulinar, 2020-03-04:
@michidk 
I don't use the `--no-sandbox` flag and yes Chromium is started as root using DietPi Autostart function.

MichaIng, 2020-03-04:
Great idea to go with X11 config file. However please add a new file for this, e.g.:
```
/etc/X11/xorg.conf.d/99-rotate_screen.conf
```
So this will be applied persistently to all Xserver applications then, better then having to run xrandr every time. Just not sure if/how it can conflict with desktop GUI-based configuration, but I guess the worst that can happen is that the resolution changes multiple times, once on Xserver init, once again when desktop loads it's e.g. xrandr based settings.

Joulinar, 2020-03-04:
@MichaIng 
Stupid question from my side. How will it behave if there are 2 config files. I guess both will be used. Or can it be specific which one to choose? 

Will try it later the evening if I'm back home.

MichaIng, 2020-03-04:
Both will be used, parsed in alphabetical order. But to be true I'm not 100% sure if two same sections are merged (only same options override each other) or those override each other fully.

Joulinar, 2020-03-04:
Ok will let you know after my test.

michidk, 2020-03-04:
Thanks guys!
I finally got it to work.
File `/etc/X11/xorg.conf.d/01-rotate-monitor`:
```
Section "Monitor"
        Identifier "Monitor0"
        Option "Rotate" "left"
EndSection

Section "Screen"
        Identifier "Screen0"
        Device "Card0"
        Monitor "Monitor0"
EndSection
```

Disclaimer: I have no clue how this xorg config file works. I just made sense of what you have sent me and now it works :D

Joulinar, 2020-03-06:
@michidk 
just for my one interest, how does your `xinit `command looks like at the end once Chromium was started? As I can see, you use quite a lot of options to call Chromium.

michidk, 2020-03-06:
@Joulinar Those parameters a based on the default dietpi chromium startup script.
I'm not sure where to find them in this repository, but you can find them here on your DietPi: `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh`
I just had to remove some parameters, because they seemed to cause chromium rendering issues.

Joulinar, 2020-03-06:
I know how the original config looks like.  but you are adding quite some parater like `--app-window-size=$RES_X,$RES_Y --incognito --noerrdialogs --disable-infobar --no-sandbox`. And as far as I can see you removed `--start-fullscreen --start-maximized`

michidk, 2020-03-06:
--start-fullscreen --start-maximized didnt do anything for me.
--app-window-size is the only size parameter that worked for me in conjunction with the kiosk mode.
Chromium wont start from the root user without --no-sandbox. But --no-sandbox will provoke a warning, which i suppress with --noerrdialogs and --disable-infobar.

Joulinar, 2020-03-06:
well you could use `--app` instead of `--kiosk`

michidk, 2020-03-06:
What would be the advantage of using  --app instead of --kiosk?

MichaIng, 2020-03-06:
@michidk 
--app mode hides/omits all error messages and prompts (e.g. ask to restore old session and such), hence can be used completely without mouse/keyboard/input, while --kiosk mode still requires input in those cases as error messages and prompts can appear and cover the website.

michidk, 2020-03-07:
Thanks for the hint!

MichaIng, 2021-09-01:
I mark this as closed. I think some of the challenges with RPi 4 and screen rotation have been solved firmware wise.

closed 2021-09-01 by MichaIng