MichaIng/DietPi #3407: error on dietpi-update

seafile007, 2020-03-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch

- Kernel version | `uname -a`
Linux DietPi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi Zero W (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 1600 mA

- SDcard used | (EG: SanDisk ultra)
San Disk ultra
#### Additional Information (if applicable)

- Can this issue be replicated on a fresh installation of DietPi?
don't tested

error ::
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
[.     ] /DietPi/dietpi/dietpi-bugreport: line 177: curl: command not found
[FAILED] DietPi-Bugreport | Failed to connect to SFTP server. Please try again later or report this to DietPi forum or GitHub repo in the first place.

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`
4219063b-eaa0-4876-b858-1549a98f9683

#### Steps to reproduce

1. dietpi-update

#### Expected behaviour

- dietpi-update ends successfully

#### Actual behaviour

- dietpi-update dont ends successfully
error : /DietPi/dietpi/dietpi-update: line 115: curl: command not found

#### Extra details

- all steps from https://dietpi.com/phpbb/viewtopic.php?t=418 don't change anything,
the error comes up again
- reboot also dosn't help

Joulinar, 2020-03-02:
Hi,

many thanks for your report. Looks like you are missing `curl ` executable. Pls can you just simply run `curl` at command line.

MichaIng, 2020-03-02:
```
apt update && apt install --reinstall curl
```
should solve it, however it got removed 🤔.

seafile007, 2020-03-03:
Hello everyone and especially @MichaIng ,

`apt update && apt install --reinstall curl`

this solved the problem. Dietpi-update is running smoothly again.
Thank you very much! :-)

closed 2020-03-03 by seafile007