MichaIng/DietPi #3408: Node-Red does not install due to nodejs V13

borpin, 2020-03-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.3
- Kernel version | `uname -a`
Linux DietPi 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
ESXI VM x86_64
- Power supply used | (EG: 5V 1A RAVpower)
VM - 240V
- SDcard used | (EG: SanDisk ultra)
Nil

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Node-Red
- Was the software title installed freshly or updated/migrated? Fresh install clean VM
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. `diepi-software install 122`

#### Expected behaviour

- Node-Red should install

#### Actual behaviour

- There is an Illegal Instruction error

#### Extra details

- Please see discussion [Cannot install Node-Red on VM](https://dietpi.com/phpbb/viewtopic.php?f=11&t=7325)

- Node-Red state that V12 is the most recent supported version of nodejs - DietPi installs V13
- The Raspbian install script installs V12 of nodejs.

Whilst I accept that some setups do work with V13, as V12 is the supported version, that is the version that should be installed by DietPi.

borpin, 2020-03-03:
I have just used the official NR install script (commented out the DietPi check) and node-red successfully installed (as user `dietpi`).

MichaIng, 2020-03-03:
@borpin 
Many thanks for your report. Yes there are other cases where lower Node version are required. Via `n` module this is luckily quite easy.

I'll also check the official NR install script, not sure if this did not exist earlier or we faced other issues with it.

borpin, 2020-03-03:
> I'll also check the official NR install script, not sure if this did not exist earlier or we faced other issues with it.

I wasn't suggesting using it, the locations of the user data precludes it I suspect, I just used it as an example of the version of nodjs it installed.

MichaIng, 2020-03-04:
@borpin 
I'll check regardless, probably the steps required afterwards (to comply with DietPi) are less effort compared to keeping up with dependencies and Node version... However a main Node version indeed is not hard to apply and we do this in other cases as well. Only issue is that with "n", the Node version is applied system wide. We're thinking about running each program with the latest Node version it supports as well: #2226

borpin, 2020-03-04:
> Ah, you were already responding to that issue as well

😀 yes, 18 months ago 😀.  I think that was because of a npm package that required a specific version of nodejs.

One other point, for me, I could not manually downgrade as nodejs V13 simply does not work - any `npm` command is greeted with an 'illegal instruction' error 😢.

As a minimum, I think the nodejs install should offer a 'do you want a specific version?' option during install (or before).

MichaIng, 2020-03-04:
@borpin 
We use the official Node installer, and generally, as well since we offer unintended installs, it should by default pull the most current version. Manual downgrading via "n" as you mentioned in the other issue is easy anyway.

> One other point, for me, I could not manually downgrade as nodejs V13 simply does not work - any npm command is greeted with an 'illegal instruction' error :cry:.

Okay, but this then means that v13 is not only incompatible with Node-RED but with your whole system? Hmm this requires additional investigation then, probably there have been made some deeper changes that require a certain Linux version or such :thinking:.

Joulinar, 2020-03-04:
As already mention on the forum entry, node-red installation is working fine on my VM as well as RPi system. Even if it pull node.js v13. So it seems something special on the ESXi Server installation. Maybe due to the server architecture.

MichaIng, 2020-03-04:
Okay good to know. Probably a kernel upgrade helps, e.g. pulling Linux 5.X from backports:
`apt install linux-image-amd64/buster-backports`

borpin, 2020-03-05:
> Okay, but this then means that v13 is not only incompatible with Node-RED but with your whole system?

@MichaIng Potentially yes. As I said it used to work but when I started with a fresh Buster image it would not work.

> node-red installation is working fine on my VM as well as RPi system. Even if it pull node.js v13.

@Joulinar As I said, I'm not denying that. But if the **official** NR install script uses V12 and their pages state that the supported version is V12, then custom installers should respect that and only install V12 (IMHO).

MichaIng, 2020-03-05:
I'll run some test on VMware later, which should behave very close to ESXI VM (?).

MichaIng, 2020-03-17:
I ran testes on VMware and everything works pretty fine. So even that Node-RED officially states Node v12 as latest supported version, I am not too keen to add a version downgrade to the installer, which we then need to remove again at a later time.

A better idea is to concentrate on debugging Node v13 on (your) ESXi Buster machine, as this seems to be the root of the issue, not Node-RED + Node v13 compatibility.

MichaIng, 2020-07-05:
I mark this as closed. VM-specific it seems, must be replicated on current ESXi with current kernel package before we can pick up.

closed 2020-07-05 by MichaIng