MichaIng/DietPi #3409: Chromium doesn't install through dietpi-software Odroid c2 Buster (workaround)

ickam, 2020-03-04:
### Creating a bug report/issue

#### Required Information
-G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng
`
10.3
Linux Odroid 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- Odroid C2
- standard odroid PSU
- Cheepo sandisk 8gb mSd card 

#### Additional Information (if applicable)
- freshly installed Chromium
- Can probably be replicated?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ... Try to install chromium via dietpi-software

#### Expected behaviour

- Install chromium

#### Actual behaviour

- Cannot find required dependencies: aml-libs-odroid
libump-odroid
xf86-video-fbturbo-odroid 
Upon manually installing dependencies I got an error about a file not being able to be overwritten, which required an extra command (see below).

#### Extra details

- I googled the packages, pulled them with wget and installed with dpkg -i (as root)
`wget https://oph.mdrjr.net/meveric/pool/main/a/aml-libs-odroid/aml-libs-odroid_0.0.1-3+deb8_arm64.deb
wget https://oph.mdrjr.net/meveric/pool/main/libu/libump-odroid/libump-odroid_0.0.1-r6p1-01rel0-1+deb9_arm64.deb
wget https://oph.mdrjr.net/meveric/pool/main/x/xf86-video-fbturbo-odroid/xf86-video-fbturbo-odroid_20170625-1+deb9_arm64.deb
``
After installing the packages I did
`cd /var/cache/apt/archives
dpkg -i --force-overwrite mali450-odroid_20161218-r6p1-1_arm64.deb`

After these steps Chromium installs through dietpi-software

MichaIng, 2020-03-04:
Many thanks for your report. Hmm that means that one or more files from the package are installed by another package as well. That is bad because then the other package might be broken by the new file. I'll try to replicate and see where it comes from.

MichaIng, 2020-03-17:
Okay the core of the issue matches #3417.
I contacted Meveric and ask to add additional Conflict/Provides entries to his Mali packages, to be compatible with Buster and probably Bullseye (future prove) and port aml-libs-odroid + libump-odroid + xf86-video-fbturbo-odroid over to the Buster repo.

MichaIng, 2020-03-19:
Issue should be solved: https://github.com/MichaIng/DietPi/issues/3417#issuecomment-601400612

closed 2020-05-03 by MichaIng