MichaIng/DietPi #3411: I dont have fail2ban

parapac, 2020-03-05:
I am trying to install Kodi but I keep getting this error.
I dont have fail2ban installed 
---------------------------------------------------------------------

#### Details:
- Date           | Thu  5 Mar 14:23:56 EET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (ID=11)
- Kernel version | Linux XU4 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Setting up fail2ban (0.9.6-2) ...
Job for fail2ban.service failed because the control process exited with error code.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
invoke-rc.d: initscript fail2ban, action "start" failed.
● fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2020-03-05 14:23:45 EET; 35ms ago
     Docs: man:fail2ban(1)
  Process: 5403 ExecStart=/usr/bin/fail2ban-client -x start (code=exited, status=255)

Joulinar, 2020-03-05:
Hi,

well it looks like there is a fail2ban installed. At least it seems some service exist. Can you past output of following?

`systemctl status fail2ban.service`

MichaIng, 2020-05-18:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-05-18 by MichaIng