MichaIng/DietPi #3412: incorrect cpu temperature ( -263200'C : -473728'F (Who put me in the freezer!) )

kt1024, 2020-03-06:
intel nuc 7i5, normally temperature should be 4x 'C. (tried with proxmox ve 6.1, same hardware), at bios it's 4x-5x 'C .
after install dietpi 'cpu' command result as below. (right after install, tried update/reboot).
```
root@DietPi:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     x86_64
 Temperature  |     -263200'C : -473728'F (Who put me in the freezer!)
 Governor     |     performance

                 Current Freq    Min Freq   Max Freq
 CPU0         |      3400 MHz      400 MHz    3400 MHz
 CPU1         |      3400 MHz      400 MHz    3400 MHz
 CPU2         |      3400 MHz      400 MHz    3400 MHz
 CPU3         |      2208 MHz      400 MHz    3400 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.

root@DietPi:~# lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
Address sizes:       39 bits physical, 48 bits virtual
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  2
Core(s) per socket:  2
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               142
Model name:          Intel(R) Core(TM) i5-7260U CPU @ 2.20GHz
Stepping:            9
CPU MHz:             3400.997
CPU max MHz:         3400.0000
CPU min MHz:         400.0000
BogoMIPS:            4416.00
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            256K
L3 cache:            4096K
NUMA node0 CPU(s):   0-3
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d
```

#### Required Information
- DietPi version
```
root@DietPi:~# cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version
```
root@DietPi:~# cat /etc/debian_version
10.3
```
- Kernel version 
```
Linux DietPi 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
```
- SBC device | Intel NUC 7i5B , 16GB RAM X2

- Power supply used | 19V 3A,
- SDcard used | no SD Card, use Intel SSD 512G, nvme interface.

MichaIng, 2020-03-06:
@kt1024 
Many thanks for your report.

This issue is similar: #3172
Simply too many different CPU temp APIs in use across x86 boards...

Can you please run the following in your console paste the output here:
```
for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input /sys/devices/platform/coretemp.[0-9]/hwmon/hwmon[0-9]/temp[0-9]_input
do
[[ -e $i ]] && echo "$i : $(<$i)"
done
```

kt1024, 2020-03-06:
@MichaIng 
thanks, here it's:
```
root@DietPi:~# for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input /sys/devices/platform/coretemp.[0-9]/hwmon/hwmon[0-9]/temp[0-9]_input
> do
> [[ -e $i ]] && echo "$i : $(<$i)"
> done
/sys/class/thermal/thermal_zone0/temp : -263200
/sys/class/thermal/thermal_zone1/temp : 34500
/sys/class/thermal/thermal_zone2/temp : 38000
/sys/class/hwmon/hwmon0/temp1_input : -263200
/sys/class/hwmon/hwmon1/temp1_input : 34500
/sys/class/hwmon/hwmon2/temp1_input : 39000
/sys/class/hwmon/hwmon2/temp2_input : 39000
/sys/class/hwmon/hwmon2/temp3_input : 36000
/sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input : 39000
/sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input : 39000
/sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input : 36000
```

MichaIng, 2020-03-06:
@kt1024 
So its again the /sys/devices/platform/coretemp API in your case. Good, so both issues are solved with the same addition then.

MichaIng, 2020-03-07:
@kt1024 
Added for v6.29: https://github.com/MichaIng/DietPi/commit/42bc75fca16993e5a1f94461bd25242b20dd2f99
Changelog: https://github.com/MichaIng/DietPi/commit/e91439174a236fd2bffcfa50285b3e0427eae6ba

LeeMenHin, 2020-03-19:
─────────────────────────────────────────────────────
 DietPi v6.28.0 : 02:30 - 五 2020年03月20日
 ─────────────────────────────────────────────────────
 - Device model : Native PC (x86_64)
 - CPU temp : 0'C : 32'F (Who put me in the freezer!)
 - LAN IP :  (eth0)
 - MOTD : Folding@home takes up the fight against 2019 Corona virus:
          -> https://foldingathome.org/2020/02/27/
          On x86_64 system you can install it via dietpi-software and contribute.
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Debian mini.iso)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input /sys/devices/platform/coretemp.[0-9]/hwmon/hwmon[0-9]/temp[0-9]_input
> do
> [[ -e $i ]] && echo "$i : $(<$i)"
> done
/sys/class/thermal/thermal_zone0/temp : 0
/sys/class/thermal/thermal_zone1/temp : -273200
/sys/class/hwmon/hwmon0/temp1_input : 0
/sys/class/hwmon/hwmon1/temp2_input : 48000
/sys/class/hwmon/hwmon1/temp3_input : 49000
/sys/class/hwmon/hwmon1/temp4_input : 50000
/sys/class/hwmon/hwmon1/temp5_input : 49000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp2_input : 48000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp3_input : 49000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp4_input : 50000
/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp5_input : 49000
root@DietPi:~#

LeeMenHin, 2020-03-19:
@MichaIng  Here it is

MichaIng, 2020-03-19:
@LeeMenHin 
Okay great, this is fixed for v6.29 already (see above). I guess your machine has 4 CPU cores? The banner/cpu command will only show the temperature for the first core (`temp2_input`), but they temperature anyway will not differ much. Impossible to have a generic method for all x86_64 machines, as you can see they do not follow any clear rules about this.

LeeMenHin, 2020-03-19:
> @LeeMenHin
> Okay great, this is fixed for v6.29 already (see above). I guess your machine has 4 CPU cores? The banner/cpu command will only show the temperature for the first core (`temp2_input`), [...]

maybe great method to detect the cpu model ?

ys is 4 core

MichaIng, 2020-03-19:
@LeeMenHin 
> maybe great method to detect the cpu model ?

What you mean? In case of x86_64 we detect that it's x86_64. We could scrape some /proc/cpuinfo vendor/model string but that would not help. There are thousands of x86 CPUs and all of them present/use different modules/the kernel API differently to present the temperatures of different mainboard parts. So it's basically to find a generic order of scanning those files that works for most of them 😉. There are other projects that only deal with detecting the CPU temperatures, but those are huge (for this single task), hence not something that we would integrate into our scripts. Others struggle with the same, like `htop`, where for years are requests open to show the CPU temperature as well 😛.

gary2002, 2020-03-19:
My unit is an Odroid-N2:

 ───────────────────────────────────────────
 DietPi v6.28.0 : 10:49 - Fri 03/20/2020
 ───────────────────────────────────────────
 - Device model : Odroid N2 (aarch64)
 - CPU temp : 40'C : 104'F (Optimal temperature)
 - LAN IP : 192.168.20.252 (eth0)
 - MOTD : Folding@home takes up the fight against 2019 Corona virus:
          -> https://foldingathome.org/2020/02/27/
          On x86_64 system you can install it via dietpi-software and contribute.
 ───────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Meveric)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@xxx:~# for i in /sys/class/thermal/thermal_zone[0-9]/temp /sys/class/hwmon/hwmon[0-9]/temp[0-9]_input /sys/devices/platform/coretemp.[0-9]/hwmon/hwmon[0-9]/temp[0-9]_input
> do
> [[ -e $i ]] && echo "$i : $(<$i)"
> done
/sys/class/thermal/thermal_zone0/temp : 38600
/sys/class/thermal/thermal_zone1/temp : 40100
root@xxx:~#

MichaIng, 2020-03-20:
@gary2002 
And your temperature is already detected correctly, isn't it?

gary2002, 2020-03-20:
Certainly seems to be, both in terms of the number reported during login (ssh) and the feeling of the temperature of the heatsink.

Currently reporting 37 degrees Celsius.

I cannot see a temperature module in Webmin and installing sensors-detect or sensors_detect doesn’t work.

Gary

gary2002, 2020-03-20:
Just noting that my SBC is the ODROID-N2 - and DietPi is running perfectly (at least as far as my configuration is concerned ;-)

Gary

MichaIng, 2020-03-21:
@gary2002 
> I cannot see a temperature module in Webmin and installing sensors-detect or sensors_detect doesn’t work.

That is indeed strange since the `/sys/class/thermal/thermal_zone[0-9]/temp` API (that we check among the others) is quite known and works fine. No idea what webmin is looking for but `apt install lm-sensors && sensors-detect` should show them as well, probably some more.

GvY85, 2020-04-17:
This seems to be fixed on the 6.29 beta on UP-Squared N4200 pc's

Joulinar, 2020-09-03:
@MichaIng 
We have a new report on similar issue with DietPi v6.32.2

https://dietpi.com/phpbb/viewtopic.php?f=11&t=8053

MichaIng, 2020-09-03:
Found it, I started to apply some general coding standards, including variable and array expansion, even if it is not strictly required in cases where the values are known to be single word without globs. But in this case, glob expansion (`*`) is required to match those coretemp file paths, so quoting the array broke it: https://github.com/MichaIng/DietPi/commit/d6a170de2b3f865c848d9ae0884e5a1fd93df6e6

closed 2020-10-17 by MichaIng