MichaIng/DietPi #3413: No execute permission for youtube-dl

ArturSierzant, 2020-03-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.28`
- Distro version | `buster`
- Kernel version | `Linux DietPi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B+ (armv7l)`
- Power supply used | 5V, 3A
- SDcard used | Goodram, 4GB

#### Additional Information (if applicable)
- Software title | O!MPD
- Was the software title installed freshly or updated/migrated? Fresh install.
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `N/A`

#### Steps to reproduce

1. Install O!MPD from dietpi-software -> Software Optimised

#### Expected behaviour

- `/usr/local/bin/youtube-dl` should have execute permission (0755: rwxr-xr-x)

#### Actual behaviour

- `/usr/local/bin/youtube-dl` lacks execute permission, it has only 0644 (rw-r--r--)  

#### Extra details

- Without execute permission for `/usr/local/bin/youtube-dl` O!MPD can't get necessary information from YouTube (like stream URL) and fails to play audio track.

MichaIng, 2020-03-07:
@ArturSierzant 
Many thanks for your report. Indeed these have been forgotten 😞. I just fixed it: https://github.com/MichaIng/DietPi/commit/0be569eaf7536fd6f7ab2ae6b5eb7f6bca941cac

ArturSierzant, 2020-03-30:
Thanks for fixing that 👍

closed 2020-03-30 by ArturSierzant