MichaIng/DietPi #3417: Kodi installation failure

ernero93, 2020-03-14:
#### Details:
- Date           | Sat 14 Mar 08:39:30 GMT 2020
- Bug report     | aa280d12-849b-42ab-b30e-d66ffb2eba97
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGI aml-libs-odroid mali450-odroid libump-odroid xf86-video-fbturbo-odroid
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Fresh install 
2. Kodi install

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: Unable to locate package aml-libs-odroid
E: Package 'libump-odroid' has no installation candidate
E: Unable to locate package xf86-video-fbturbo-odroid
```

MichaIng, 2020-03-14:
@ernero93 
Many thanks for your report.

Did you use the new beta image? https://github.com/MichaIng/DietPi/issues/2979
Can you please paste the following output:
```
cat /etc/apt/sources.list{,.d/*.list}
```

MichaIng, 2020-03-14:
Okay it should be possible to add the Stretch packages. The packages have no specific dependencies and should basically work on newer distro versions as well. Please try the following:
```
cp /etc/apt/sources.list.d/meveric-buster-main.list /etc/apt/sources.list.d/meveric-stretch-main.list
cp /etc/apt/sources.list.d/meveric-buster-backports.list /etc/apt/sources.list.d/meveric-stretch-backports.list
sed -i 's/buster/stretch/g' /etc/apt/sources.list.d/meveric-buster-{main,backports}.list
```

MichaIng, 2020-03-14:
I just checked the other Odroids. Either the driver packages are in the Buster suite for those or in the SBC-specific (distro-independent) components. So Odroid C2 is a special case.

MichaIng, 2020-03-14:
I updated the Odroid C2 beta image with the above changes and some other minor tweaks + kernel upgrade.

ernero93, 2020-03-14:
I flashed the lastest beta image, but still had the same failure when installing kodi.

#### Details:
- Date           | Sat 14 Mar 21:00:38 GMT 2020
- Bug report     | e204dec9-4373-48fc-90d9-cf6c4addde61
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGI aml-libs-odroid mali450-odroid libump-odroid xf86-video-fbturbo-odroid
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Selecting previously unselected package libc-dev-bin.
(Reading database ... 17103 files and directories currently installed.)
Preparing to unpack .../0-libc-dev-bin_2.28-10_arm64.deb ...
Unpacking libc-dev-bin (2.28-10) ...
Selecting previously unselected package linux-libc-dev:arm64.
Preparing to unpack .../1-linux-libc-dev_4.19.98-1_arm64.deb ...
Unpacking linux-libc-dev:arm64 (4.19.98-1) ...
Selecting previously unselected package libc6-dev:arm64.
Preparing to unpack .../2-libc6-dev_2.28-10_arm64.deb ...
Unpacking libc6-dev:arm64 (2.28-10) ...
Selecting previously unselected package aml-libs-odroid.
Preparing to unpack .../3-aml-libs-odroid_0.0.1-2+deb9_arm64.deb ...
Unpacking aml-libs-odroid (0.0.1-2+deb9) ...
Selecting previously unselected package libump-odroid.
Preparing to unpack .../4-libump-odroid_0.0.1-r6p1-01rel0-1+deb9_arm64.deb ...
Unpacking libump-odroid (0.0.1-r6p1-01rel0-1+deb9) ...
Selecting previously unselected package mali450-odroid.
Preparing to unpack .../5-mali450-odroid_20161218-r6p1-1_arm64.deb ...
Unpacking mali450-odroid (20161218-r6p1-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-UFXFcr/5-mali450-odroid_20161218-r6p1-1_arm64.deb (--unpack):
 trying to overwrite '/usr/lib/aarch64-linux-gnu/libEGL.so.1', which is also in package libegl1:arm64 1.1.0-1
Selecting previously unselected package xf86-video-fbturbo-odroid.
Preparing to unpack .../6-xf86-video-fbturbo-odroid_20170625-1+deb9_arm64.deb ...
Unpacking xf86-video-fbturbo-odroid (20170625-1+deb9) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-UFXFcr/5-mali450-odroid_20161218-r6p1-1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-03-15:
@ernero93 
Please do:
```
apt purge libegl1
apt install mali450-odroid
```
Then retry install.

This is an issue due to Stretch => Buster OpenGL package renaming. The packages aim to replace the generic GL dispatchers. `libgles1-mesa` was the name on Debian Stretch but since Buster it's actually `libegl1` without mesa suffix.

I'll contact Meveric, probably we can repack the packages with the required additional package names it replaces.

ernero93, 2020-03-15:
Always no luck !!
What I did:
1) Fresh and completed install of beta build
2) ran the three commands
 sed -i 's/ mesa-utils mesa-utils-extra//' /DietPi/dietpi/dietpi-software
 apt purge libegl1 
apt install mali450-odroid xserver-xorg
all completed successfully.

3) Kodi installation.

What I get :

#### Details:
- Date           | Sun 15 Mar 13:43:35 GMT 2020
- Bug report     | b1d0dc17-ec70-4155-9e5c-c011919ff0f2
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.81+ #1 SMP PREEMPT Mon Jan 20 13:45:12 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGI libcec4 libcurl3-gnutls libnfs12 kodi-odroid
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...

#### Actual behaviour:

- ...
#### Extra details:

- ...

#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Let me know.
Regards

MichaIng, 2020-03-15:
@ernero93 
Can you run this manually and paste the full output:
```
apt install libcec4 libcurl3-gnutls libnfs12 kodi-odroid
```

ernero93, 2020-03-15:
root@DietPi:~# apt install libcec4 libcurl3-gnutls libnfs12 kodi-odroid
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package kodi-odroid
root@DietPi:~#

ernero93, 2020-03-15:
### 

### 
root@DietPi:~# apt install libcec4 libcurl3-gnutls libnfs12 kodi-odroid
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package kodi-odroid
root@DietPi:~#
****

MichaIng, 2020-03-15:
@ernero93 
Okay that is strange now since it's in the same repo as `xf86-video-fbturbo-odroid`, btw also not available in Meverics Buster repo yet for ARMv8, but ARMv7 only.

I'll run some tests here.

MichaIng, 2020-03-15:
@ernero93 
Okay the package is available but it depends on a bunch of Stretch library versions, impossible to get this running on Buster. So we need to wait for Meveric to port Kodi for ARMv8 onto Buster. It might not be trivial as this likely includes an upgrade to Kodi 18.

MichaIng, 2020-03-19:
Okay, the required packages have been ported by Meveric to his Buster suite:
- http://fuzon.co.uk/meveric/pool/main/x/
- http://fuzon.co.uk/meveric/pool/main/a/aml-libs-odroid/

Kodi for arm64 Odroid is still missing: http://fuzon.co.uk/meveric/pool/main/k/kodi-odroid/
However this is a larger task as well since it will include upgrade to version 18.5 as well.

MichaIng, 2020-03-23:
Okay for now dietpi-software will check on arm64 Odroids if the package is available, else disables Kodi for install: https://github.com/MichaIng/DietPi/commit/984d8e5e56ac556aca811c4725a1df9ef8b39ff0
So as fast as one is done, it will be available in DietPi as well. I add the workaround label for this to review and in case remove workaround with a later release.

MichaIng, 2020-10-02:
I'll mark this as closed. Will review state from time to time, however `dietpi-software` will automatically show Kodi for install as fast as it's available in the repository 🙂.

closed 2020-10-02 by MichaIng