MichaIng/DietPi #3419: (eth0) display error

msongz, 2020-03-15:
DietPi v6.28.0 : 00:45 - Mon 03/16/2020
---
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 34'C : 93'F (Cool runnings)
 - LAN IP : 192.168.2.213 (eth0)
---

1.using **wire** connection all the beginning
2.then set up **wireless** connection
3.**unplug** the wire(because route ip confilct)
4.reboot
5.ssh to pi 4 **wireless**
6.banner still showing **eth0** LAN IP but connected with wi-fi

Joulinar, 2020-03-15:
Hi,

Thy for your report. Do you use static IP on ethernet connection? Pls can you disable ethernet?

msongz, 2020-03-15:
@Joulinar no,i using DHCP all the way
after disable the eth, banner shows properly (wlan0) without reboot.

Joulinar, 2020-03-15:
Did you reboot your system after you unplug the eth cable ?

msongz, 2020-03-15:
@Joulinar yes, but do not remember the way, maybe using reboot command or just cut off the power

Joulinar, 2020-03-15:
Reboot is always the better option as DietPi contains some config files that are locked on RamDisk. 

Can you try to reactivate eth adapter? Does it change the banner?

MichaIng, 2020-03-15:
For reference: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/obtain_network_details
As long as there is any Ethernet adapter connected to any network, has an IP assigned and as well shows status "UP", it will be shown in banner, regardless if it is actually used for internet connection or not.

Since this behaviour is not optimal, I aim to change it: https://github.com/MichaIng/DietPi/projects?query=is%3Aopen+sort%3Acreated-asc
For now, as long as you use WiFi for internet connection, the Ethernet adapter must be either unplugged or kept unconfigured (disabled via `dietpi-config`) on boot to have the banner showing WiFi correctly. However this is mostly a visual one and has rarely any effect on our scripts, at least nothing serious.

MichaIng, 2021-11-06:
This has been fixed with DietPi v7.7 as now the default route has highest priority for the shown IP/interface. Only if no default route is assigned, Ethernet interfaces with same state (`UP`) have preference over WiFi interfaces, simply because this is the order in which `ip a` lists them :wink:.

closed 2021-11-06 by MichaIng