MichaIng/DietPi #3420: hdparm spindown not working after boot

lukaszm89, 2020-03-15:
- DietPi version | v6.28.0 (MichaIng/master) 
- Device model : RPi 4 Model B (armv7l)
- Distro | buster (ID=5)
- hdparm v9.58

I found a problem with setting spin down time for disks using dietp-drive_manager.
It is producing wrongly formatted **/etc/hdparm.conf** witch is than read by function **hdparm_options()** from **/lib/hdparm/hdparm-functions** during udev events handling.

**Current format (wrong):**

/dev/mmcblk0 
{
   apm = 127
   spindown_time = 24
}
/dev/sda 
{
   apm = 127
   spindown_time = 24
}

**Working version (correct):**

/dev/mmcblk0 **{**
   apm = 127
   spindown_time = 24
}
/dev/sda **{**
   apm = 127
   spindown_time = 24
}

First curly bracket needs to be in the same line as drive name. 
Fix needs to be made to lines 988-997 of dietpi-drive_manager file.

File format part from **man hdparm.conf**:
       /sbin/hdparm is not run unless a block of the form:

       DEV {

       option

       option

       }

MichaIng, 2020-03-16:
@lukaszm89 
Many thanks for your report. This seems to be the reason for #3309 as well. Interesting since somehow I remember that we changed this somewhere in the past due to wrong syntax. Probably something got changed back and forth, or we just did a mistake.

Okay while there is more to do, your suggestion can be done quickly for now. I also checked my own tests and indeed I also have `/dev/sda {` without a newline.

MichaIng, 2020-03-16:
@lukaszm89 
Out of interest related to the other issue, what does the following return in your case:
```
hdparm -B 127 /dev/sda
hdparm -B 127 /dev/mmcblk0
```

lukaszm89, 2020-03-16:
```
root@DietPi:~# hdparm -B 127 /dev/sda

/dev/sda:
 setting Advanced Power Management level to 0x7f (127)
 APM_level      = 127

root@DietPi:~# hdparm -B 127 /dev/mmcblk0

/dev/mmcblk0:
 setting Advanced Power Management level to 0x7f (127)
 HDIO_DRIVE_CMD failed: Invalid argument
 HDIO_DRIVE_CMD(identify) failed: Invalid argument

```

MichaIng, 2020-03-16:
@lukaszm89 
I found a better solution compared to adding a dedicated block for every drive, checking it it is even a spinning drive etc: https://github.com/MichaIng/DietPi/commit/8f5fe825d62aae43e7b479afb641ee0e40553555
If settings are added outside a block, they are interpreted as defaults to be used for all drives for which no block is present. Much cleaner, exactly what we want and compatible with all Debian versions.

GvY85, 2020-04-18:
Problem with this in general is that hdparm does not work with all externall hdd's. I have 2 Western Digital's (from 2019 an 2020) and a Seagate (from 2019) and all 3 are not supported by hdparm for some reason. 
After reading the guide linked below I just installed hd-idle (with hdparm still installed since dietpi-drove_manager needs it) and out of the box without any additional config (just apt install hd-idle) now all 3 hdd's, spread over 2 Intel mini-pc's spin down after 10min. 10min is fine for me so I didnt bother trying to adjust this. Also, it doest seem to interfere with the excisting hdparm installation.
Perhaps dietpi-drive_manager should migrate to hd-idle? 

guide used: https://www.htpcguides.com/spin-down-and-manage-hard-drive-power-on-raspberry-pi/

MichaIng, 2020-04-21:
@GvY85 
Hmm, does `hdparm -S /dev/sdX` spin down the drive immediately?
Depending on hdparm version it might still be the reason that APM needs to be supported by the drive to have spindown applied on boot, while applying spindown manually actually works. To check: `hdparm -B 127 /dev/sdX`
But the last time I tested, it should work at least until Buster and with v6.29 on Bullseye as well: https://github.com/MichaIng/DietPi/issues/3309#issuecomment-599549655

GvY85, 2020-04-22:
Hey,
I removed hdparm and dont want to fiddle with it because I finally got my drives spinning down (had Smartmontools also installed and apperently that creates a service called 'smartd' that spins up the drive every half hour......)
But, what I remember is that while hdparm gives some errors it can actually spin down the drive with 'hdparm -S /dev/sdX' and even seems to set the APM bit with 'hdparm -B 127 /dev/sdX' sometimes, while it does give errors, but never persistent throught reboots. 
For now, what you have in 6.29 works better than in <6.29 and should do it for most people if hdparm supports the drive.
hd-idle is indeed mucht simpler and lightweight but yes, only comes in backports but I think planning to change hdparm to hd-idle in 6.30 sound like a nice improvement.

MichaIng, 2020-04-22:
@GvY85 
>  with 'hdparm -S /dev/sdX' and even seems to set the APM bit with 'hdparm -B 127 /dev/sdX' sometimes, while it does give errors, but never persistent throught reboots.

Yes, setting spindown/APM via hdparm command is not boot persistent. For this /etc/hdparm.conf needs to be adjusted. However there is this issue (the other linked issue above) where the hdparm udev rules do not apply spindown times on boot if the drive does not support APM. There have been made some changes from Debian Stretch to Buster to Bullseye and it is kinda messy with different scripts being invoked.

`hdparm -I /dev/sdX` should btw read out all information including APM (Advanced Power Management) support.

closed 2020-05-03 by MichaIng