MichaIng/DietPi #3421: Add 32bit EFI support on x86_64 hardware

lhartmann, 2020-03-16:
Intel Atom Z3735F based windows 8/10 tablets are now sold extremely cheap, unsupported by newer windows releases, but still pretty nice platforms. Unfortunately they run 32bit EFI on a 64bit processor, which prevents the current Native EFI image from booting. 

The clonezilla-based native installer works perfectly, but the installed image lacks 32bit EFI support for grub.

I was able to boot the installed system by copying clonezilla's grub, i386-efi modules, kernel, and initrd into the FAT32 partition, and setting a grub.cfg accordingly. Too much of a hack to release, though.

MichaIng, 2020-03-16:
@lhartmann 
Many thanks for your request.

Please see: #1531

Just to be sure, is really the i386 version of kernel and initrd required? I.e. is boot broken if you install/reconfigure `linux-image-amd64` and/or run `update-initramfs -u`? This would mean that the system itself runs as i386, e.g. `uname -m` shows this?
- This would then mean that we cannot add support for this.

From what I see, the [grub-efi-ia32](https://packages.debian.org/grub-efi-ia32) packages can be installed on amd64/x86_64 systems, hence package dependency-wise do not require an i386 kernel or initrd.
- In this case, support would be thinkable, either via separate image or via selection in Clonezilla installer. If there is a reliable way to auto-detect if a system runs with x64 or x32 EFI, that could be even automated.

lhartmann, 2020-03-16:
Ok, I got things a little bit cleaner now, DietPi minimal install is running just about perfect.

- Installed DietPi normally, via clonezilla-based tool, to an external SD (not the internal eMMC).
- Built 32bit EFI grub following [this tutorial](https://github.com/jfwells/linux-asus-t100ta/tree/master/boot) on my desktop.
- Copied resulting `bootia32.efi` to the EFI partition.
- Copied resulting `*.mod` to `/boot/grub/ia32-efi/*.mod`.
- Created a 1.5GB swap partition, otherwise first-run wizard breaks in cryptic ways due to lack of memory (1GB only on my system).
- Copied `(hd0,gpt1)/EFI/debian/grub.cfg` to `(hd0,gpt1)/boot/grub/grub.efi`, because I do not know the right way to make grub find it's config.

Kernel, initrd and everything else unchanged.

As far as I can tell, both 32 and 64 bit versions of grub can coexist in the file system. lubuntu, clonezilla and debian-multiarch have that.

```
root@DietPi:~# uname -a
Linux DietPi 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
```

MichaIng, 2020-03-17:
@lhartmann 
Indeed then it should be possible to simply install the i386 grub EFI packages on the initial image before wrapping it into the Clonezilla installer. The the hardware seems to grab the right matching one then automatically. I'll do that with the next image.